Sloeber/arduino-eclipse-plugin #1251: Flashing different partitions

sbehnsen, 2020-11-12:
The function to flash an image into the target under Sloeber is very nice. I particulary like that I don't have to stop the serial monitor (using and ESP32 target). I can just click flash target and the serial connection is automatically stopped, the flash procedure takes place, and then the serial connection works again.

Now to my issue: There are some more partitions that are not flashed in this procedure, and also they typically are flashed seperately. For example a SPIFFS Filesystem or NVR Memory. 
It is easy to create a make target under Eclipse to do this job, be in this case the serial monitor must be manually deactivated before and reactivated after the flash procedure... 

I am looking for two kinds of solution:

1. A way to put this deactivation and reactivation of the serial monitor as a call into the makefile. Basically a small tool that would do this (not sure if this can be done easily).

2. A new feature in Sloeber to configure different flash jobs to be called with one click.  
Basically to have a flash all partitions, flash only the App, or App + SPIFF or just SPIFFS etc...

jantje, 2020-11-12:
I'm not well known to ESP32 so it is not easy for me to picture your need.
How are these other "flashing commands" defined in Arduino world? Are they in the platform.txt? boards.txt? platform.txt? or not defined at all?
Because if they are defined it should be as simple as changing the uploader (Though I understand that is not really what you want)
I'm not sure why you would want to use makefiles to do this. The upload itself is not done in a makefile. A eclipse command would be lots better but that will need some work to get to the Sloeber/CDT info

FYI I see a link to issue #585 as the bootloader should also have to stop the serial monitor and restart it.

sbehnsen, 2020-11-13:
The tools are external programms tools called from Eclipse/Sloeber. Typically esptool.exe (or esptool.py to run under python) to flash. There are other tools like mkspiffs.exe to generate an image for the SPIFFS filesystem.

> FYI I see a link to issue #585 as the bootloader should also have to stop the serial monitor and restart it.
Yes this is exactly what I was referencing to. It is done in there somewhere, and the solution I was looking for was a way to call this functionality somehow.

Why am I using makefiles (makefile.init to be exact) for this: By adding a maketarget I get a "button" (the make target in the project Explorer) to call this function with one click.
So for example I have a maketarget that will generate a SPIFFS filesystem image (from one directory in the project) and then flash this image. That is one click. Very convinient. Also this is part of the prject and henace can be configured project specific.
I have another maketarget to clear the NVR memory partition from the flash. Another one could flash a predetermined NVR memory partition... This simply is the most convinient way to do this.
The only slightly unconvinient thing is that the serial monitor needs to be stoped before and restarted after. Basically if there is a way to call that functionality than the rest can be solved with the "normal" features within Eclipse.

And yes, some things are defined through the boards.txt and platform.txt files. But at the end there is only one Flash button in Sloeber. It may be possible to change the behaviour to have for example the SPIFFS partition included in the flash process. But since these things are not the same in all projects (e.g. the partition scheme changes) it again is an issue.

I have one other idea: 
Since there are different partition schemes  (e.g. ...arduinoPlugin\packages\esp32\hardware\esp32\1.0.4\tools\partitions\...) maybe it would be possible to have the functionalitly to flash the entire set or just a configurable subset from Sloeber). So basically have a "flash target" button with a drop down list of different configurable flash target variants...
But this is obviously a feature that would need some work to be implemented. Access to a serial monitor on/off function that we can but in makefiles (or whatever people use for their build process) would probably be much easier to do.

jantje, 2020-11-13:
I didn't know about this makefile feature integration with CDT. Interesting.
Unfortunately I do not think it is possible to put a command in the makefile to tell sloeber to disconnect the serial.
This is one of the reasons why the upload is not in the makefile but a command run in Sloeber.
You do not want to disconnect the serial monitor when building so one should know whether disconnection is needed or not (this could be based on the targetname though).
I should check whether I can add pre/post commands to the build.

sbehnsen, 2020-11-13:
Can you point me to the place in the source where this serial monitor connect/disconnect is done? Maybe I can find a way...

jantje, 2020-11-13:
It is in the api part 
https://github.com/Sloeber/arduino-eclipse-plugin/blob/master/io.sloeber.core/src/io/sloeber/core/api/SerialManager.java#L39
look at the variable WeStoppedTheComPort to see how Sloeber does it during upload
https://github.com/Sloeber/arduino-eclipse-plugin/blob/master/io.sloeber.core/src/io/sloeber/core/tools/uploaders/UploadSketchWrapper.java#L192

jantje, 2021-07-17:
Did you make progress?

sbehnsen, 2021-07-17:
Unfortunately I have had no time to look into this issue. Since it is mostly an inconvinience it is not that high on my priority list :-/

jantje, 2021-08-22:
As I've been working on #585 I will add a stop and start for running targets.
I'm implementing a stop and start the serial communication based on a list of target names.

So basically
If the target you are building is in the list of targetnames (managed in the gui) and the project contains a serial port name and that serial port name is connected in the serial port; disconnect; build the target; reconnect else build the target

Does that suit your needs?

sbehnsen, 2021-08-23:
Yes, that is perfect !!! Thank you.