Sloeber/arduino-eclipse-plugin #1254: Upload is not working

wimjongman, 2020-11-14:
I am working on a new projject and I reinstalled sloeber in c:\apps\sloeber

I can't get the sketch to compile, any idea?

Starting upload
Uploading project "blink" with "AVRDUDE"
Starting reset using DTR toggle process  -> I see board action
Continuing to use "COM3"
Ending reset

Launching:  
Output:
Cannot run program "" (in directory "C:\Users\jongw\workspaces\slbr"): CreateProcess error=87, The parameter is incorrect
failed to upload "L/blink/blink}/Release/blink.hex"

[image]

wimjongman, 2020-11-14:
Ok, I see many duplicates and the resolution is to poke around in the settings but leaving it like this is not acceptable. The packagemanager should continue when there is an issue downloading.

Stacktrace is:

java.io.IOException: Failed to download url http://downloads.arduino.cc/cores/ArduinoCore-mbed-nonexistant.tar.bz2 error code is: 404
	at io.sloeber.core.api.PackageManager.myCopy(PackageManager.java:778)
	at io.sloeber.core.api.PackageManager.myCopy(PackageManager.java:740)
	at io.sloeber.core.managers.InternalPackageManager.downloadAndInstall(InternalPackageManager.java:351)
	at io.sloeber.core.managers.InternalPackageManager.downloadAndInstall(InternalPackageManager.java:130)
	at io.sloeber.core.managers.InternalPackageManager.startup_Pluging(InternalPackageManager.java:101)
	at io.sloeber.core.Activator$2.run(Activator.java:247)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:63)

wimjongman, 2020-11-14:
Ok, im running my own build and it seems resolved. Time for a new release?

jantje, 2020-11-14:
I've been planning on a new release because of CDT10 and all related changes (like no longer uppercasing in windows which may be related to your issue).
However there are still some loose ends I'm working on and I don't think lots of people tested yet.

wimjongman, 2020-11-14:
Can't we release a patch on the current version that solves the crashing of the config?

jantje, 2020-11-14:
The crashing of the config is caused by arduino including "bad pl;atforms" in their json file with Sloeber taking this bad one :-(

wimjongman, 2020-11-14:
Why is it not crashing with the manual build? Did you already make a patch for it?

wimjongman, 2020-11-14:
Ok, I was referring to the wrong issue. I removed all unrelated comments and answers.

This is about the upload not working. Is this caused by CDT10 you think?

The first message I see is:

`
A.tools.avrdude.upload..pattern : not found in the platform.txt file`

There is a double point between upload and pattern. Is that a red herring?

wimjongman, 2020-11-14:
I assume the platform.txt is refering to pre_processing_platform.txt which is rebuilt one every restart. My file does not conain the menioned key ^^
[pre_processing_platform.txt](https://github.com/Sloeber/arduino-eclipse-plugin/files/5541374/pre_processing_platform.txt)

jantje, 2020-11-14:
you can't use cdt10 with the latest sloeber release on windows
why 
pre cdt10 all environment variables were uppercased on windows
post cdt10 no uppercasing happens
cdt has always been case sensitive
as the current final release was pre cdt 10 it assumes cdt uppercases the environment variables

wimjongman, 2020-11-14:
I debugged the program and found a bug in the code below. When the networkPrefix is empty, the key will get two dots and can not be resolved to a value. When the dot is removed, it is reolved.

'
        if (isNetworkUpload()) {
            networkPrefix = Const.NETWORK_PREFIX;
        }
        String key = Const.A_TOOLS + upLoadTool + DOT + action + DOT + networkPrefix + DOT + Const.PATTERN;
'

Still this does not solve it completely,

wimjongman, 2020-11-14:
`
Launching:  -C -patmega328p -carduino -PCOM3 -b115200 -D -Uflash:w:C:\Users\jongw\workspaces\runtime-arduino.product\sloeber\Release/sloeber.hex:i 
Output:
Cannot run program "" (in directory "C:\Users\jongw\workspaces\runtime-arduino.product"): CreateProcess error=87, The parameter is incorrect
`

What should it put there? "make" or "avrdude"?

jantje, 2020-11-15:
>What should it put there? "make" or "avrdude"?

That is defined by the boards.txt/platform.txt

jantje, 2020-11-15:
The DOT problem is indeed a bug but it is a bug in the nightly introduced during the upperCasing removal.
And why I'm reluctant to release

jantje, 2020-11-15:
The DOT is not the only problem. Seems workaround needs a fix to.

jantje, 2020-11-15:
Works now on my system

wimjongman, 2020-11-15:
Can you push to a branch.

wimjongman, 2020-11-15:
> 
> 
> > What should it put there? "make" or "avrdude"?
> 
> That is defined by the boards.txt/platform.txt

What is the key for that specific part?

wimjongman, 2020-11-15:
Ok, I looked at platform.txt. It contains the following:

```
tools.avrdude.cmd.path={path}/bin/avrdude
<snip>
tools.avrdude.upload.pattern="{cmd.path}" "-C{config.path}" {upload.verbose} {upload.verify} -p{build.mcu}
```
It looks to me that "cmd.path" should be "tools.avrdude.cmd.path"

jantje, 2020-11-15:
>It looks to me that "cmd.path" should be "tools.avrdude.cmd.path"

yes
That is fixed in https://github.com/Sloeber/arduino-eclipse-plugin/commit/52a3422bc73307e37393e2144dd706a78224cef6
There was still a toupper

jantje, 2020-11-15:
sorry should have been https://github.com/Sloeber/arduino-eclipse-plugin/commit/1a25c1b421fa23010f7523ce32dda1b66b343dc4
The other is the cosmetic stuff

wimjongman, 2020-11-15:
I'm running with that commit. Do I have to remove everything first?

jantje, 2020-11-15:
please see your whatsapp/mail I'm waiting in video chat for you

wimjongman, 2020-11-15:
> 
> 
> I'm running with that commit. Do I have to remove everything first?

I removed the arduinoPlugin directory and now it finally uploads.

closed 2020-11-16 by jantje