Sloeber/arduino-eclipse-plugin #1257: jssc crashed Java 11

wimjongman, 2020-11-14:
I have started devmode to figure out what the issue is with the uploader. I started sloeber from the IDE. Creating a sketch and compiling works fine but the avrdude upload crashes the jvm due to a jssc crash. 

I have found this issue:
[https://github.com/arduino/Arduino/issues/6788](https://github.com/arduino/Arduino/issues/6788
)
[javadump.txt](https://github.com/Sloeber/arduino-eclipse-plugin/files/5541432/javadump.txt)
[javacore.20201101.221210.14140.0002.txt](https://github.com/Sloeber/arduino-eclipse-plugin/files/5541434/javacore.20201101.221210.14140.0002.txt)

### Main stack dump snippet
3XMTHREADINFO      "Worker-3: avrdude" J9VMThread:0x00000000008DE700, omrthread_t:0x000002F3AF11CC30, java/lang/Thread:0x00000000E2050028, state:R, prio=5
3XMJAVALTHREAD            (java/lang/Thread getId:0x48, isDaemon:false)
3XMTHREADINFO1            (native thread ID:0xAB90, native priority:0x5, native policy:UNKNOWN, vmstate:R, vm thread flags:0x00000020)
3XMCPUTIME               CPU usage total: 0.203125000 secs, user: 0.015625000 secs, system: 0.187500000 secs, current category="Application"
3XMHEAPALLOC             Heap bytes allocated since last GC cycle=200400 (0x30ED0)
3XMTHREADINFO3           Java callstack:
4XESTACKTRACE                at jssc/SerialNativeInterface.openPort(Native Method)
4XESTACKTRACE                at jssc/SerialPort.openPort(SerialPort.java:158)
4XESTACKTRACE                at io/sloeber/core/api/Serial.connect(Serial.java:168)

wimjongman, 2020-11-14:
I found a newer version of the jssc wrapper that solved my issue.

https://github.com/java-native/jssc

I also noticed that cdt has their own serial bundle. I guess we should switch.

jantje, 2020-11-14:
I'm fine with that :-)
If we can use the cdt one that would be great

wimjongman, 2021-03-25:
I think this one is done. Correct?

jantje, 2021-03-25:
I think so. 
I'm trying to get my buildserver back up so nightly will be 4.4 hopefully soon

closed 2021-06-04 by jantje