Sloeber/arduino-eclipse-plugin #1264: Sloeber fails to install because it can't get data from a specific URL

ralphc-1965, 2020-11-18:
Once I install Sloeber it complains that the install can't be completed due to missing files.

From then on when restarted it gives the error about configuring Sloeber:
Failed to configer Sloeber
Failed to install [DEPRECATED] Arduino mbed-enabled Boards
Failed to download "http://downloads.arduino.cc/cores/ArduinoCore-mbed-nonexistant.tar.bz2".
Failed to download url http://downloads.arduino.cc/cores/ArduinoCore-mbed-nonexistant.tar.bz2 error code is: 404

Surprised that no one has an issue like this and created a bug report.

I am not behind a firewall but I am in the US on a standard US network (Frontier).

[image]

jantje, 2020-11-18:
This is duplicate of #1196

Sloeber does not come with any platform (read boards) so it tries to install a platform and the platform sloeber tries to install has been deprecated.
Solution
install the platform of your choice at window->preferences->arduino->platform&boards
press on the > untill you can select a version you want to install
select apply and close and wait for it to finish (look at bottom right)
You are ready to roll

ralphc-1965, 2020-11-19:
Thank you Jantje,

Sorry for writing a bogus problem report.

All works great now.

Ralph

jantje, 2020-11-22:
reopening because many people do not search the closed issues.

btribley, 2021-01-29:
I installed three times: twice on Windows because the first time I just left the path at the default and got the windows path nastygram. After installing in C:\eclipse and downloading the plugin ide I got this. Then I tried the plugin only and got this. Then I found the Sloeber one, installed it and got this. I was reaching for my Mac. Fortunately I found this post.

This is a nasty installer bug. No one will be able to install then start a project without getting knocked in the head. Suggest two possible solutions:
1. Strip out all default platforms and put up a big dialog box at the end of the install that says to go get a platform.
2. Instead of the completely cryptic error message of no board found in boards.txt at the expected location (which is empty) say "No boards found. Configure a platform in Windows-Preferences-Arduino and add a json file or board. 

That would fix it! Many thanks for putting this tool together. I am hoping to be able to use it for the new AVR128DA* series because the Atmel Studio 7 is totally broken for them. The Arduino IDE works after installing DxCore but I can't work with such a limited tool.

jantje, 2021-01-29:
Thanks for your input.

>Strip out all default platforms and put up a big dialog box at the end of the install that says to go get a platform.

I'm a strong believer of "working out of the box". That is why I opted to install a platform if none is present.
Unfortunately -after a Arduino json update- the "selection options" Sloeber uses no longer selected the arduino avr platform but selects a platform that Arduino made uninstallable (I would have removed the platform from the json)

>Instead of the completely cryptic error message of no board found in boards.txt at the expected location (which is empty) say "No boards found. Configure a platform in Windows-Preferences-Arduino and add a json file or board.

I get your point. The problem is ... this should never have happened. The root cause is Arduino who publishes a platform that can not be installed. Which -I assume by accident- is picked by Sloeber as the platform to install if no platform is found. 
When Sloeber starts the first time it adds: examples, make(on windows); libraries and a platform. Normally all this should work.
Because people start Sloeber and don't know this is happening the message is cryptic.
If the user initiated the installation of a platform the message would be way less cryptic. Especially because the name of the platform contains "deprecated" ;-)

closed 2021-07-11 by jantje