Sloeber/arduino-eclipse-plugin #1268: Better support for version control

jantje, 2020-11-26:
currently the Sloeber configuration data is stored in eclipse environment variables. This proposal is to use Sloeber dedicated files and a build environment variable provider instead.

The current solution forces the A. notation on Sloeber and makes checking in eclipse configuration files necessary.

This issue is to switch to using build environment variable provider and Sloeber configuration files
There would be one config file for each configuration so you do not need to share all your configs with other team members
The config file would contain json url, platform name, platform version,board id, board options, and other user settings
This allows to "download and install" the platform after install (future plans ;-) ) 
Plan is to avoid any platform dependent info (like path names) to ease checking in/out.

I'm not sure yet about where to set the upload port as this is system specific

jantje, 2020-12-01:
I turned off my build engine so this change will not affect the nightly untill I'm fully happy.
The main reason for this is that old projects will stop working with this change

jantje, 2020-12-22:
I'm making serious progress but there is still lots of work.

uzi18, 2021-01-09:
@jantje maybe just store port per platform, it is hard to get same port number on different windows machines, so almost always you need to change it

jantje, 2021-01-09:
The sloeber.cfg will not contain the port and upload protocol

uzi18, 2021-01-09:
Will try to investigate your changes but java is not my fluent language, in fact it is something new so maybe will learn something ;)

jantje, 2021-01-09:
Here is the boarddescription environment variable deliverer for version control
https://github.com/Sloeber/arduino-eclipse-plugin/blob/4b9e63db2b84b7831af53d68a8f1cc09e92e803e/io.sloeber.core/src/io/sloeber/core/api/BoardDescription.java#L823

closed 2021-06-04 by jantje