Sloeber/arduino-eclipse-plugin #1269: Using only "myfile.h" file without a "myfile.c", slobed does not include when compiling.

marcelocsi, 2020-12-01:
I created a sketch from scratch and followed directions to add custom libraries.
I have also couple custom files specific for my project and they are contained in subfolder.

The project structure is as follow:
> MySketch
    > libraries (includes correctly all libraries required by sloeber)
    > MySpecificStuff
          > MyCode1
                   > MyCode.cpp
                   > MyCode.h
          > MySettings
                   > MySettings.h
    > MySketch.ino

when I hit the verify button, the MySettings.h is not seen by a custom library file contained in the libraries folder. After reading issue #1249, I attempted creating a "MySettings.c" file and it worked like a charm.

Setup description:
- Windows 10 64-bit machine
- Eclipse IDE for C/C++ Developers (includes Incubating components) Version: 2020-09 (4.17.0).

jantje, 2020-12-01:
Please provide screendumps of the project setup (as working and as not working) and the compile command of sloeber.ino.cpp

I ask that because your information is inconsistent. you sayy "they are contained in subfolder' but your sketch folder does not contain subfolders 

That being said. If you want a header file to be found it needs to be in the "include path". 
Putting a header file in a subfolder means you have to do one of the option below 
1) include the header file like #include "[subfolder]/header file.h" (should also work in arduino ide) 
2) add [subfolder]  to the include path in project properties->C/C++ general-> paths and symbols (will not work in arduino ide)

marcelocsi, 2020-12-01:
There is no working project yet. :( My hapinness was too short. 

Here goes the requested images, there is not working:
[image]

Regarding your suggestions:
1. I want to avoid it since the files are locked up in a specific structure.
2. Does it work or not in the sloeber?

Ps.: curiosity, what is the meaning of sloeber?

jantje, 2020-12-01:
Why is your remark hiden?
Both 1 and 2 work in sloeber. IMHO 2  means that your project will not work in arduino IDE

marcelocsi, 2020-12-01:
> Why is your remark hiden?
> Both 1 and 2 work in sloeber. IMHO 2 means that your project will not work in arduino IDE

It was hidden because I was still placing information and doing some tests and you were not supposed to read it. lol 

I am still struggling now with 2, sloeber is too cool to get back to Arduino IDE.

Congrats for your job! I am gonna donate soon to support you.

marcelocsi, 2020-12-01:
@jantje my isseus seems to be gone and I am sharing few notes on my learning since it might help somebody that hits here.

1. adding the path to "paths & symbols" under C/C++ General tab solved the including issue.
2. when using sloeber, the camel case of #include "MyFile.h" must match the the name of the file. ie. myfile.h does not work if it includes "MyFile.h" -> hint: Arduino IDE does not care about it.
3. the Sloeber Arduino Menu then, Preferences, then Paths, helps to find only libraries.
4. even if the gcc is used amound Arduino IDE and Eclipse with Sloeber, its settings might be different and several other errors might occur. Arduino was compiling with some non ANSI-C string management while gcc for Sloeber is complaining a lot.

marcelocsi, 2020-12-01:
@jantje I would like to suggest to automatically add to the top of "Path & Symbols" anything that it is not pointed in the Arduino menu. It looks like Eclipse could see and properly find the references while compiler was expecting to add the include. It would be very helpful if slober make does look through folders within the Eclipse project and automatically addes the folder with .h, .c/.cpp files to the makefile.

jantje, 2020-12-01:
>It looks like Eclipse could see and properly find the references while compiler was expecting to add the include. 

Yes, the CDT indexer does not look at the compile include path nor to the includes in your source file. That is for a very good reason. 

>I would like to suggest to automatically add to the top of "Path & Symbols" anything that it is not pointed in the Arduino menu

Sloeber is build on top of CDT. "Path & Symbols" is a CDT thing so it doesn't know about arduino/sloeber so it cant sort based on.
Note that I call the need to modify the "Path & Symbols" advanced usage and it will not survive a git checkin and checkout on another computer.

> It would be very helpful if slober make does look through folders within the Eclipse project and automatically addes the folder with .h, .c/.cpp files to the makefile.

That would be very un C/C++ like

marcelocsi, 2020-12-01:
@jantje I am not sure about this issue now; I have a library which cames by default with Arduino IDE and I moved the library to my one, see the picture. The bluetooth one can't be seen at all while the others can be seen.

Any clue on what I am doing wrong are very welcome!

jantje, 2020-12-01:
It is unclear to me what you are trying to do.
I do feel however you are trying to bite off more than you can chew.
Sloeber does lots of things to make arduino world work in plain old C by telling CDT (build on eclipse) things. As soon as you step away from default arduino way you Sloeber doesn't tell anything to CDT and you need to tell these things to CDT yourself.
If you are not well known to Sloeber that means learning Sloeber, arduino framework, CDT and eclipse. That is not a comfortable chew.

marcelocsi, 2020-12-01:
Well, if I am not able to chew, then I have to slice it down in parts where I can learn.

I am trying to make life easier by properly setting the Git as versioning tool. So far, Arduino IDE is a pain in the ass when you wanna deal with version. I worked for years in a company where for each module we had strict control of versions. I wanna apply the same concept for my personal projects.

1. application specific: stuff specific for each project and also paremeters for each project as well
2. use library from vender, like bluetooth, wifi, etc.
3. my custom library which depends on parameters from 1 and talks to 2 and do something according to my needs and might be shared with other project as well.

I am struggling on 2. I don't want to use automatic version from the cloud, even if it has fixes. if so, I wanna test each of them for me before accepting them. My computer just crashed and I realized Arduino IDE does not have this control easily when you reinstall everything. For sloeber, I wanna setup each module (set of .h/.cpp files) under spefic versioning and install them directly by Git itself. The only bad part here is the manual setup of the Eclipse project, well, I am preparing also a template.

The weird part is the "BLEDevice.h" file. Eclipse cant see neither in the esp32/hardware folder nor in the folder I copy. In my folder, I can see other libraries I manually copied but I can't see this one. Maybe there is a flag somewhere that expects something and blocks the filter of my eclipse? I am guessing how to overcome.

jantje, 2020-12-01:
> So far, Arduino IDE is a pain in the ass when you wanna deal with version

One of the big selling points of Sloeber

>I don't want to use automatic version from the cloud, even if it has fixes

Nor do I. That is why a sloeber project points to a dedicated platform version and you can have multiple version of a platform.
This way each and every project can be on a different version.

>I realized Arduino IDE does not have this control easily when you reinstall everything

At install time both Sloeber and arduino IDE install the latest version. For both you can change that to a older version (actually I'm not sure this is true for arduino avr). The difference is that
A) Arduino ide only allows 1 version of a platform
B) Arduino ide does not store the version used in the project.

>, I wanna setup each module (set of .h/.cpp files) under spefic versioning and install them directly by Git itself. 

For my main project I've taken that route, However it is hard, I mean really hard (And I have been a version architect for many years)
The first problem is that Sloeber does not allow for different version of libraries.It is technically very complex to do so.
The second problem is that most code comes already from a repository. So the problem you have is ... how do I sync over multiple repositories.
I opted to have all my personal libraries in 1 github repository (actually 2;  one I share and one I don't share). This way I have 3 projects in my workspace 
1 shared libraries repository https://github.com/jantje/marvinLibraries
1 private libraries repository
1 sloeber project 
By adding the 2 library repositories to the private libraries folder this works like a charm except for
1)Because files are linked in eclipse, eclipse does not always detect that the project has changed. You need to refresh the projects from time to time
2) As it are 3 repositories you need to checkin/tag/... to keep the 3 repositories in sync
3)Arduino opted for 1 repository=1library approach. So you are Arduino incompatible. 
4) There is no github zip a subfolder thing making zips for the arduino IDE not possible.
5)When looking at the libraries folder there is no easy way to see wether libraries are used outside of you 2 repositories. Welll you will have hardware libraries which is ok because the platform is versioned.

Add to that that if you want top use a github library you may need to fork it (and then keep it in sync) and add it to the private libraries one by one. You could use 2 github homes.

As you can see it becomes a narrow path.
As to version control, I'm currently working on #1268 to make version control easier.

marcelocsi, 2020-12-01:
I got your point and I am pretty much on the same page.

The approach I am using is to use Git to control the version. I need to get used to the "Git team menu" in the Eclipse because I am using the client TortoiseGit on Windows, so, there is no "implicit communication" between git and sloeber, since I am doing the file management outside the Eclipse.

For the version, yes, I am gonna have three projects on the Git. One is the main project itself, and the other two will be seem as libraries (one for Arduino vendor source code and other one is my libraries) and those latter two will be added as sub projects to the main project. I made some tests and it looks fine from the TortoiseGit/Windows explorer perspective where I can control each subproject, pull requests, etc. right now, I am struggling to make Sloeber just recognize the source code in the paths I am placing the code.

I read #1268 and I have the feeling you are redoing the versioning control by proposing a new one; maybe the approach I am taking solves what you are doing.

jantje, 2020-12-01:
Redoing version control.. no way. I'm not smart enough for that rofl
Currently you need to check in .cproject .project and settings info to get a reproducible result
What I'm doing is that you should only check in a sloeber.txt file and after import recreate the project.
This because currently system dependent info (like / versus \ and upload ports) is mixed with info you want in version control

marcelocsi, 2020-12-02:
Oh I see what you are doing, hopefully it works! 

By the way, I realized what was missing. I have a folder with named BLE in my library folder however, it didn't have any BLE.h nor BLE.cpp file in there. Once I created dummies files named BLE.h and BLE.cpp without any content and included BLE.h in any of my files, they showed up and all source files were compiled.

If possible, all header and source files pointed by that Arduino hardware/library folders could be compiled and let the combiner (I know it as linker) decide what symbols will be linked?

P.s.: I followed your advice and had too much beers, time to sleep now!

closed 2020-12-02 by marcelocsi

jantje, 2020-12-02:
>I followed your advice and had too much beers, 

😆 I don't recall advising you to drink to much beers 🥼

marcelocsi, 2020-12-02:
> > I followed your advice and had too much beers,
> 
> 😆 I don't recall advising you to drink to much beers 🥼

I read somewhere else in other issue I guess, anyway, I accepted. lol