Sloeber/arduino-eclipse-plugin #1270: How do I configure boards.txt in Sloeber?

marcelocsi, 2020-12-02:
I am coming from Arduino IDE and I have an ESP32 project running fine with some partition made in the boards.txt.
I setup the same project in Sloeber and I could compile and link without major issues, however, after flashing, the board is reseting. It was same symptom I had before configuring properly the boards.txt in the Arduino IDE, so, how do I set boards.txt with partitions in Sloeber?

jantje, 2020-12-02:
In the project explorer, goto your project->core->variant right click and select view in system explorer
In the parent folder you can find the boards.txt (newer versions will also have a boards.sloeber.txt if so delete that one)
After you changed the boards.txt goto your project properties->arduino select apply and close

marcelocsi, 2020-12-02:
It worked like a charm! 

Hey, give me directions to donate for you man; I do appreciate your very short time to respond and precise responses.

closed 2020-12-02 by marcelocsi