Sloeber/arduino-eclipse-plugin #1274: recipe.hooks.linking.prelink.NN with NN > 10 are not executed

hreintke, 2020-12-10:
Hi,

Sloeber 4.3.3, Windows 10

In the esp8266 platform there is a commit which updates platform.txt 
from : 
```
## Build the app.ld linker file
recipe.hooks.linking.prelink.1.pattern="{compiler.path}{compiler.c.cmd}" -CC -E -P {build.vtable_flags} "{runtime.platform.path}/tools/sdk/ld/eagle.app.v6.common.ld.h" -o "{build.path}/local.eagle.app.v6.common.ld"
```
to : 
```
## Build the app.ld linker file
recipe.hooks.linking.prelink.11.pattern=mkdir -p "{build.path}/ld_h/"
recipe.hooks.linking.prelink.12.pattern=cp "{runtime.platform.path}/tools/sdk/ld/{build.flash_ld}" "{build.path}/ld_h/local.eagle.flash.ld.h"
recipe.hooks.linking.prelink.13.pattern="{compiler.path}{compiler.c.cmd}" -CC -E -P {build.vtable_flags} {build.mmuflags} "{build.path}/ld_h/local.eagle.flash.ld.h" -o "{build.path}/local.eagle.flash.ld"
recipe.hooks.linking.prelink.14.pattern="{compiler.path}{compiler.c.cmd}" -CC -E -P {build.vtable_flags} {build.mmuflags} "{runtime.platform.path}/tools/sdk/ld/eagle.app.v6.common.ld.h" -o "{build.path}/local.eagle.app.v6.common.ld"
recipe.hooks.linking.prelink.11.pattern.windows=cmd /v:on /e:on /c "if not exist {build.path}\ld_h\ (mkdir {build.path}\ld_h\ )"
recipe.hooks.linking.prelink.12.pattern.windows=copy "{runtime.platform.path}\tools\sdk\ld\\{build.flash_ld}" "{build.path}\ld_h\local.eagle.flash.ld.h"
```
After that commit the prelink commands are not executed.

Changing to NN < 10 : 

```
## Build the app.ld linker file
recipe.hooks.linking.prelink.1.pattern=mkdir -p "{build.path}/ld_h/"
recipe.hooks.linking.prelink.2.pattern=cp "{runtime.platform.path}/tools/sdk/ld/{build.flash_ld}" "{build.path}/ld_h/local.eagle.flash.ld.h"
recipe.hooks.linking.prelink.3.pattern="{compiler.path}{compiler.c.cmd}" -CC -E -P {build.vtable_flags} {build.mmuflags} "{build.path}/ld_h/local.eagle.flash.ld.h" -o "{build.path}/local.eagle.flash.ld"
recipe.hooks.linking.prelink.4.pattern="{compiler.path}{compiler.c.cmd}" -CC -E -P {build.vtable_flags} {build.mmuflags} "{runtime.platform.path}/tools/sdk/ld/eagle.app.v6.common.ld.h" -o "{build.path}/local.eagle.app.v6.common.ld"
recipe.hooks.linking.prelink.1.pattern.windows=cmd /v:on /e:on /c "if not exist {build.path}\ld_h\ (mkdir {build.path}\ld_h\ )"
recipe.hooks.linking.prelink.2.pattern.windows=copy "{runtime.platform.path}\tools\sdk\ld\\{build.flash_ld}" "{build.path}\ld_h\local.eagle.flash.ld.h"
```
Makes it run OK.

jantje, 2020-12-10:
sloeber stops at the first non existing number
So that seems to be compliant with what you see

jantje, 2020-12-10:
Checking https://arduino.github.io/arduino-cli/latest/platform-specification/ it doesn't say you need to start at 1 and have consequent numbers. It does say however:
"If you need to write more than 10 recipes for a single hook, pad the number with a zero, for example:"
Which does imply numbers should start at 1 and no gaps.

hreintke, 2020-12-10:
Others read the same documentation and come to other conclusion -> all should be executed based on the sorted number.

This issue is solved by esp8266 with changing numbering starting from 1.
https://github.com/esp8266/Arduino/pull/7752

I you agree, I keep this issue open as I also have some problems with added recipes running itself.

hreintke, 2020-12-11:
OK, This is what I think is happening.

One of the new prelink commands is : 

`recipe.hooks.linking.prelink.2.pattern.windows=cmd /c "copy {runtime.platform.path}\tools\sdk\ld\{build.flash_ld}" "{build.path}\ld_h\local.eagle.flash.ld.h"`

That expands to : 

`cmd /c "copy C:/Userdata/ArduinoGit_HR\tools\sdk\ld\eagle.flash.4m2m.ld" "C:\Userdata\ESP8266\Projects\MDNSWebserver\Release\ld_h\local.eagle.flash.ld.h"`

which results in : 

`The syntax of the command is incorrect.`

The problem is the forward slashes in the first part of the copy `"copy C:/Userdata/ArduinoGit_HR`
Replacing that with `"copy C:\Userdata\ArduinoGit_HR` makes it work OK.

The environment variables for this are : 

[image]

As this is from a private hardware path, I also checked a boardmanager installed version.
That has the same problem.

[image]

My version of Sloeber is : 

[image]

If you need more info/tests just let me know.

jantje, 2020-12-11:
I think the \ / mix is fixed in the nightly

jantje, 2020-12-11:
http://eclipse.baeyens.it/nightly.php

hreintke, 2020-12-11:
Following that I get to the site : 
[image]

Following the mentioned link `This download page is the travis build. Download the tar.gz for your os.`
I come to the page I mentioned with the outdated versions.

I do not want to upgrade my current working version until I know it works OK.
So I downloaded the current release version and tried to get the update with no luck.

Also tried to build from git master and run, but in that I cannot create create any new sketch.

Prefer to have a clean download of the nightly for windows 64 bits (or a version which should have fixed the issue) to verify.

BTW : As I read there is quite some development on nightly now. 
What is the current status of nightly. Should it work OK or ids it a WIP with known/unknown issues ?

jantje, 2020-12-11:
I switched off nightly creation as master is WIP #1268 
That page needs so me updates but http://eclipse.baeyens.it/update/V4/nightly is the update site

hreintke, 2020-12-12:
Finally got it running.
Would be much easier/cleaner when nightly is just the zip from the build.
The there is also no risk that old configuration is taken over.

The issue mentioned above is solved in this version.

What doen't work is creating new cpp project.
It creates : 
```
// Only modify this file to include
// - function definitions (prototypes)
// - include files
// - extern variable definitions
// In the appropriate section

#ifndef _{title}_H_
#define _{title}_H_
#include "{Include}"
//add your includes for the project {title} here

//end of add your includes here

//add your function definitions for the project {title} here

//Do not add code below this line
#endif /* _{title}_H_ */
```
and
```
// Do not remove the include below
#include "{title}.h"

//The setup function is called once at startup of the sketch
void setup()
{
// Add your initialization code here
}

// The loop function is called in an endless loop
void loop()
{
//Add your repeated code here
}
```

Creating a new ino project works OK.

jantje, 2020-12-12:
The cpp project creation problem is regression from https://github.com/Sloeber/arduino-eclipse-plugin/commit/86d89b7b004f37c91d4f208ac1fe647aff9adbdf at 25 sept

hreintke, 2020-12-14:
Too fast.
I do have issues running sloeber with java later than 8.
Will open a new issue as that is not related to the recipe.prelink

closed 2020-12-14 by hreintke