Sloeber/arduino-eclipse-plugin #1276: [proposal] Automatically build releases with maven from GitHub Actions

Kashouryo, 2020-12-12:
I am sorry to hear that your server got ddosed.
As an alternative way to provide nightly builds, you can try using GitHub Actions with maven to automatically publish nightly releases when you push a change to the code base.
https://docs.github.com/en/free-pro-team@latest/actions/guides/building-and-testing-java-with-maven

jantje, 2020-12-12:
Thanks for the info. 
First of I should remove the ddos message on the site. There used to be a list of all versions on the site and I had to remove that. This message is related to build output storage that used to be on baeyens.it but had to be moved because of the ddos attacj.
is no longer relevant.

The build is a maven build and is done with travis right now. https://travis-ci.org/github/Sloeber/arduino-eclipse-plugin
There is a long lasting issue #641 on the travis build and a new problem now with maven download
```
$ sh -c "if $TRAVIS_SECURE_ENV_VARS; then echo Travis SECURE available -- Deploy; else echo Travis SECURE NOT available --  Verify; fi"

Travis SECURE available -- Deploy

The command "sh -c "if $TRAVIS_SECURE_ENV_VARS; then echo Travis SECURE available -- Deploy; else echo Travis SECURE NOT available --  Verify; fi"" exited with 0.

360.03s$ sh -c "if $TRAVIS_SECURE_ENV_VARS; then ./travis_exec_and_print_time.sh 'mvn verify -Dtest=RegressionTest  --quiet --settings settings.xml'; else ./travis_exec_and_print_time.sh 'mvn verify -Dtest=RegressionTest --quiet --settings settings.xml'; fi"

Exec and print time periodically - overcoming long quiet times

CMD=mvn verify -Dtest=RegressionTest  --quiet --settings settings.xml

Testing for PID=4397

PID=4397 still running at 180 seconds

Dec 05, 2020 11:39:54 PM org.apache.http.impl.execchain.RetryExec execute

INFO: I/O exception (org.apache.http.NoHttpResponseException) caught when processing request to {s}->https://eclipse.mirror.rafal.ca:443: The target server failed to respond
```
@wimjongman  @rlogiacco should we consider moving travis build to github.
The first thing I notice is that travis uses linux and github seems to use windows/mac.

rlogiacco, 2021-01-08:
Apologise @jantje for the late answer, anyways it looks like you fixed this yourself as travis build is back online... I personally prefer to build on unix systems if possible than mac/windows, so my opinion is _stay on travis, if possible_

jantje, 2021-01-08:
@rlogiacco Thanks for answering and happy new year
I'm not sure about the travis state right now. There seems to be problems with accessing a download site and for all I know only god knows what the root cause of the problem is.
I agree on your "better to build on linux" mostly because right now I mostly work on windows.

wimjongman, 2021-06-18:
Fixed

closed 2021-06-18 by wimjongman