Sloeber/arduino-eclipse-plugin #1277: Sloeber crash when opening serial port, java version 10, 11, 15

hreintke, 2020-12-14:
Windows info : 

Edition	Windows 10 Home
Version	20H2
Installed on	‎13/‎11/‎2020
OS build	19042.685
Experience	Windows Feature Experience Pack 120.2212.551.0

Sloeber version :  both current and nightly
Multiple platforms same issue (STM32, ESP32, ESP8266)

When running with java version 8, current = OK, nightly -> not running due to java version.

When running either version with version 10, 11 or 15 sloeber crashes when opening serial port.
If no serial port connect there is then there is the correct message -> Port not found.
When connected this is from first part of the eclipse log : 

```
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  EXCEPTION_ACCESS_VIOLATION (0xc0000005) at pc=0x000000007110b5db, pid=8204, tid=4840
#
# JRE version: Java(TM) SE Runtime Environment (10.0.2+13) (build 10.0.2+13)
# Java VM: Java HotSpot(TM) 64-Bit Server VM (10.0.2+13, mixed mode, tiered, compressed oops, g1 gc, windows-amd64)
# Problematic frame:
# C  [jSSC-2.8_x86_64.dll+0xb5db]
#
# No core dump will be written. Minidumps are not enabled by default on client versions of Windows
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#

---------------  S U M M A R Y ------------

Command Line: -Dosgi.requiredJavaVersion=1.8 -Dorg.eclipse.update.reconcile=false -Xms40m -Xmx2048m --add-modules=ALL-SYSTEM C:\Utilities\Sloeber\\plugins/org.eclipse.equinox.launcher_1.5.500.v20190715-1310.jar -os win32 -ws win32 -arch x86_64 -showsplash -launcher C:\Utilities\Sloeber\sloeber-ide.exe -name Sloeber-ide --launcher.library C:\Utilities\Sloeber\\plugins/org.eclipse.equinox.launcher.win32.win32.x86_64_1.1.1100.v20190907-0426\eclipse_1902.dll -startup C:\Utilities\Sloeber\\plugins/org.eclipse.equinox.launcher_1.5.500.v20190715-1310.jar --launcher.appendVmargs -exitdata 1878_c4 -perspective io.sloeber.application.perspective -vm C:\Program Files (x86)\Common Files\Oracle\Java\javapath\javaw.exe -vmargs -Dosgi.requiredJavaVersion=1.8 -Dorg.eclipse.update.reconcile=false -Xms40m -Xmx2048m --add-modules=ALL-SYSTEM -jar C:\Utilities\Sloeber\\plugins/org.eclipse.equinox.launcher_1.5.500.v20190715-1310.jar

Host: Intel(R) Core(TM) i7-10510U CPU @ 1.80GHz, 8 cores, 15G,  Windows 10 , 64 bit Build 19041 (10.0.19041.662)
Time: Mon Dec 14 12:04:31 2020 W. Europe Standard Time elapsed time: 115 seconds (0d 0h 1m 55s)

---------------  T H R E A D  ---------------

Current thread (0x0000015973c9c800):  JavaThread "Worker-34: esptool" [_thread_in_native, id=4840, stack(0x0000004ea5b00000,0x0000004ea5c00000)]

Stack: [0x0000004ea5b00000,0x0000004ea5c00000],  sp=0x0000004ea5bfe8f0,  free space=1018k
Native frames: (J=compiled Java code, j=interpreted, Vv=VM code, C=native code)
C  [jSSC-2.8_x86_64.dll+0xb5db]

Java frames: (J=compiled Java code, j=interpreted, Vv=VM code)
j  jssc.SerialNativeInterface.openPort(Ljava/lang/String;Z)J+0
j  jssc.SerialPort.openPort()Z+65
j  io.sloeber.core.api.Serial.connect(I)V+28
j  io.sloeber.core.api.Serial.connect()V+2
j  io.sloeber.core.api.Serial.<init>(Ljava/lang/String;ICIFZ)V+99
j  io.sloeber.core.api.Serial.<init>(Ljava/lang/String;I)V+9
j  io.sloeber.core.communication.ArduinoSerial.makeArduinoUploadready(Lorg/eclipse/ui/console/MessageConsoleStream;Lorg/eclipse/core/resources/IProject;Lorg/eclipse/cdt/core/settings/model/ICConfigurationDescription;Lio/sloeber/core/api/BoardDescriptor;)Ljava/lang/String;+269
j  io.sloeber.core.tools.uploaders.arduinoUploader.uploadUsingPreferences(Lorg/eclipse/core/resources/IFile;Lio/sloeber/core/api/BoardDescriptor;Lorg/eclipse/core/runtime/IProgressMonitor;Lorg/eclipse/ui/console/MessageConsoleStream;Lorg/eclipse/ui/console/MessageConsoleStream;Lorg/eclipse/ui/console/MessageConsoleStream;)Z+80
j  io.sloeber.core.tools.uploaders.UploadSketchWrapper$UploadJobWrapper.run(Lorg/eclipse/core/runtime/IProgressMonitor;)Lorg/eclipse/core/runtime/IStatus;+249
j  org.eclipse.core.internal.jobs.Worker.run()V+33
v  ~StubRoutines::call_stub

siginfo: EXCEPTION_ACCESS_VIOLATION (0xc0000005), reading address 0x000000006fdeff79
```

I searched the internet and found multiple identical issues in several projects.

If you want to have specific versions/tests used, just let me know.

jantje, 2020-12-15:
@wimjongman 
Wim have you seen this? The serial monitor disappears on my system as well after trying to open a serial connection

hreintke, 2020-12-15:
With me it is not the serial monitor. I don't use that.

Here it is the upload using serial connection -> cannot upload anything to devices

jantje, 2020-12-15:
Both use the same component that @wimjongman changed in #1261 
So I think there is something very fundamentally wrong that is probably easy to fix for someone who knows what is going on in that change.
Given the changes I'm currently doing in master, and the problems you have with the latest release I accept this is a  nasty situation for you.

wimjongman, 2020-12-15:
I see you are running with Java 10 which is end-of-live since 2018. Did you test on java 11?

wimjongman, 2020-12-15:
Yes, there are two pull requests waiting. 

#1262 switches to a new serial library which comes from CDT.

#1258 which uses a new jssc library. This worked for me but I am using the AdoptOpenJDK 11 which was also tested in the jssc 
Please read issue https://github.com/java-native/jssc/issues/65

hreintke, 2020-12-15:
I tried java 10, 11 and 15. All Oracle java.  
Just to check whether I could solve by using another java version. 
The error logging is from java 10.

jantje, 2020-12-15:
@wimjongman 
Thanks for your input.
I thought these were already merged. 
I guess they don't need to be merged both. Do you advice #1262 or #1258 to be merged?
I  tend to #1258 as setDTR and setRTS are actually used in esp8266 related code 
Can you advise

jantje, 2020-12-15:
@hreintke 
The latest version is internally seriously different from previous versions.
Regression test and build tests run successful . I didn't run the upload tests yet and this issue shows they will fail.
However project upgrades are not yet supported and multi config setups are still buggy.
Most test are currently done automated so I fear that as a user you may come in "difficult situations"
I however am looking forward to testers :-)

wimjongman, 2020-12-16:
#1258 upgrades to a newer jssc version so you can try that first. However. judging from java-native/jssc#65, people have still issues with it.

hreintke, 2020-12-16:
Think I have positive news.

I applied PR 1258 on my local version,
```
C:\Userdata\arduino-eclipse-plugin>\Utilities\SmartGit\git\bin\git fetch origin pull/1258/head:jssc
From https://github.com/Sloeber/arduino-eclipse-plugin
 * [new ref]           refs/pull/1258/head -> jssc
```
Which results in the git : 

[image]

Building that locally gives a sloeber version which does not have the serial issue, although not tested extensive.

I am not sure on which commit the latest nightly is based, expect the commit marked `jm' in the above picture.

This version still has the "make.exe" version issue #1198

wimjongman, 2020-12-16:
Thanks for testing. I have released the patch. If you build locally again, then you should have the latest and the greatest.

> I am not sure on which commit the latest nightly is based,

The nightly is built on the latest commit.

hreintke, 2020-12-16:
Quote from @jantje 

>I switched off nightly creation as master is WIP

[image]

Think it would be good to build a nightly without the WIP and #1258 added.

wimjongman, 2020-12-16:
Master is always WIP. Nightlies are inherently unstable.

If you want to publish "stable nightlies" then maybe we should have a "milestone" repo where we can store a stable nightly whilst still building the nightly.

jantje, 2020-12-16:
>Master is always WIP. Nightlies are inherently unstable.

Indeed but there are currently big works going on. That is why I turned off the nightly. I could have created a branch but due to normal community effort I didn't bother.

>If you want to publish "stable nightlies" then maybe we should have a "milestone" repo where we can store a stable nightly whilst still building the nightly.

IMHO this is lots of work for a problem that hardly ever happens.

jantje, 2020-12-16:
>Think it would be good to build a nightly without the WIP and #1258 added.

Unless you know someone who could do this or do some WIP work,  that would delay the WIP. I only have limited energy and dedication. I focus on WIP right now. I do not expect WIP to stabilize enough to restart the nightly builds this year.

jantje, 2020-12-16:
works on my system as well :-)
Thanks Wim

hreintke, 2020-12-16:
@jantje 
>Unless you know someone who could do this or do some WIP work, that would delay the WIP. I only have limited energy and dedication. I focus on WIP right now. I do not expect WIP to stabilize enough to restart the nightly builds this year.

I think my wording was not the best.
What I ment to say is : 
I think it is good to add one more nightly build, which is the current with #1258 included. 
After that no further nightly until your WIP is finished.
Currently the nightly is broken and cannot be used, adding this makes it a usable version.

>I however am looking forward to testers :-)

As soon as your WIP stabilizes, I am happy to help in testing.

wimjongman, 2020-12-16:
Yes, that is a good idea. Do we know the commit that was used to build the current nightly? 

Everything after that should be the WIP branch. This allows for some bugfixing.

jantje, 2020-12-17:
IMHO The last nightly is from 2020 November 24 started at 05:22 military time so shortly after midnight.
The last build contains pull request #1229 (you can test by creating a new project and on the last page you have no file as option)
[image]

You do not want #1267 from November 24 see comment https://github.com/Sloeber/arduino-eclipse-plugin/issues/1267#issuecomment-731811410 for why

I'll see whether I can create a branch there and fix. No clue how though 😕

jantje, 2020-12-17:
I succeeded in creating the branch https://github.com/Sloeber/arduino-eclipse-plugin/tree/nightly-fix
I can setup a buildserver to process this branch
But cherry picking to get the commit in there ..... please please help

wimjongman, 2020-12-17:
Switch to the nightly-fix branch, then in the history view, find the commit you want, and select cherry-pick:

[image]

jantje, 2020-12-17:
Thanks Wim.
That seems to have worked. I started a nightly build. Lets keep fingers crossed

jantje, 2020-12-17:
For testing I wanted to verify a crash before upgradin to nightly and my production setup does not crash 😕 
Seems it does not crash with oracle java jdk 11.0.4

hreintke, 2020-12-17:
I use Oracle java jdk 11.0.9.

Probably can test with 11.0.4 tomorrow.
Please let me know which sloeber versions you want me to use.

jantje, 2020-12-17:
seems like I'll get the nightly to work with the Wim fix :-) so there is no need to test the old nightly.
I was just very surprised it did not crash with that setup on my machine

jantje, 2020-12-17:
The update site is available but due to plugin interdependencies I fail to update my production setup
Can someone else test?

wimjongman, 2020-12-17:
You also have to add the following eclipse update sites:

http://download.eclipse.org/eclipse/updates/latest
http://download.eclipse.org/releases/latest

jantje, 2020-12-17:
I did not have 
But still no go. He litterally doesn't find anything.
```
An error occurred while collecting items to be installed
session context was:(profile=DefaultProfile, phase=org.eclipse.equinox.internal.p2.engine.phases.Collect, operand=, action=).
No repository found containing: osgi.bundle,com.sun.xml.bind,2.3.3.v20201118-1818
No repository found containing: osgi.bundle,io.sloeber.application,4.3.3.202012171725
No repository found containing: osgi.bundle,io.sloeber.core,4.3.3.202012171725
No repository found containing: osgi.bundle,io.sloeber.core.nl1,4.3.3.202012171725
No repository found containing: osgi.bundle,io.sloeber.ui,4.3.3.202012171725
No repository found containing: osgi.bundle,io.sloeber.ui.nl1,4.3.3.202012171725
No repository found containing: osgi.bundle,org.apache.ws.commons.util,1.0.2.v20160817-1930
No repository found containing: osgi.bundle,org.eclipse.cdt,10.1.0.202010062020
No repository found containing: osgi.bundle,org.eclipse.cdt.autotools.core,2.1.100.202010121414
No repository found containing: osgi.bundle,org.eclipse.cdt.autotools.docs,2.0.600.202010121439
No repository found containing: osgi.bundle,org.eclipse.cdt.autotools.ui,2.1.100.202010121414
No repository found containing: osgi.bundle,org.eclipse.cdt.codan.checkers,3.4.100.202010121414
No repository found containing: osgi.bundle,org.eclipse.cdt.codan.core,4.1.100.202010121414
No repository found containing: osgi.bundle,org.eclipse.cdt.codan.core.cxx,3.5.100.202011201553
No repository found containing: osgi.bundle,org.eclipse.cdt.codan.ui,3.4.100.202011201553
No repository found containing: osgi.bundle,org.eclipse.cdt.core,7.1.100.202012020953
No repository found containing: osgi.bundle,org.eclipse.cdt.core.native,6.1.0.202011181638
No repository found containing: osgi.bundle,org.eclipse.cdt.core.win32,6.0.100.202011111729
No repository found containing: osgi.bundle,org.eclipse.cdt.core.win32.x86_64,6.0.100.202011060124
No repository found containing: osgi.bundle,org.eclipse.cdt.debug.core,8.6.0.202011181638
No repository found containing: osgi.bundle,org.eclipse.cdt.debug.gdbjtag.core,10.1.0.202011291939
No repository found containing: osgi.bundle,org.eclipse.cdt.debug.gdbjtag.ui,9.0.100.202011030619
No repository found containing: osgi.bundle,org.eclipse.cdt.debug.ui,8.4.100.202011181638
No repository found containing: osgi.bundle,org.eclipse.cdt.doc.user,5.4.600.202011181638
No repository found containing: osgi.bundle,org.eclipse.cdt.dsf,2.10.0.202011291939
No repository found containing: osgi.bundle,org.eclipse.cdt.dsf.gdb,6.1.0.202011291941
No repository found containing: osgi.bundle,org.eclipse.cdt.dsf.gdb.ui,2.7.100.202011181638
No repository found containing: osgi.bundle,org.eclipse.cdt.dsf.ui,2.6.100.202011291939
No repository found containing: osgi.bundle,org.eclipse.cdt.launch,10.1.0.202010181249
No repository found containing: osgi.bundle,org.eclipse.cdt.make.core,7.5.100.202011171450
No repository found containing: osgi.bundle,org.eclipse.cdt.make.ui,8.0.100.202011181841
No repository found containing: osgi.bundle,org.eclipse.cdt.managedbuilder.core,9.1.0.202012011707
No repository found containing: osgi.bundle,org.eclipse.cdt.managedbuilder.ui,9.3.0.202012011707
No repository found containing: osgi.bundle,org.eclipse.cdt.native.serial,1.2.100.202011111956
No repository found containing: osgi.bundle,org.eclipse.cdt.platform.branding,10.1.0.202010062020
No repository found containing: osgi.bundle,org.eclipse.cdt.testsrunner,8.2.100.202010121414
No repository found containing: osgi.bundle,org.eclipse.cdt.ui,7.1.0.202011292014
No repository found containing: osgi.bundle,org.eclipse.egit,5.10.0.202012080955-r
No repository found containing: osgi.bundle,org.eclipse.egit.core,5.10.0.202012080955-r
No repository found containing: osgi.bundle,org.eclipse.egit.doc,5.10.0.202012080955-r
No repository found containing: osgi.bundle,org.eclipse.egit.ui,5.10.0.202012080955-r
No repository found containing: osgi.bundle,org.eclipse.epp.mpc.core,1.9.0.v20201105-1601
No repository found containing: osgi.bundle,org.eclipse.epp.mpc.core.win32,1.9.0.v20201105-1601
No repository found containing: osgi.bundle,org.eclipse.epp.mpc.help.ui,1.9.0.v20201105-1601
No repository found containing: osgi.bundle,org.eclipse.epp.mpc.ui,1.9.0.v20201125-0811
No repository found containing: osgi.bundle,org.eclipse.epp.mpc.ui.css,1.9.0.v20201125-0811
No repository found containing: osgi.bundle,org.eclipse.jgit,5.10.0.202012080955-r
No repository found containing: osgi.bundle,org.eclipse.jgit.archive,5.10.0.202012080955-r
No repository found containing: osgi.bundle,org.eclipse.jgit.gpg.bc,5.10.0.202012080955-r
No repository found containing: osgi.bundle,org.eclipse.jgit.http.apache,5.10.0.202012080955-r
No repository found containing: osgi.bundle,org.eclipse.jgit.ssh.apache,5.10.0.202012080955-r
No repository found containing: osgi.bundle,org.eclipse.jgit.ssh.jsch,5.10.0.202012080955-r
No repository found containing: osgi.bundle,org.eclipse.launchbar.core,2.4.100.202010121414
No repository found containing: osgi.bundle,org.eclipse.launchbar.ui,2.4.100.202012041937
No repository found containing: osgi.bundle,org.eclipse.tools.templates.freemarker,1.2.100.202011191931
No repository found containing: osgi.bundle,org.eclipse.tools.templates.ui,1.2.100.202010121414
No repository found containing: org.eclipse.update.feature,io.sloeber.feature,4.3.3.202012171725
[... cut ...]
No repository found containing: org.eclipse.update.feature,org.eclipse.jgit,5.10.0.202012080955-r
No repository found containing: org.eclipse.update.feature,org.eclipse.jgit.gpg.bc,5.10.0.202012080955-r
No repository found containing: org.eclipse.update.feature,org.eclipse.jgit.http.apache,5.10.0.202012080955-r
No repository found containing: org.eclipse.update.feature,org.eclipse.jgit.ssh.apache,5.10.0.202012080955-r
No repository found containing: org.eclipse.update.feature,org.eclipse.jgit.ssh.jsch,5.10.0.202012080955-r
```

jantje, 2020-12-18:
made changes to the update sites in the plugins. Remaking the nightly right now. Hope this will work.
As I assumed this is taking lots of time.

jantje, 2020-12-18:
Still doesn't work.
When I try to upgrade cdt only I also get an error. (see below)
Seems like org.eclipse.cdt.gnu.debug.feature.jar (recently added to cdt) is causing the problem.
As if Sloeber insists on10.0 and CDT on10.1
@hreintke can you update using the nightly?

```
Cannot complete the install because of a conflicting dependency.
  Software being installed: C/C++ GNU Toolchain Debug Support 10.1.0.202010181249 (org.eclipse.cdt.gnu.debug.feature.group 10.1.0.202010181249)
  Software currently installed: Sloeber 4.3.3.202011240522 (io.sloeber.product 4.3.3.202011240522)
  Only one of the following can be installed at once: 
    C/C++ GNU Toolchain Debug Support 10.0.0.202009010132 (org.eclipse.cdt.gnu.debug.feature.jar 10.0.0.202009010132)
    C/C++ GNU Toolchain Debug Support 10.1.0.202010181249 (org.eclipse.cdt.gnu.debug.feature.jar 10.1.0.202010181249)
  Cannot satisfy dependency:
    From: Sloeber 4.3.3.202011240522 (io.sloeber.product 4.3.3.202011240522)
    To: org.eclipse.equinox.p2.iu; org.eclipse.cdt.gnu.debug.feature.group [10.0.0.202009010132,10.0.0.202009010132]
  Cannot satisfy dependency:
    From: C/C++ GNU Toolchain Debug Support 10.0.0.202009010132 (org.eclipse.cdt.gnu.debug.feature.group 10.0.0.202009010132)
    To: org.eclipse.equinox.p2.iu; org.eclipse.cdt.gnu.debug.feature.jar [10.0.0.202009010132,10.0.0.202009010132]
  Cannot satisfy dependency:
    From: C/C++ GNU Toolchain Debug Support 10.1.0.202010181249 (org.eclipse.cdt.gnu.debug.feature.group 10.1.0.202010181249)
    To: org.eclipse.equinox.p2.iu; org.eclipse.cdt.gnu.debug.feature.jar [10.1.0.202010181249,10.1.0.202010181249]
```

erdemuncuoglu, 2020-12-18:
> I however am looking forward to testers :-)

My latest Sloeber platform is...
```
OS      : Linux 4.19.0-0.bpo.9-amd64
Eclipse : 4.18.0.I20201202-1800
JAVA    : OpenJDK  11.0.6
CDT     : 10.1.0.202012020953
Sloeber : 4.3.3.202012180113 (plugin version)
```

Testing with ESP8266...
```
Platform : 2.7.4
Build    : ok
Upload   : ok
Serial   : ok
```

Testing with ESP32...
```
Platform : 1.0.4
Build    : ok
Upload   : ok
Serial   : ok
```

I wish this counts as a valid test

jantje, 2020-12-18:
@erdemuncuoglu Did you upgrade using the nightly update site?

jantje, 2020-12-19:
Upgrade seems to work from stable to nightly.
Well you do need to replace java8 by java 11 but upgrade finds all needed stuff.
So I'm closing this one as fixed in nightly.

jantje, 2020-12-19:
Thanks all for the help :-)

closed 2020-12-19 by jantje

hreintke, 2020-12-19:
@jantje Final question on this.

I changed the java version in use by removing the old and installing the new java version system wide.
When I have multiple versions of java installed, is there a way to direct sloeber to use a specific one ?

wimjongman, 2020-12-19:
You change that in sloeber.ini. Google for eclipse.ini

jantje, 2020-12-19:
> 
> You change that in sloeber.ini. Google for eclipse.ini

Indeed. I have in some .ini files
```
-vm
C:/Program Files/Java/jdk-11.0.4/bin
```

erdemuncuoglu, 2020-12-20:
> Did you upgrade using the nightly update site?

Yes it was. And by the time I read your comment, I got an other update to  `4.3.3.202012182318` and I did the same tests again. Results were all ok.

jantje, 2020-12-20:
> 
> 
> > Did you upgrade using the nightly update site?
> 
> Yes it was. And by the time I read your comment, I got an other update to `4.3.3.202012182318` and I did the same tests again. Results were all ok.

Thanks