Sloeber/arduino-eclipse-plugin #1280: Devstyle Darkest Dark plugin removing all UI and build tools from Sloeber.

MangyDogUK, 2020-12-26:
As the topic says. Install a fresh download of sloeber from the website, install devstyle darkest dark off market place and while devstyle installs fine, the UI looses all build tools and controls relating to sloeber or arduino. You can select Arduino perspective, but thats it. Nothing actually loads. 

I cant be the only one with this issue?

jantje, 2020-12-26:
I guess devstyle requires (installed) a cdt version 10 or higher or java version 11 or higher.
Both resulting in sloeber not being loaded as it requires java8 and CDT version <10
This is simply the eclipse plugin version nightmare that caused us to create a single package to install.

wimjongman, 2020-12-26:
I don't think devstyle requires cdt. I also don't think Sloeber requires java 8.

@MangyDogUK can you attach the .log from the workspace .metadata directory?

rmspacefish, 2021-02-02:
I have the same issue. I attached the log file.

rmspacefish, 2021-02-02:
I also had to delete the old Sloeber folder and replace it to get the icons back.

jantje, 2021-02-02:
@rmspacefish based on your .log it looks to me your sloeber install updated the eclipse version. Which has the same root cause as #1263 
Basically sloeber 3.3 needs CDT<=9 needs old eclipse needs java 8
Sloeber nightly needs CDT>=10 needs new eclipse needs java 11

We are currently in the final stages of releasing Sloeber 4.4 who has some serious internal changes; so it is going to be a big (technical) jump from 3.3 to 3.4.

Therefore  we need testers https://github.com/Sloeber/arduino-eclipse-plugin/discussions/1290
Please subscribe for testing.

uzi18, 2021-02-02:
My vote for dark mode 👍

jantje, 2021-02-02:
> 
> 
> My vote for dark mode 👍

I'm confused. How do you relate this comment to the issue?

uzi18, 2021-02-02:
Just a joke, also had some problems with this mode but after some magic it is working on 3.3.
Always Backup Sloeber folder before update anything ;)

jantje, 2021-02-02:
> Always Backup Sloeber folder before update anything ;)

Happy to see you follow my advice ;-) http://eclipse.baeyens.it/how_to.shtml#/n

rmspacefish, 2021-02-15:
I tested installing the dark style plugin on Windows and Linux, and it worked without issues.

jantje, 2021-02-15:
>The default color theme on Linux for a dark Ubuntu settings wasn't really ideal either (awful contrasts making some monitor/views unreadable), but I guess that is an Eclipse specific problem.

Yes indeed Sloeber does not come with color themes. It is one of the benefits/disadvantages of being in a bigger package ;-)

uzi18, 2021-02-15:
Confirmed, it works without issues :)

jantje, 2021-02-15:
thanks for confirming

closed 2021-02-15 by jantje