Sloeber/arduino-eclipse-plugin #1284: esp32 macOS Big Sur error

asat232, 2021-01-06:
**Describe the bug**
Plugin was working fine for ESP32 Dev Module on previous macOS Catalina, but after updating to macOS Big Sur on Verify sketch having error described in "Additional context". Temporary workaround i'm using now for existing projects described in "Additional context" as well.

**To Reproduce**
Steps to reproduce the behavior:
Attempt to Verify existing sketch for ESP32 Dev Module.

**Expected behavior**
Program verifying without issues.

**Desktop (please complete the following information):**
 - macOS Big Sur Version 11.1 (20C69)
 - Eclipse Version: 2020-09 (4.17.0) Build id: 20200910-1200
 - Sloeber (plugin version)	4.3.3.202005042010

**Additional context**
Error on verifying sketch:

make all 
Do all objcopy commands
python "/Applications/Eclipse.app/Contents/Eclipse//arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/gen_esp32part.py" -q "/Applications/Eclipse.app/Contents/Eclipse//arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/partitions/default.csv" "/Users/a1/projects_test/arduino_test/esp_arduino_serial/esp_serial/Release/esp_serial.partitions.bin"
"/Applications/Eclipse.app/Contents/Eclipse//arduinoPlugin/packages/esp32/tools/esptool_py/2.6.1/esptool" --chip esp32 elf2image --flash_mode "dio" --flash_freq "80m" --flash_size "4MB" -o "/Users/a1/projects_test/arduino_test/esp_arduino_serial/esp_serial/Release/esp_serial.bin" "/Users/a1/projects_test/arduino_test/esp_arduino_serial/esp_serial/Release/esp_serial.elf"
Traceback (most recent call last):
  File "esptool.py", line 57, in <module>
  File "/Library/Python/2.7/site-packages/PyInstaller/loader/pyimod03_importers.py", line 389, in load_module
  File "serial/tools/list_ports.py", line 29, in <module>
  File "/Library/Python/2.7/site-packages/PyInstaller/loader/pyimod03_importers.py", line 389, in load_module
  File "serial/tools/list_ports_posix.py", line 31, in <module>
  File "/Library/Python/2.7/site-packages/PyInstaller/loader/pyimod03_importers.py", line 389, in load_module
  File "serial/tools/list_ports_osx.py", line 32, in <module>
ValueError: dlsym(RTLD_DEFAULT, kIOMasterPortDefault): symbol not found
Failed to execute script esptool
make: *** [esp_serial.hex] Error 255

Temporary workaround for existing projects:
Notes: paths can differ from mine, on my mac python2.7 having installed package pyserial (if not then should be installed, for example by using pip)

copy file:
/Applications/Eclipse.app/Contents/Eclipse/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/esptool.py
to directory:
/Applications/Eclipse.app/Contents/Eclipse/arduinoPlugin/packages/esp32/tools/esptool_py/2.6.1/

edit file:
/Applications/Eclipse.app/Contents/Eclipse/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/platform.txt
line to modify:
tools.esptool_py.cmd=esptool
line modified:
tools.esptool_py.cmd=esptool.py

edit file inside project directory:
.settings/org.eclipse.cdt.core.prefs
line to modify:
environment/project/io.sloeber.core.toolChain.release.9935190/A.TOOLS.ESPTOOL_PY.CMD/value=esptool
line modified:
environment/project/io.sloeber.core.toolChain.release.9935190/A.TOOLS.ESPTOOL_PY.CMD/value=esptool.py

jantje, 2021-01-07:
Looking at the workaround you propose this comes to me as a esp32 issue and not a Sloeber issue.
Can you elaborate why you think this is a Sloeber issue?

closed 2021-01-07 by asat232

jantje, 2021-01-07:
Can you elaborate on why you closed the issue?