Sloeber/arduino-eclipse-plugin #1285: Neither plugin nor bundle working

wf2020, 2021-01-14:
Installation fails because files for downloading don't exist at arduino.cc:

`!ENTRY io.sloeber.core 2 0 2021-01-14 14:44:05.178
!MESSAGE Failed to download url http://downloads.arduino.cc/cores/ArduinoCore-mbed-nonexistant.tar.bz2 error code is: 404

!ENTRY io.sloeber.core 2 0 2021-01-14 14:44:05.178
!MESSAGE Failed to download url http://downloads.arduino.cc/cores/ArduinoCore-mbed-nonexistant.tar.bz2
!STACK 0
java.io.IOException: Failed to download url http://downloads.arduino.cc/cores/ArduinoCore-mbed-nonexistant.tar.bz2 error code is: 404
	at io.sloeber.core.api.PackageManager.myCopy(PackageManager.java:778)
	at io.sloeber.core.api.PackageManager.myCopy(PackageManager.java:740)
	at io.sloeber.core.managers.InternalPackageManager.downloadAndInstall(InternalPackageManager.java:351)
	at io.sloeber.core.managers.InternalPackageManager.downloadAndInstall(InternalPackageManager.java:130)
	at io.sloeber.core.managers.InternalPackageManager.startup_Pluging(InternalPackageManager.java:101)
	at io.sloeber.core.Activator$2.run(Activator.java:247)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:63)

!ENTRY io.sloeber.core 4 0 2021-01-14 14:44:05.209
!MESSAGE Failed to configer Sloeber
!SUBENTRY 1 io.sloeber.core 4 0 2021-01-14 14:44:05.209
!MESSAGE Failed to install [DEPRECATED] Arduino mbed-enabled Boards
!SUBENTRY 2 io.sloeber.core 4 0 2021-01-14 14:44:05.209
!MESSAGE Failed to download "http://downloads.arduino.cc/cores/ArduinoCore-mbed-nonexistant.tar.bz2".
!STACK 0
java.io.IOException: Failed to download url http://downloads.arduino.cc/cores/ArduinoCore-mbed-nonexistant.tar.bz2 error code is: 404
	at io.sloeber.core.api.PackageManager.myCopy(PackageManager.java:778)
	at io.sloeber.core.api.PackageManager.myCopy(PackageManager.java:740)
	at io.sloeber.core.managers.InternalPackageManager.downloadAndInstall(InternalPackageManager.java:351)
	at io.sloeber.core.managers.InternalPackageManager.downloadAndInstall(InternalPackageManager.java:130)
	at io.sloeber.core.managers.InternalPackageManager.startup_Pluging(InternalPackageManager.java:101)
	at io.sloeber.core.Activator$2.run(Activator.java:247)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:63)
`

jantje, 2021-01-14:
duplicate of #1264

closed 2021-01-14 by jantje

wf2020, 2021-01-14:
Thought it sholuld be fixed after all the time...
Not really helpful that installation just fails without any hint!

jantje, 2021-01-14:
The hint is "!MESSAGE Failed to download "http:downloads.**arduino**.cc/cores/ArduinoCore-mbed-**nonexistant**.tar.bz2"."
I say: "Not really helpful of arduino to publish platforms that that need to install nonexistant files"