Sloeber/arduino-eclipse-plugin #1287: Build failed for sduino STM8 board

pbtsrc, 2021-01-28:
Boards Manager URL (used both in Sloeber and Arduino IDE):
https://github.com/tenbaht/sduino/raw/master/package_sduino_stm8_index.json
STM8S103F3 Breakout Board

Tested: standard example Blink.ino
In the Arduino IDE the build finished without errors.

In Sloeber the build failed:

`Starting archiver
"/opt/eclipse//arduinoPlugin/packages/sduino/tools/STM8Tools/2019.02.05/wrapper/sdar.sh" "/opt/eclipse//arduinoPlugin/packages/sduino/tools/sdcc/build.11242/bin/sdar"    /home/user/Projects/test_stm8_blink/Release/arduino.ar  ./core/core/HardwareSerial.c-1.c.o ./core/core/HardwareSerial.c-10.c.o ./core/core/HardwareSerial.c-2.c.o ./core/core/HardwareSerial.c-3.c.o ./core/core/HardwareSerial.c-4.c.o ./core/core/HardwareSerial.c-5.c.o ./core/core/HardwareSerial.c-6.c.o ./core/core/HardwareSerial.c-7.c.o ./core/core/HardwareSerial.c-8.c.o ./core/core/HardwareSerial.c-9.c.o ./core/core/Print-float.c.o ./core/core/Print.c.o ./core/core/SPI.c.o ./core/core/WInterrupts.c.o ./core/core/WMath.c.o ./core/core/main.c.o ./core/core/weak_atexit.c.o ./core/core/weak_initVariant.c.o ./core/core/weak_serialEvent.c.o ./core/core/wiring-alternate-function.c.o ./core/core/wiring-delay-microseconds.c.o ./core/core/wiring-delay.c.o ./core/core/wiring-init.c.o ./core/core/wiring-micros.c.o ./core/core/wiring-millis.c.o ./core/core/wiring_analog.c.o ./core/core/wiring_digital-tables.c.o ./core/core/wiring_digital.c-1.c.o ./core/core/wiring_digital.c-2.c.o ./core/core/wiring_digital.c-3.c.o ./core/core/wiring_digital.c-4.c.o ./core/core/wiring_digital.c-5.c.o ./core/core/wiring_digital.c-6.c.o ./core/core/wiring_digital.c-7.c.o ./core/core/wiring_pulse.c.o ./core/core/wiring_shift.c.o     re4  rcs
/opt/eclipse//arduinoPlugin/packages/sduino/tools/sdcc/build.11242/bin/sdar /home/user/Projects/test_stm8_blink/Release/arduino.ar ./core/core/HardwareSerial.c-1.c.o ./core/core/HardwareSerial.c-10.c.o ./core/core/HardwareSerial.c-2.c.o ./core/core/HardwareSerial.c-3.c.o ./core/core/HardwareSerial.c-4.c.o ./core/core/HardwareSerial.c-5.c.o ./core/core/HardwareSerial.c-6.c.o ./core/core/HardwareSerial.c-7.c.o ./core/core/HardwareSerial.c-8.c.o ./core/core/HardwareSerial.c-9.c.o ./core/core/Print-float.c.o ./core/core/Print.c.o ./core/core/SPI.c.o ./core/core/WInterrupts.c.o ./core/core/WMath.c.o ./core/core/main.c.o ./core/core/weak_atexit.c.o ./core/core/weak_initVariant.c.o ./core/core/weak_serialEvent.c.o ./core/core/wiring-alternate-function.c.o ./core/core/wiring-delay-microseconds.c.o ./core/core/wiring-delay.c.o ./core/core/wiring-init.c.o ./core/core/wiring-micros.c.o ./core/core/wiring-millis.c.o ./core/core/wiring_analog.c.o ./core/core/wiring_digital-tables.c.o ./core/core/wiring_digital.c-1.c.o ./core/core/wiring_digital.c-2.c.o ./core/core/wiring_digital.c-3.c.o ./core/core/wiring_digital.c-4.c.o ./core/core/wiring_digital.c-5.c.o ./core/core/wiring_digital.c-6.c.o ./core/core/wiring_digital.c-7.c.o ./core/core/wiring_pulse.c.o ./core/core/wiring_shift.c.o re4 rcs
Mark ./core/core/HardwareSerial.c-10.c.o:/opt/eclipse//arduinoPlugin/packages/sduino/tools/sdcc/build.11242/bin/sdar ./core/core/HardwareSerial.c-2.c.o ./core/core/HardwareSerial.c-3.c.o ./core/core/HardwareSerial.c-4.c.o ./core/core/HardwareSerial.c-5.c.o ./core/core/HardwareSerial.c-6.c.o ./core/core/HardwareSerial.c-7.c.o ./core/core/HardwareSerial.c-8.c.o ./core/core/HardwareSerial.c-9.c.o ./core/core/Print-float.c.o ./core/core/Print.c.o ./core/core/SPI.c.o ./core/core/WInterrupts.c.o ./core/core/WMath.c.o ./core/core/main.c.o ./core/core/weak_atexit.c.o ./core/core/weak_initVariant.c.o ./core/core/weak_serialEvent.c.o ./core/core/wiring-alternate-function.c.o ./core/core/wiring-delay-microseconds.c.o ./core/core/wiring-delay.c.o ./core/core/wiring-init.c.o ./core/core/wiring-micros.c.o ./core/core/wiring-millis.c.o ./core/core/wiring_analog.c.o ./core/core/wiring_digital-tables.c.o ./core/core/wiring_digital.c-1.c.o ./core/core/wiring_digital.c-2.c.o ./core/core/wiring_digital.c-3.c.o ./core/core/wiring_digital.c-4.c.o ./core/core/wiring_digital.c-5.c.o ./core/core/wiring_digital.c-6.c.o ./core/core/wiring_digital.c-7.c.o ./core/core/wiring_pulse.c.o ./core/core/wiring_shift.c.o re4 rcs /home/user/Projects/test_stm8_blink/Release/arduino.ar ./core/core/HardwareSerial.c-1.c.rel
/opt/eclipse//arduinoPlugin/packages/sduino/tools/sdcc/build.11242/bin/sdar: wrong key — «.»`

The corresponding Arduino IDE output:

`/home/user/.arduino15/packages/sduino/tools/STM8Tools/2019.02.05/wrapper/sdar.sh /home/user/.arduino15/packages/sduino/tools/sdcc/build.11242/bin/sdar /tmp/arduino_build_459865/core/core.a /tmp/arduino_build_459865/core/HardwareSerial.c-1.c.o re4 rcs
/home/user/.arduino15/packages/sduino/tools/sdcc/build.11242/bin/sdar /tmp/arduino_build_459865/core/core.a /tmp/arduino_build_459865/core/HardwareSerial.c-1.c.o re4 rcs
Mark re4:/home/user/.arduino15/packages/sduino/tools/sdcc/build.11242/bin/sdar rcs /tmp/arduino_build_459865/core/core.a /tmp/arduino_build_459865/core/HardwareSerial.c-1.c.rel
/home/user/.arduino15/packages/sduino/tools/STM8Tools/2019.02.05/wrapper/sdar.sh /home/user/.arduino15/packages/sduino/tools/sdcc/build.11242/bin/sdar /tmp/arduino_build_459865/core/core.a /tmp/arduino_build_459865/core/HardwareSerial.c-10.c.o re4 rcs
/home/user/.arduino15/packages/sduino/tools/sdcc/build.11242/bin/sdar /tmp/arduino_build_459865/core/core.a /tmp/arduino_build_459865/core/HardwareSerial.c-10.c.o re4 rcs
Mark re4:/home/user/.arduino15/packages/sduino/tools/sdcc/build.11242/bin/sdar rcs /tmp/arduino_build_459865/core/core.a /tmp/arduino_build_459865/core/HardwareSerial.c-10.c.rel`

It seems the main difference is that the Sloeber runs bin/sdar with relative paths (and so the error is wrong key — «.»), and the Arduino IDE uses absolute paths.

jantje, 2021-01-29:
I see that the sloeber build has way more HardwareSerial.c files than the arduino IDE builds.
Are you sure your setup is ok?

pbtsrc, 2021-01-29:
I compared files in 
/opt/eclipse/arduinoPlugin/packages/sduino/hardware/stm8/0.5/
and
/home/user/.arduino15/packages/sduino/hardware/stm8/0.5.0/
The directories are identical (including the files inside them)

jantje, 2021-01-29:
I see that I excluded this package from testing with the comment "uses busybox so command line issues I think"
https://github.com/Sloeber/arduino-eclipse-plugin/blob/75737d2fa454d3d2d67e6e811857caf611e2ae15/io.sloeber.tests/src/io/sloeber/core/CreateAndCompileDefaultInoOnAllBoardsTest.java#L70
The package is removed for all oses which probably means the boards failed a test on windows and linux.

To get this to work there is a need for someone who knows the toolchain and can explain what causes errors (like: /sduino/tools/sdcc/build.11242/bin/sdar: wrong key — «.») and how to fix the command lines.

Note that in regards of build process the behaviour between Sloeber and Arduino IDE are different in regards to the archiver. Arduino IDE builds the archive "one by one" and Sloeber builds it in one go.
There have been issues due to this different behaviour.

Also note that my original thought was :""this is the windows command line to long issue" but as you seem to be on linux ...

Also note that I turned off the nightly because of disruptive changes so anyone involved in this fix (at this moment in time) should be willing to work from sloeber source code.

pbtsrc, 2021-07-13:
Tested on Sloeber-4.4. The issue is still there, unfortunately. The build log is the same too

jantje, 2021-07-13:
I'll try this on linux.
If you havn't heard of me in a week, remind me.

jantje, 2021-07-17:
I tried this on Linux and I get the same result.
When I run the command from the Terminal I get a very similar error
I tried to simplify the command and make it all full path commands but still a similar error
`/home/jan/beta/Sloeber/arduinoPlugin/packages/sduino/tools/sdcc/build.11242/bin/sdar /home/jan/beta/version/ffff/Release/arduino.ar ~/beta/version/ffff/Release/core/core/wiring_shift.c.o re4 rcs`

```
/home/jan/beta/Sloeber/arduinoPlugin/packages/sduino/tools/sdcc/build.11242/bin/sdar: invalid option -- '/'
Usage: /home/jan/beta/Sloeber/arduinoPlugin/packages/sduino/tools/sdcc/build.11242/bin/sdar [emulation options] [-]{dmpqrstx}[abcDfilMNoPsSTuvV] [--plugin <name>] [member-name] [count] archive-file file...
       /home/jan/beta/Sloeber/arduinoPlugin/packages/sduino/tools/sdcc/build.11242/bin/sdar -M [<mri-script]
 commands:
  d            - delete file(s) from the archive
  m[ab]        - move file(s) in the archive
  p            - print file(s) found in the archive
  q[f]         - quick append file(s) to the archive
  r[ab][f][u]  - replace existing or insert new file(s) into the archive
  s            - act as ranlib
  t            - display contents of archive
  x[o]         - extract file(s) from the archive
 command specific modifiers:
  [a]          - put file(s) after [member-name]
  [b]          - put file(s) before [member-name] (same as [i])
  [D]          - use zero for timestamps and uids/gids
  [U]          - use actual timestamps and uids/gids (default)
  [N]          - use instance [count] of name
  [f]          - truncate inserted file names
  [P]          - use full path names when matching
  [o]          - preserve original dates
  [u]          - only replace files that are newer than current archive contents
 generic modifiers:
  [c]          - do not warn if the library had to be created
  [s]          - create an archive index (cf. ranlib)
  [S]          - do not build a symbol table
  [T]          - make a thin archive
  [v]          - be verbose
  [V]          - display the version number
  @<file>      - read options from <file>
  --target=BFDNAME - specify the target object format as BFDNAME
 optional:
  --plugin <p> - load the specified plugin
 emulation options: 
  No emulation specific options
/home/jan/beta/Sloeber/arduinoPlugin/packages/sduino/tools/sdcc/build.11242/bin/sdar: supported targets: asxxxx plugin srec symbolsrec verilog tekhex binary ihex
jan@jan-ubuntu:~/beta/version/ffff/Release$
```

I can't help any further as this is a cryptic error from a tool called by a wrapper.
If you want to progress this issue you will have to contact someone with more knowledge of the tools used for this board.