Sloeber/arduino-eclipse-plugin #1292: When will this be fixed?

cahhoffman, 2021-02-01:
**Describe the bug**
Any idea where to find the boards file for Arduino nRF52. On both mac and windows get same following error. Failed to install [DEPRECATED] Arduino mbed-enabled Boards Failed to download "http://downloads.arduino.cc/cores/ArduinoCore-mbed-nonexistant.tar.bz2".

wm1962, 2021-02-03:
I get the same error. Is there any solution?

FranzDurst, 2021-02-03:
Same here. Wanted to try Sloeber for the first time and neither recommended IDE nor Eclipse plugin works. At finishing Installation phase the mentioned ([DEPRECATED] ....) massage arises. Later it's not possible to create a new sketch due to "no boards.txt found". 
Tried on a different, "clean" computer --> same problem.

jandrassy, 2021-02-03:
Arduino removed version 1.2 of mbed boards. 
install 1.3

FranzDurst, 2021-02-03:
jandrassy, thanks a lot!
Found out that this is a known issue (with Arduino, not directly Soeber). [Issue](https://github.com/Sloeber/arduino-eclipse-plugin/issues/1264)

So just a minor glitch but can confuse you a lot if new to Soeber.

wm1962, 2021-02-03:
Thanks a lot

closed 2021-03-22 by jantje