Sloeber/arduino-eclipse-plugin #1293: Upload sketch does not upload a sketch

Roycinger, 2021-02-03:
Hi all together,
I just transitioned from Arduino IDE to Sloeber.
I went through the standalone installation process and the How to section on the page.
After that, I pasted a simple blink sketch into Sloeber and pressed the upload sketch button.
The build seems to be successful. However, after an apparently successful flash (no error messages), the Arduino is still running the old sketch that was previously flashed with Arduino IDE.
Here's a screenshot output of the console:
[image]
This is the complete line:
Launching: D:\Sloeber\arduinoPlugin\packages\arduino\tools\avrdude\6.3.0-arduino17/bin/avrdude -CD:\Sloeber\arduinoPlugin\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf -patmega328p -cstk500v2 -Pusb -Uflash:w:D:\Sloeber_workspace\Surric_x\Release/Surric_x.hex:i
These are my board settings (pretty much the same as in Arduino IDE:
[image]

Am I missing something?
Thanks for your help in advance

jantje, 2021-02-03:
Default is the default upload protocol. Try that.

Roycinger, 2021-02-03:
That helped, thank you!
Out of curiosity: Why doesn't it work with a specific programmer selected?
In case I want to program an ATTiny85 later, I'd have to use Arduino as ISP. Would be a bit uncool if this wouldn't work :)

jantje, 2021-02-03:
If you want to use a programmer you need to use the selected programmer otherwise use default

closed 2021-02-04 by Roycinger

uzi18, 2021-02-04:
@Roycinger do you want to say: when you choose board "default" programmer was not selected by default ;) ?

Roycinger, 2021-02-04:
> @Roycinger do you want to say: when you choose board "default" programmer was not selected by default ;) ?

Well, I set the board up like I did in Arduino, with AVR ISP MKII as programmer / upload protocol.
In the Arduino IDE, there is no "default" option.

jantje, 2021-02-04:
>Well, I set the board up like I did in Arduino, with AVR ISP MKII as programmer / upload protocol.
In the Arduino IDE, there is no "default" option.

To me the Arduino ide is very confusing in regards what happens in regards to upload and programmers.
Basically most people will never use a programmer though the programmers are really visible (which is in contrast to normal arduino IDE behaviour which is hide things that are hardly used to simplify).

[image]
Given the menu of arduino IDE above I would compare Arduino IDE behaviour to Sloeber behaviour like
upload = default programmer
upload using programmer= select a programmer

Roycinger, 2021-02-04:
Ah, then I was just as confused as you are with my statement :)
I was always referring to this menu:
[image]

So I guess this menu from my screenshot refers to actual physical programmers (that should be used when choosing "Upload using programmer".
If this is the case, I wonder why an upload with Arduino as ISP always worked for me when hitting the usual upload button

jantje, 2021-02-04:
> So I guess this menu from my screenshot refers to actual physical programmers (that should be used when choosing "Upload using programmer".

yes

>If this is the case, I wonder why an upload with Arduino as ISP always worked for me when hitting the usual upload button

because upload is upload and not upload using programmer.

It took me a long time to realize the selected programmer is only used when you use "upload using programmer" In all other cases it is simply ignored.