Sloeber/arduino-eclipse-plugin #1294: Sloeber consumes lot of CPU if Arduino is disconnected and Serial monitor left open

FranzDurst, 2021-02-04:
Hi,
Sloeber consumes lot of CPU if Arduino is disconnected and Serial monitor left open.
At the very moment the Monitor is closed CPU settles down.
Not a big issue of course.

Arduino Nano, latest Sloeber IDE (Feb/2021), Win 10

Btw, I'm using Sloeber since two days and love it! 
Used Eclipse with some plug ins and occasionally this caused quite a lot of frustration (libraries not recognised in a random way, minimum 1h to find a way and next start the same ... or not)
Greetings, Franz

jantje, 2021-02-04:
In 4.4 the jdms has been upgraded and that is likely the problem.
Beta testing for 4.4 is planned to start this month
https://github.com/Sloeber/arduino-eclipse-plugin/discussions/1290

tyllurius, 2021-03-22:
I can confirm this error with Beta 4.4 on Win 10.