Sloeber/arduino-eclipse-plugin #1295: Build dependency issue with ESP32 and debug level

HiroP0, 2021-02-05:
**Description**
espressif ESP32 (Board: Dev Module or M5Stack)  provides logging by macros reacting to the "Core Debug Level" board setting. 
(https://dl.espressif.com/dl/package_esp32_index.json, v1.0.7)

Once a log level is set FreeRTOS library gets included, but the builds fail due to an unresolved dependency to <avr/io.h>. Setting log level back to none and detaching FreeRTOS library enables successful builds again. Arduino IDE builds successfully. (Issue persits with Sloeber independently from whether Arduino IDE is installed or not).
So ESP32 default logging does not work.

**To Reproduce**

Source ino:

> #include "Arduino.h"
> void setup() {
> 	log_e("Baaad.");
> }
> void loop() {}

1. Create ESP32 project, "Core Debug Level" None
2. Use full source ino from above
3. Build --> successful
4. Set Project>Preferences>Ardunio: "Core Debug Level": "Error"
5. Build -->FreeRTOS gets included
6. See error:

18:15:37 **** Incremental Build of configuration Release for project log_test ****
"E:\\dev\\eclipse\\arduinoPlugin\\tools\\make\\make" all 
'Building file: ..\sloeber.ino.cpp'
'Starting C++ compile'
"E:\dev\eclipse\arduinoPlugin\packages\m5stack\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX "-IE:\dev\eclipse\/arduinoPlugin/packages/m5stack/hardware/esp32/1.0.7/tools/sdk/include/config" "-IE:\dev\eclipse\/arduinoPlugin/packages/m5stack/hardware/esp32/1.0.7/tools/sdk/include/app_trace" "-IE:\dev\eclipse\/arduinoPlugin/packages/m5stack/hardware/esp32/1.0.7/tools/sdk/include/app_update" "-IE:\dev\eclipse\/arduinoPlugin/packages/m5stack/hardware/esp32/1.0.7/tools/sdk/include/asio" "-IE:\dev\eclipse\/arduinoPlugin/packages/m5stack/hardware/esp32/1.0.7/tools/sdk/include/bootloader_support" "-IE:\dev\eclipse\/arduinoPlugin/packages/m5stack/hardware/esp32/1.0.7/tools/sdk/include/bt" "-IE:\dev\eclipse\/arduinoPlugin/packages/m5stack/hardware/esp32/1.0.7/tools/sdk/include/coap" "-IE:\dev\eclipse\/arduinoPlugin/packages/m5stack/hardware/esp32/1.0.7/tools/sdk/include/console" "-IE:\dev\eclipse\/arduinoPlugin/packages/m5stack/hardware/esp32/1.0.7/tools/sdk/include/driver" "-IE:\dev\eclipse\/arduinoPlugin/packages/m5stack/hardware/esp32/1.0.7/tools/sdk/include/esp-tls" "-IE:\dev\eclipse\/arduinoPlugin/packages/m5stack/hardware/esp32/1.0.7/tools/sdk/include/esp32" "-IE:\dev\eclipse\/arduinoPlugin/packages/m5stack/hardware/esp32/1.0.7/tools/sdk/include/esp_adc_cal" "-IE:\dev\eclipse\/arduinoPlugin/packages/m5stack/hardware/esp32/1.0.7/tools/sdk/include/esp_event" "-IE:\dev\eclipse\/arduinoPlugin/packages/m5stack/hardware/esp32/1.0.7/tools/sdk/include/esp_http_client" "-IE:\dev\eclipse\/arduinoPlugin/packages/m5stack/hardware/esp32/1.0.7/tools/sdk/include/esp_http_server" "-IE:\dev\eclipse\/arduinoPlugin/packages/m5stack/hardware/esp32/1.0.7/tools/sdk/include/esp_https_ota" "-IE:\dev\eclipse\/arduinoPlugin/packages/m5stack/hardware/esp32/1.0.7/tools/sdk/include/esp_ringbuf" "-IE:\dev\eclipse\/arduinoPlugin/packages/m5stack/hardware/esp32/1.0.7/tools/sdk/include/ethernet" "-IE:\dev\eclipse\/arduinoPlugin/packages/m5stack/hardware/esp32/1.0.7/tools/sdk/include/expat" "-IE:\dev\eclipse\/arduinoPlugin/packages/m5stack/hardware/esp32/1.0.7/tools/sdk/include/fatfs" "-IE:\dev\eclipse\/arduinoPlugin/packages/m5stack/hardware/esp32/1.0.7/tools/sdk/include/freemodbus" "-IE:\dev\eclipse\/arduinoPlugin/packages/m5stack/hardware/esp32/1.0.7/tools/sdk/include/freertos" "-IE:\dev\eclipse\/arduinoPlugin/packages/m5stack/hardware/esp32/1.0.7/tools/sdk/include/heap" "-IE:\dev\eclipse\/arduinoPlugin/packages/m5stack/hardware/esp32/1.0.7/tools/sdk/include/idf_test" "-IE:\dev\eclipse\/arduinoPlugin/packages/m5stack/hardware/esp32/1.0.7/tools/sdk/include/jsmn" "-IE:\dev\eclipse\/arduinoPlugin/packages/m5stack/hardware/esp32/1.0.7/tools/sdk/include/json" "-IE:\dev\eclipse\/arduinoPlugin/packages/m5stack/hardware/esp32/1.0.7/tools/sdk/include/libsodium" "-IE:\dev\eclipse\/arduinoPlugin/packages/m5stack/hardware/esp32/1.0.7/tools/sdk/include/log" "-IE:\dev\eclipse\/arduinoPlugin/packages/m5stack/hardware/esp32/1.0.7/tools/sdk/include/lwip" "-IE:\dev\eclipse\/arduinoPlugin/packages/m5stack/hardware/esp32/1.0.7/tools/sdk/include/mbedtls" "-IE:\dev\eclipse\/arduinoPlugin/packages/m5stack/hardware/esp32/1.0.7/tools/sdk/include/mdns" "-IE:\dev\eclipse\/arduinoPlugin/packages/m5stack/hardware/esp32/1.0.7/tools/sdk/include/micro-ecc" "-IE:\dev\eclipse\/arduinoPlugin/packages/m5stack/hardware/esp32/1.0.7/tools/sdk/include/mqtt" "-IE:\dev\eclipse\/arduinoPlugin/packages/m5stack/hardware/esp32/1.0.7/tools/sdk/include/newlib" "-IE:\dev\eclipse\/arduinoPlugin/packages/m5stack/hardware/esp32/1.0.7/tools/sdk/include/nghttp" "-IE:\dev\eclipse\/arduinoPlugin/packages/m5stack/hardware/esp32/1.0.7/tools/sdk/include/nvs_flash" "-IE:\dev\eclipse\/arduinoPlugin/packages/m5stack/hardware/esp32/1.0.7/tools/sdk/include/openssl" "-IE:\dev\eclipse\/arduinoPlugin/packages/m5stack/hardware/esp32/1.0.7/tools/sdk/include/protobuf-c" "-IE:\dev\eclipse\/arduinoPlugin/packages/m5stack/hardware/esp32/1.0.7/tools/sdk/include/protocomm" "-IE:\dev\eclipse\/arduinoPlugin/packages/m5stack/hardware/esp32/1.0.7/tools/sdk/include/pthread" "-IE:\dev\eclipse\/arduinoPlugin/packages/m5stack/hardware/esp32/1.0.7/tools/sdk/include/sdmmc" "-IE:\dev\eclipse\/arduinoPlugin/packages/m5stack/hardware/esp32/1.0.7/tools/sdk/include/smartconfig_ack" "-IE:\dev\eclipse\/arduinoPlugin/packages/m5stack/hardware/esp32/1.0.7/tools/sdk/include/soc" "-IE:\dev\eclipse\/arduinoPlugin/packages/m5stack/hardware/esp32/1.0.7/tools/sdk/include/spi_flash" "-IE:\dev\eclipse\/arduinoPlugin/packages/m5stack/hardware/esp32/1.0.7/tools/sdk/include/spiffs" "-IE:\dev\eclipse\/arduinoPlugin/packages/m5stack/hardware/esp32/1.0.7/tools/sdk/include/tcp_transport" "-IE:\dev\eclipse\/arduinoPlugin/packages/m5stack/hardware/esp32/1.0.7/tools/sdk/include/tcpip_adapter" "-IE:\dev\eclipse\/arduinoPlugin/packages/m5stack/hardware/esp32/1.0.7/tools/sdk/include/ulp" "-IE:\dev\eclipse\/arduinoPlugin/packages/m5stack/hardware/esp32/1.0.7/tools/sdk/include/vfs" "-IE:\dev\eclipse\/arduinoPlugin/packages/m5stack/hardware/esp32/1.0.7/tools/sdk/include/wear_levelling" "-IE:\dev\eclipse\/arduinoPlugin/packages/m5stack/hardware/esp32/1.0.7/tools/sdk/include/wifi_provisioning" "-IE:\dev\eclipse\/arduinoPlugin/packages/m5stack/hardware/esp32/1.0.7/tools/sdk/include/wpa_supplicant" "-IE:\dev\eclipse\/arduinoPlugin/packages/m5stack/hardware/esp32/1.0.7/tools/sdk/include/xtensa-debug-module" "-IE:\dev\eclipse\/arduinoPlugin/packages/m5stack/hardware/esp32/1.0.7/tools/sdk/include/esp-face" "-IE:\dev\eclipse\/arduinoPlugin/packages/m5stack/hardware/esp32/1.0.7/tools/sdk/include/esp32-camera" "-IE:\dev\eclipse\/arduinoPlugin/packages/m5stack/hardware/esp32/1.0.7/tools/sdk/include/esp-face" "-IE:\dev\eclipse\/arduinoPlugin/packages/m5stack/hardware/esp32/1.0.7/tools/sdk/include/fb_gfx" -std=gnu++11 -Os -g3 -Wpointer-arith -fexceptions -fstack-protector -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -Wall -Werror=all -Wextra -Wno-error=maybe-uninitialized -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-unused-but-set-parameter -Wno-missing-field-initializers -Wno-sign-compare -fno-rtti -MMD -c -DF_CPU=240000000L -DARDUINO=10812 -DARDUINO_M5STACK_Core2 -DARDUINO_ARCH_ESP32 "-DARDUINO_BOARD=\"M5STACK_Core2\"" -DARDUINO_VARIANT="m5stack_core2"  -DESP32 -DCORE_DEBUG_LEVEL=1 -DBOARD_HAS_PSRAM -mfix-esp32-psram-cache-issue   -I"E:\dev\eclipse\arduinoPlugin\packages\m5stack\hardware\esp32\1.0.7\cores\esp32" -I"E:\dev\eclipse\arduinoPlugin\packages\m5stack\hardware\esp32\1.0.7\variants\m5stack_core2" -I"E:\dev\eclipse\arduinoPlugin\libraries\FreeRTOS\10.4.3-8\src" -MMD -MP -MF"sloeber.ino.cpp.d" -MT"sloeber.ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "..\sloeber.ino.cpp"   -o "sloeber.ino.cpp.o"
'Finished building: ..\sloeber.ino.cpp'
' '
'Building file: E:\dev\eclipse\arduinoPlugin\libraries\FreeRTOS\10.4.3-8\src\croutine.c'
'Starting C compile'
"E:\dev\eclipse\arduinoPlugin\packages\m5stack\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-gcc" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX "-IE:\dev\eclipse\/arduinoPlugin/packages/ [... cut ...]
**E:\dev\eclipse\arduinoPlugin\libraries\FreeRTOS\10.4.3-8\src\FreeRTOSConfig.h:30:20: fatal error: avr/io.h: No such file or directory**
compilation terminated.
libraries\FreeRTOS\src\subdir.mk:49: recipe for target 'libraries\FreeRTOS\src\croutine.c.o' failed
make: *** [libraries\FreeRTOS\src\croutine.c.o] Error 1
"E:/dev/eclipse/arduinoPlugin/tools/make/make all" terminated with exit code 2. Build might be incomplete.

18:15:39 Build Failed. 3 errors, 0 warnings. (took 1s.266ms)

**Expected behavior**
Successful build & logging.

[image]

jantje, 2021-02-05:
If I understand correctly the freeRtos library is included but you do not need/use it.
In other words I assume Sloeber falsely identifies freeRtos as a library your project needs and the solution is to stop Sloeber doing so.

First of all: you can tell Sloeber to add libraries automatically or not. Disabling the automatically addition of libraries is a guaranteed solution but due to the fact this is a workspace setting .... tedious when creating new projects.

Check in windows->preferences

[image]

So sometimes Sloeber add libraries that are not needed. Sloeber searches for libraries (when the option is turned on) based on "unresolved includes" provided by the CDT indexer.
Falsely identified "unresolved includes" can be caused by a confused indexer, a wrongly configured indexer or by something in the code that makes the indexer think there are unresolved includes (I know there is 1 board that always fails the build test due to this issue).
It is possible the confusion is already fixed and then simply deleting the libraries from the libraries folder solves that problem. 

[image]

If the libraries are automatically reattached you can force the indexer to rebuild.

[image]

Then delete the libraries.
If they are still reattached I exclude the library from the build using the resource configuration.

[image]
[image]
The visualisation of the folder changes showing it is excluded
[image]

HiroP0, 2021-02-06:
Workaround (not including libraries automatically) helps and is acceptable. 
Thanks and Cheers!

Brolensky, 2021-02-13:
Hello,
I have got a problem with ESP32 with basic projet.
Here is my all in one view:
[image]

The problem is
If i <clean project>, <index freshen, rebuild> and put libraries and <build all> or <buildd project>, it will be ok or no. 
When it is good it is only few time, after Sloeber.ino.h got the libraries include and the problems appears.
Even i erease the includes it still non working!

I have have configure the enable parrallel built to 8.
If have made the git ESP32 external project import and inclusion like in the vidéo 5 times and now step by step.
I am trying to adding a reference to  
"${workspace_loc:/Toto4/core/variant}"
"${workspace_loc:/Toto4/core/core}"
"${C:\Users\cabrol\git\arduino-esp32\libraries}"
the problem is with or without last line

So what is my mystake?
Thank for your Help 
Cedric

Brolensky, 2021-02-13:
Event i comment the includes, sloeber.ino.cpp ask to put a new version with the includes...

Brolensky, 2021-02-13:
Here is the out console message for a working project Toto5, the same then Toto4 before including anything
the whitespace problem seems to have not importance.

08:58:50 **** Incremental Build of configuration Release for project Toto5 ****
"C:\\Sloeber\\arduinoPlugin\\tools\\make\\make" all 
C:/Sloeber/arduinoPlugin/tools/make/make --no-print-directory pre-build
cmd /c if exist "C:\Sloeber\Workspace2\Toto5\partitions.csv" copy /y "C:\Sloeber\Workspace2\Toto5\partitions.csv" "C:\Sloeber\Workspace2\Toto5\Release\partitions.csv"
cmd /c if not exist "C:\Sloeber\Workspace2\Toto5\Release\partitions.csv" copy "C:/Users/cabrol/git/arduino-esp32\tools\partitionsdefault_ffat.csv" "C:\Sloeber\Workspace2\Toto5\Release\partitions.csv"
' '
C:/Sloeber/arduinoPlugin/tools/make/make --no-print-directory main-build
'Building file: ..\sloeber.ino.cpp'
'Starting C++ compile'
"C:/Users/cabrol/git/arduino-esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/config" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/app_trace" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/app_update" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/asio" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/bootloader_support" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/bt" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/coap" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/console" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/driver" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/efuse" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/esp-tls" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/esp32" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/esp_adc_cal" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/esp_event" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/esp_http_client" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/esp_http_server" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/esp_https_ota" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/esp_https_server" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/esp_ringbuf" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/esp_websocket_client" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/espcoredump" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/ethernet" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/expat" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/fatfs" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/freemodbus" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/freertos" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/heap" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/idf_test" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/jsmn" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/json" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/libsodium" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/log" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/lwip" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/mbedtls" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/mdns" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/micro-ecc" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/mqtt" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/newlib" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/nghttp" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/nimble" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/nvs_flash" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/openssl" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/protobuf-c" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/protocomm" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/pthread" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/sdmmc" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/smartconfig_ack" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/soc" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/spi_flash" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/spiffs" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/tcp_transport" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/tcpip_adapter" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/ulp" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/unity" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/vfs" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/wear_levelling" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/wifi_provisioning" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/wpa_supplicant" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/xtensa-debug-module" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/esp-face" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/esp32-camera" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/esp-face" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/fb_gfx" -std=gnu++11 -Os -g3 -Wpointer-arith -fexceptions -fstack-protector -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -Wall -Werror=all -Wextra -Wno-error=maybe-uninitialized -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=d [... cut ...] ls/sdk/include/heap" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/idf_test" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/jsmn" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/json" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/libsodium" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/log" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/lwip" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/mbedtls" "-IC:/Users/cabrol/git/arduino-esp32/tool

Brolensky, 2021-02-13:
A) Adding an #include "SD.h" in .ino and direct buid
kill the project 

B) Multi step
 1 ) Creating project with 8 parrallel built, OK.
 2) Adding an #include "SD.h" in .ino underligne the #include "SD.h"
 3) Searching unresolved inclusions says everithing is OK
 4) Build project says OK, but SD.h is undertligned SD begin is wrong too & searching unresolved is OK???

C)
1) Searching unresolved before a build says OK, but it is underligned and the loop SD.begin() Serial