Sloeber/arduino-eclipse-plugin #1296: Other bad includes with ESP32

Brolensky, 2021-02-13:
Hello,
I have got a problem with ESP32 with basic projet.
Here is my all in one view:
[image]

The problem is
If i "clean project", "index freshen, rebuild" and put libraries and "build all" or "build project", it will be ok or no. 
When it is good it is only few time, after Sloeber.ino.h got the libraries include and the problems appears.
Even i erease the includes it still non working!

I have have configure the enable parrallel built to 8.
If have made the git ESP32 external project import and inclusion like in the vidéo 5 times and now step by step.
(The only difference are the use versions, i reproduce the new project before get procedure and the project doesnt create at this step (invisible), i erase and made a new after get procedure)
I am trying to adding a reference to  
"${workspace_loc:/Toto4/core/variant}"
"${workspace_loc:/Toto4/core/core}"
"${C:\Users\cabrol\git\arduino-esp32\libraries}"
the problem is with or without last line

So what is my mystake?
Thank for your Help 
Cedric

_Originally posted by @Brolensky in https://github.com/Sloeber/arduino-eclipse-plugin/issues/1295#issuecomment-778575290_

Brolensky, 2021-02-13:
Even i comment the includes, sloeber.ino.cpp ask to put a new version with the includes...
If i exclude sloeber.ino.cpp from built i have got error form setup and loop.
when i re-include, even the #include "SD.h" is comment, the comment is underline with the error for library 
Refresh doesnt do anything

Here is the complete console output message (with out any visible reference to SD & WiFi):

08:30:47 **** Incremental Build of configuration Release for project Toto4 ****
"C:\\Sloeber\\arduinoPlugin\\tools\\make\\make" -j8 all 
C:/Sloeber/arduinoPlugin/tools/make/make --no-print-directory pre-build
echo prebuild;cmd /c if exist "C:\Sloeber\Workspace2\Toto4\partitions.csv" copy /y "C:\Sloeber\Workspace2\Toto4\partitions.csv" "C:\Sloeber\Workspace2\Toto4\Release\partitions.csv"
prebuild;cmd /c if exist "C:\Sloeber\Workspace2\Toto4\partitions.csv" copy /y "C:\Sloeber\Workspace2\Toto4\partitions.csv" "C:\Sloeber\Workspace2\Toto4\Release\partitions.csv"
cmd /c if exist "C:\Sloeber\Workspace2\Toto4\partitions.csv" copy /y "C:\Sloeber\Workspace2\Toto4\partitions.csv" "C:\Sloeber\Workspace2\Toto4\Release\partitions.csv"
        1 fichier(s) copi‚(s).
cmd /c if not exist "C:\Sloeber\Workspace2\Toto4\Release\partitions.csv" copy "C:/Users/cabrol/git/arduino-esp32\tools\partitionsdefault_ffat.csv" "C:\Sloeber\Workspace2\Toto4\Release\partitions.csv"
' '
C:/Sloeber/arduinoPlugin/tools/make/make --no-print-directory main-build
'Building file: ..\sloeber.ino.cpp'
'Building file: C:\Users\cabrol\git\arduino-esp32\cores\esp32\libb64\cdecode.c'
'Starting C++ compile'
"C:/Users/cabrol/git/arduino-esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/config" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/app_trace" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/app_update" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/asio" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/bootloader_support" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/bt" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/coap" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/console" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/driver" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/efuse" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/esp-tls" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/esp32" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/esp_adc_cal" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/esp_event" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/esp_http_client" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/esp_http_server" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/esp_https_ota" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/esp_https_server" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/esp_ringbuf" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/esp_websocket_client" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/espcoredump" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/ethernet" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/expat" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/fatfs" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/freemodbus" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/freertos" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/heap" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/idf_test" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/jsmn" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/json" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/libsodium" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/log" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/lwip" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/mbedtls" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/mdns" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/micro-ecc" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/mqtt" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/newlib" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/nghttp" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/nimble" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/nvs_flash" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/openssl" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/protobuf-c" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/protocomm" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/pthread" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/sdmmc" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/smartconfig_ack" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/soc" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/spi_flash" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/spiffs" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/tcp_transport" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/tcpip_adapter" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/ulp" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/unity" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/vfs" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/wear_levelling" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/wifi_provisioning" "-IC:/Users/ [... cut ...]
'Finished building: C:\Users\cabrol\git\arduino-esp32\cores\esp32\Esp.cpp'
'Finished building: C:\Users\cabrol\git\arduino-esp32\cores\esp32\IPv6Address.cpp'
' '
' '
'Finished building: C:\Users\cabrol\git\arduino-esp32\cores\esp32\MD5Builder.cpp'
' '
'Finished building: C:\Users\cabrol\git\arduino-esp32\cores\esp32\Print.cpp'
' '
make: *** [makefile:65: all] Error 2
"C:/Sloeber/arduinoPlugin/tools/make/make -j8 all" terminated with exit code 2. Build might be incomplete.

08:30:49 Build Failed. 3 errors, 10 warnings. (took 2s.660ms)

Brolensky, 2021-02-13:
Here is the out console message for a working project Toto5, the same then Toto4 before including anything
the whitespace problem seems to have not importance.

08:58:50 **** Incremental Build of configuration Release for project Toto5 ****
"C:\\Sloeber\\arduinoPlugin\\tools\\make\\make" all 
C:/Sloeber/arduinoPlugin/tools/make/make --no-print-directory pre-build
cmd /c if exist "C:\Sloeber\Workspace2\Toto5\partitions.csv" copy /y "C:\Sloeber\Workspace2\Toto5\partitions.csv" "C:\Sloeber\Workspace2\Toto5\Release\partitions.csv"
cmd /c if not exist "C:\Sloeber\Workspace2\Toto5\Release\partitions.csv" copy "C:/Users/cabrol/git/arduino-esp32\tools\partitionsdefault_ffat.csv" "C:\Sloeber\Workspace2\Toto5\Release\partitions.csv"
' '
C:/Sloeber/arduinoPlugin/tools/make/make --no-print-directory main-build
'Building file: ..\sloeber.ino.cpp'
'Starting C++ compile'
"C:/Users/cabrol/git/arduino-esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/config" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/app_trace" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/app_update" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/asio" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/bootloader_support" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/bt" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/coap" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/console" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/driver" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/efuse" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/esp-tls" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/esp32" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/esp_adc_cal" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/esp_event" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/esp_http_client" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/esp_http_server" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/esp_https_ota" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/esp_https_server" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/esp_ringbuf" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/esp_websocket_client" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/espcoredump" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/ethernet" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/expat" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/fatfs" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/freemodbus" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/freertos" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/heap" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/idf_test" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/jsmn" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/json" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/libsodium" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/log" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/lwip" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/mbedtls" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/mdns" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/micro-ecc" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/mqtt" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/newlib" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/nghttp" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/nimble" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/nvs_flash" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/openssl" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/protobuf-c" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/protocomm" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/pthread" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/sdmmc" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/smartconfig_ack" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/soc" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/spi_flash" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/spiffs" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/tcp_transport" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/tcpip_adapter" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/ulp" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/unity" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/vfs" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/wear_levelling" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/wifi_provisioning" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/wpa_supplicant" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/xtensa-debug-module" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/esp-face" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/esp32-camera" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/esp-face" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/fb_gfx" -std=gnu++11 -Os -g3 -Wpointer-arith -fexceptions -fstack-protector -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -Wall -Werror=all -Wextra -Wno-error=maybe-uninitialized -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=d [... cut ...] ls/sdk/include/heap" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/idf_test" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/jsmn" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/json" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/libsodium" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/log" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/lwip" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/mbedtls" "-IC:/Users/cabrol/git/arduino-esp32/tool

jantje, 2021-02-13:
you are probably hitting #1184 and #1295 (containing workaround)
#1184 should be fixed in the next release that we are currently testing in beta https://github.com/Sloeber/arduino-eclipse-plugin/discussions/1290
and turn off parallel  build when trouble shooting.

Brolensky, 2021-02-13:
fHi thanks Jantje, 
I am going to test yours solutions... and says what is correct to fix that.

Brolensky, 2021-02-13:
In #1184 your asking for checking project->properties->C/C++ build->environment. 
and find something like A.RUNTIME.TOOLS.XTENSA-AVR-ELF-GCC-PATH (for AVR)

I instal ESP32 as your vidéo tutotrial so i dont put esp32S2
my string is:
=>A.RUNTIME.TOOLSXTENSA-ESP32-ELF-GCC.PATH   
=>${A.RUNTINE.PLATFORM.PATH}/tools/extensa-esp32-elf
=>USER: CONFIG

So i think it is ok for that, iam going to read the next step in #1184

Brolensky, 2021-02-13:
I have looking all #1184, i am using 4.3.3 
It difficult for me to understand what i have really to do, but i am going to try...

jantje, 2021-02-13:
>I have uncheck expand environment variable
and stop parrallels build.
Have i to delete or modify a file?

No but removing the Release folder is making sure 100%

Using a github repository as private hardware is difficult.

Note that my workaround statement was only for #1295 there is no workaround for #1184

Brolensky, 2021-02-13:
I put partitions in the general folder

Brolensky, 2021-02-13:
I have to make my big project.
I want to mesure color of the soil.
So i have to take picture, get pixels, and communicate with WiFi to an ESP8266 and MEGA board to get position with telemeter...

For the first time (pixels) i run it on IDE, but including all libraries i hope working with Sloeber.

Brolensky, 2021-02-13:
I put the project on a ESP32 1.0.4: i have got 225 bads includes!!!
I going to rebuild a new Toto project

Brolensky, 2021-02-13:
I have better results refering copy of github repertory than using the ESP32 1.0.4
Do you confirm it right having 2 errors vs 234! Or have i got an other mystake?

On problem with ESP32 Reference project is the integration of memory, how could i make a string path to have tools in the environnement?

Brolensky, 2021-02-13:
New test with Github Esp32 (not 1.0.4)
1) Erase Release
2) Comment Include (on cpp & .ino)
3) Clean
4) unresolve libraireies says SD is not ok?
   if i do good theres no libraries on the ino & cpp (clean, refresh, not paste...)
   for 1 build without expension variables i have got fantoms errors of SD that doesnt exist...

Brolensky, 2021-02-13:
So i have analysed the process (and close and save the files)

adding SD.h,

Builds N°:
1)  not ok beacause SD.begin is not solved, but no problems with SD.h it is founded!
2) SD.begin() is OK, SD.h is OK in ino files!!!   But not ok in .cpp SD.h not found!!!

And it is possible to see SD.h ok on a file and not ok on the other...
If i look in the SD.h FS.h is bad, but if i write FS.h in the ino it is OK! 
>I think this is the problem! And maybe the solution

Brolensky, 2021-02-13:
If in .ino or sloeber.ino.cpp: "SD.h' is not ok <SD.h> make it Ok

Brolensky, 2021-02-13:
I think it is quasi sure it is better to refer to the gitub project, despite warning "unresolve include" will appear turning off the reference

Brolensky, 2021-02-13:
-  But the SD.h in the reference github project is on error with or without the ""<>
- Reincluding the project Githib = no error

Brolensky, 2021-02-13:
last observation:
if i change the ino file <SD.h> to "SD.h", if i save sloeber change and the error give up
if i prepared the solution and not saving the build all  dont change the sloeber so it is a bug but , if the system construct an edit the new sloeber before the build maybe he willnot be on error.

For changing "" to <> it is possible by creating a script based on pair and not pair compile or comment uncomment before ech compile.

Brolensky, 2021-02-13:
Last observation, simple:
if the project refered to github, when de include change for exemple erase " to remplace with " the underline diseaper  in Sloeber.ino.cpp (but the point need a clean to escape)

Brolensky, 2021-02-13:
"-----------------------------------"
So for partial fixing
"-----------------------------------"

- Partial because it is working with "Wifi.h" but not with "SD.h"
- Use a board as a Github hub 
- AND make an external Reference to the Github hub project
- check generate automaticaly makefiles
- dont check expand variables
- use parrallel job 2 times (i dont test others)
- dont check build on resource save
- check Buil incremental
- dont check Clean

And so WiFi will work if you #include "WiFi.h" all  the time
But       SD will not work....

in the project reference of C++ build are 
-I"C:\Users\cabrol\git\arduino-esp32\cores\esp32" 
-I"C:\Users\cabrol\git\arduino-esp32\variants\esp32"
so i think we have to add tools wich make SD crash because of <memory>
And i dont know to do correctly  this point...
I am going to try working without SD!i!i!i!i!i!i!i!i!i!i!i!i!i!i!i!i!i!i!i!!i!i!i!i!i!i!i!i!i!i!i!i!i!i!!i!i!i!D<[]]]])

Brolensky, 2021-02-13:
I include 3 libraries, the problem is always the 1 library of the sloeber.ino.cpp

Brolensky, 2021-02-13:
All time the bad include is correctly fix but in this extract

1:48:14 **** Incremental Build of configuration Release for project Toto6 ****
"C:\\Sloeber\\arduinoPlugin\\tools\\make\\make" all 
C:/Sloeber/arduinoPlugin/tools/make/make --no-print-directory pre-build
echo preBuild
preBuild
cmd /c if exist "C:\Sloeber\Workspace2\Toto6\partitions.csv" copy /y "C:\Sloeber\Workspace2\Toto6\partitions.csv" "C:\Sloeber\Workspace2\Toto6\Release\partitions.csv"
        1 fichier(s) copi‚(s).
cmd /c if not exist "C:\Sloeber\Workspace2\Toto6\Release\partitions.csv" copy "C:/Users/cabrol/git/arduino-esp32\tools\partitionsdefault_ffat.csv" "C:\Sloeber\Workspace2\Toto6\Release\partitions.csv"
' '
C:/Sloeber/arduinoPlugin/tools/make/make --no-print-directory main-build
'Building file: ..\sloeber.ino.cpp'
'Starting C++ compile'
"C:/Users/cabrol/git/arduino-esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/config" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/app_trace" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/app_update" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/asio" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/bootloader_support" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/bt" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/coap" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/console" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/driver" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/efuse" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/esp-tls" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/esp32" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/esp_adc_cal" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/esp_event" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/esp_http_client" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/esp_http_server" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/esp_https_ota" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/esp_https_server" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/esp_ringbuf" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/esp_websocket_client" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/espcoredump" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/ethernet" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/expat" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/fatfs" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/freemodbus" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/freertos" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/heap" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/idf_test" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/jsmn" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/json" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/libsodium" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/log" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/lwip" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/mbedtls" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/mdns" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/micro-ecc" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/mqtt" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/newlib" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/nghttp" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/nimble" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/nvs_flash" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/openssl" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/protobuf-c" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/protocomm" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/pthread" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/sdmmc" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/smartconfig_ack" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/soc" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/spi_flash" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/spiffs" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/tcp_transport" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/tcpip_adapter" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/ulp" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/unity" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/vfs" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/wear_levelling" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/wifi_provisioning" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/wpa_supplicant" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/xtensa-debug-module" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/esp-face" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/esp32-camera" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/esp-face" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/fb_gfx" -std=gnu++11 -Os -g3 -Wpointer-arith -fexceptions -fstack-protector -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -Wall -Werror=all -Wextra -Wno-error=maybe-uninitialized -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter  [... cut ...]
compilation terminated.
make[1]: *** [subdir.mk:20: sloeber.ino.cpp.o] Error 1
make: *** [makefile:65: all] Error 2
"C:/Sloeber/arduinoPlugin/tools/make/make all" terminated with exit code 2. Build might be incomplete.

21:48:15 Build Failed. 3 errors, 1 warnings. (took 747ms)

we cant see <command-line>:0:21: warning: ISO C++11 requires whitespace after the macro name
and just after the bad include. Maybe it is include but the old error is read instead of the itisnotanerror? Reading a boolean not put to false?

Brolensky, 2021-02-13:
[image]
Why Path is the only in red?

jantje, 2021-02-13:
>Why Path is the only in red?

because it is/overwrites a system variable

jantje, 2021-02-13:
I think you are also hitting #1210

Brolensky, 2021-02-14:
Hello @jantje,
Thanks for your analysis, i have already do your recomandation and download the path make... so i dont think the truncated is present.

>The problem of truncating commands is in the windows component cmd.exe.

>The difference between the 2 make.zip's is that one contains make.exe and sh.exe the other only make.exe.

>If there is a sh.exe in the folder next to make.exe; make.exe will use sh.exe and not cmd.exe (at least with the supplied make.exe)

>As sh.exe does not have the truncation problem I hoped to switch to sh.exe.

>However

>sh.exe behaves very similarly to cmd.exe

>but there are differences.

>These differences between make.exe and sh.exe cause problems for certain boards.

>So the zip with make.exe and sh.exe can be used as a workaround but not as general solution.

What do you think about whitespaces?
Is it possible the system made an error with whitspace (but i think it is resolved and not a matter), but maybe the boolean error will be true and next step error is initialized with true so it is crashing without error?

Brolensky, 2021-02-14:
What about mixing "\" &"/" in the file code
C:\Users\cabrol\git\arduino-esp32\libraries\SD\src/SD.h

Problematic or no?

Brolensky, 2021-02-14:
Absolute path in FS give bugg on FS interagration
using #include "C:\Users\cabrol\git\arduino-esp32\tools\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\memory"
>C:\Users\cabrol\git\arduino-esp32\libraries\FS\src\FS.h:24:112: fatal error: C:\Users\cabrol\git\arduino-esp32\tools\xtensa->esp32-elf\xtensa-esp32-elf\include\c++\5.2.0: No such file or directory

A wrong path gives same result
using #include "C:\Users\cabrol\git\arduino-esp32\tools\includes\memory"
>C:\Users\cabrol\git\arduino-esp32\libraries\FS\src\FS.h:24:112: fatal error: C:\Users\cabrol\git\arduino-esp32\tools\xtensa->esp32-elf\xtensa-esp32-elf\include\c++\5.2.0: No such file or directory

Brolensky, 2021-02-14:
What about an other library?
Using WiFi crash...surprising?

But in WiFi there are more inclusions! This is the list:

>#ifndef WiFi_h
>#define WiFi_h

>#include <stdint.h>

>#include "Print.h"
>#include "IPAddress.h"
>#include "IPv6Address.h"

>#include "WiFiType.h"
>#include "WiFiSTA.h"
>#include "WiFiAP.h"
>#include "WiFiScan.h"
>#include "WiFiGeneric.h"

>#include "WiFiClient.h"
>#include "WiFiServer.h"
>#include "WiFiUdp.h"

The message is 
>C:\Users\cabrol\git\arduino-esp32\libraries\WiFi\src\WiFiClient.h:42:10: error: 'shared_ptr' in namespace 'std' does not name a template type

So sysytem have done 9 inclusion fefore crash! What are the include in WiFiClient?
This is the list

>#ifndef _WIFICLIENT_H_
>#define _WIFICLIENT_H_

>#include "Arduino.h"
>#include "Client.h"
>#include <memory>

<memory>?

Brolensky, 2021-02-14:
If i made a multiple jobs without stopping on errors it seems it is good after second job
in .ino files but error on Wifitypes in sloeber.ino.cpp
for resolve, i have to put absolute path

> #include <Arduino.h>
> 	//#include "C:\Users\cabrol\git\arduino-esp32\libraries\FS\src\FS.h"
> 	//#include "C:\Users\cabrol\git\arduino-esp32\libraries\SD\src\SD.h"//C:\Users\cabrol\git\arduino-esp32\libraries\SD\src\//
> //#include "C:\Users\cabrol\git\arduino-esp32\libraries\WiFiScan\src\WiFiScan.h"
> //#include "C:\Users\cabrol\git\arduino-esp32\libraries\WiFi\src\WiFi.h"
> 
> #include <stdint.h>
> 
> #include "Print.h"
> #include "IPAddress.h"
> #include "IPv6Address.h"
> 
> #include "C:\Users\cabrol\git\arduino-esp32\libraries\WiFi\src\WiFiType.h"
> #include "C:\Users\cabrol\git\arduino-esp32\libraries\WiFi\src\WiFiSTA.h"
> #include "C:\Users\cabrol\git\arduino-esp32\libraries\WiFi\src\WiFiAP.h"
> #include "C:\Users\cabrol\git\arduino-esp32\libraries\WiFi\src\WiFiScan.h"
> #include "C:\Users\cabrol\git\arduino-esp32\libraries\WiFi\src\WiFiGeneric.h"
> 
> #include "C:\Users\cabrol\git\arduino-esp32\libraries\WiFi\src\WiFiClient.h"
> #include "C:\Users\cabrol\git\arduino-esp32\libraries\WiFi\src\WiFiServer.h"
> #include "C:\Users\cabrol\git\arduino-esp32\libraries\WiFi\src\WiFiUdp.h"
> #include "C:\Users\cabrol\git\arduino-esp32\libraries\WiFi\src\WiFi.h"
> //The setup function is called once at startup of the sketch
> void setup()
> {
>      Serial.begin(9600);
> }
> 
> // The loop function is called in an endless loop
> void loop()
> {
> 	WiFi.begin();
> 	//SD.begin();
> 	Serial.println("Hey");
> }

for finaly have 

> C:\Users\cabrol\git\arduino-esp32\libraries\WiFi\src\WiFiClient.h:42:10: error: 'shared_ptr' in namespace 'std' does not name a template type

CONCLUSION
For WiFi inclusion no path problem but type problem!

Brolensky, 2021-02-14:
After runing 8 time without stops, and warning only on declarations problems:
- no errors with all WiFi libraries in .ino and sloeber.ino.cpp! 
- some errors and this

> String EspClass::getSketchMD5()
> {
>     static String result;
>     if (result.length()) {
>         return result;
>     }
>     uint32_t lengthLeft = getSketchSize();
> 
>     const esp_partition_t *running = esp_ota_get_running_partition();
>     if (!running) {
>         log_e("Partition could not be found");
> 
>         return String();
>     }
>     const size_t bufSize = SPI_FLASH_SEC_SIZE;
>     std::unique_ptr<uint8_t[]> buf(new uint8_t[bufSize]);
>     uint32_t offset = 0;
>     if(!buf.get()) {
>         log_e("Not enough memory to allocate buffer");
> 
>         return String();
>     }
>     MD5Builder md5;
>     md5.begin();
>     while( lengthLeft > 0) {
>         size_t readBytes = (lengthLeft < bufSize) ? lengthLeft : bufSize;
>         if (!ESP.flashRead(running->address + offset, reinterpret_cast<uint32_t*>(buf.get()), (readBytes + 3) & ~3)) {
>             log_e("Could not read buffer from flash");
> 
>             return String();
>         }
>         md5.add(buf.get(), readBytes);
>         lengthLeft -= readBytes;
>         offset += readBytes;
>     }
>     md5.calculate();
>     result = md5.toString();
>     return result;
> }

the error is here buf(new uint8_t[bufSize]); buf is non declared in the scope

And i dont know what doing with this complex declaration

CONCLUSION
- for memory problem if call by SD (FS), no problem if call direct FS
- for WiFi.h inclusion OK, but declaration problem for buff in ESP.cpp ligne 180

Brolensky, 2021-02-14:
For an other library esp_camera same problem with unique_ptr ...

> > 1:15:09 **** Incremental Build of configuration Release for project Toto6 ****
> "C:\\Sloeber\\arduinoPlugin\\tools\\make\\make" -k -j8 all 
> C:/Sloeber/arduinoPlugin/tools/make/make --no-print-directory pre-build
> echo preBuild
> preBuild
> cmd /c if exist "C:\Sloeber\Workspace2\Toto6\partitions.csv" copy /y "C:\Sloeber\Workspace2\Toto6\partitions.csv" "C:\Sloeber\Workspace2\Toto6\Release\partitions.csv"
>         1 fichier(s) copi‚(s).
> cmd /c if not exist "C:\Sloeber\Workspace2\Toto6\Release\partitions.csv" copy "C:/Users/cabrol/git/arduino-esp32\tools\partitionsdefault_ffat.csv" "C:\Sloeber\Workspace2\Toto6\Release\partitions.csv"
> ' '
> C:/Sloeber/arduinoPlugin/tools/make/make --no-print-directory main-build
> 'Building file: C:\Users\cabrol\git\arduino-esp32\cores\esp32\Esp.cpp'
> 'Starting C++ compile'
> "C:/Users/cabrol/git/arduino-esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/config" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/app_trace" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/app_update" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/asio" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/bootloader_support" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/bt" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/coap" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/console" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/driver" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/efuse" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/esp-tls" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/esp32" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/esp_adc_cal" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/esp_event" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/esp_http_client" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/esp_http_server" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/esp_https_ota" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/esp_https_server" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/esp_ringbuf" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/esp_websocket_client" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/espcoredump" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/ethernet" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/expat" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/fatfs" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/freemodbus" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/freertos" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/heap" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/idf_test" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/jsmn" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/json" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/libsodium" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/log" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/lwip" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/mbedtls" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/mdns" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/micro-ecc" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/mqtt" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/newlib" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/nghttp" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/nimble" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/nvs_flash" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/openssl" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/protobuf-c" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/protocomm" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/pthread" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/sdmmc" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/smartconfig_ack" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/soc" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/spi_flash" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/spiffs" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/tcp_transport" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/tcpip_adapter" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/ulp" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/unity" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/vfs" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/wear_levelling" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/wifi_provisioning" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/wpa_supplicant" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/xtensa-debug-module" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/esp-face" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/esp32-camera" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/esp-face" "-IC:/Users/cabrol/git/arduino-esp32/tools/sdk/include/fb_gfx" -std=gnu++11 -Os -g3 -Wpointer-arith -fexceptions -fstack-protector -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -Wall -Werror=all -Wextra -Wno-error=maybe-uninitialized -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error= [... cut ...]
> C:\Users\cabrol\git\arduino-esp32\cores\esp32\Esp.cpp:180:56: error: 'buf' was not declared in this scope
>      std::unique_ptr<uint8_t[]> buf(new uint8_t[bufSize]);
>                                                         ^
> make[1]: *** [core\core\subdir.mk:123: core\core\Esp.cpp.o] Error 1
> make[1]: Target 'main-build' not remade because of errors.
> make: *** [makefile:65: all] Error 2
> "C:/Sloeber/arduinoPlugin/tools/make/make -k -j8 all" terminated with exit code 2. Build might be incomplete.
> 
> 11:15:11 Build Failed. 6 errors, 1 warnings. (took 1s.413ms)

uzi18, 2021-02-14:
tried this example: https://randomnerdtutorials.com/esp32-data-logging-temperature-to-microsd-card/

used sloeber 4.4 and 4.3 with victory :)

Please attach your compressed project to test compilation

Brolensky, 2021-02-14:
Hello @uzi18,
I'am just try to make #include SD.h and SD.begin();
I am not shure complex project is OK (difficult working with Sloeber and IDE)
Maybe have a look on this, it is my base project and it is never working on sloeber
https://www.instructables.com/Arduino-Selfie-Camera/
Thanks

jantje, 2021-02-14:
As I stated in https://github.com/Sloeber/arduino-eclipse-plugin/issues/1296#issuecomment-778622359
you are probably hitting #1184 because you are using the esp32 github repository. There is nothing you can do about that in your code. You may upgrade to the latest nightly Sloeber but that will probably cause other issues.
There is no need to bombard us with data as due to #1184 it won't work.

@me-no-dev  what is the status of the version causing #1184 ? is it already released. Will you help test it with V4.4 beta? 
https://github.com/Sloeber/arduino-eclipse-plugin/discussions/1290
https://github.com/Sloeber/arduino-eclipse-plugin/releases/tag/V4_4_0-beta

Brolensky, 2021-02-14:
> > As I stated in #1296 (comment)
> you are probably hitting #1184 because you are using the esp32 github repository. There is nothing you can do about that in your code. You may upgrade to the latest nightly Sloeber but that will probably cause other issues.
> There is no need to bombard us with data as due to #1184 it won't work.
> 
> @me-no-dev what is the status of the version causing #1184 ? is it already released. Will you help test it with V4.4 beta?
> #1290
> https://github.com/Sloeber/arduino-eclipse-plugin/releases/tag/V4_4_0-beta

jantje, 2021-02-14:
>Sorrry, OK i am going on 4,4,0 thanks

Note that 4.4 is early beta and project migration from 4.3 is still problematic.

Brolensky, 2021-02-14:
@jantje thanks : I'll going to try . 
I have download jdk 11.0. I have got a directory; where do i have to put it?

Brolensky, 2021-02-14:
OK, Sloeber is running!

uzi18, 2021-02-14:
This code also works ok:
```
#include "Arduino.h"

#include "SD.h"

void setup(){
    SD.begin();

}

void loop(){
Serial.println("Hey");
}

```

uzi18, 2021-02-14:
You can use only 11 for Sloeber 4.3 and 4.4

uzi18, 2021-02-14:
But if you can zip your project and paste here it will be more easy to find what is wrong.

Brolensky, 2021-02-14:
Ok @uzi18 

> This code also works ok:
> 
> #include "Arduino.h"
> 
> #include "SD.h"
> 
> void setup(){
>     SD.begin();
> 
> }
> 
> void loop(){
> Serial.println("Hey");
> }

Could you test #include Wifi.h Wifi.begin();? It doesn't work for me...
Maybe i have got an other problem? 
I'am using java 11 for 4,4 Windows 10 and make modified (for long strings)

uzi18, 2021-02-14:
Yes it works:
```
#include "Arduino.h"

#include "SD.h"
#include "WiFi.h"

void setup(){
    SD.begin();
    WiFi.begin();
}

void loop(){
Serial.println("Hey");
}

Finished building target: esp32_test
 

20:41:50 Build Finished. 0 errors, 0 warnings. (took 5s.407ms)
```

Brolensky, 2021-02-14:
Are you using ESP32S2?
For me a complex MEGA2560 compile too

Brolensky, 2021-02-14:
#inclide WiFi.h is a problem for me because of #include <stdint.h>, even i put it on board esp32 1.0.4

uzi18, 2021-02-14:
please use "project->export" and choose general->archive file, paste here

Brolensky, 2021-02-14:
OK, thanks, 
> please use "project->export" and choose general->archive file, paste here
but i haven't got any complex ESP32 project yet: i complexifie and now it is unsuable on arduino IDE and Sloeber?
(I transposate my ESP8266 to an ESP32, it is worki,g stade)
But if you like i put it, with the ESP8266 with receive values from the MEGA, he likes $ and #$
My problem is that i couldnt make with ESP1.0.4 too, i'am making an other import ESP 1.0.4  and investigate
[Compacto_ESP32_CAM.zip](https://github.com/Sloeber/arduino-eclipse-plugin/files/5978540/Compacto_ESP32_CAM.zip)

Brolensky, 2021-02-14:
This is my problem (for information). i'll going to resolve!

> nstalling Arduino Board Platforms
OK
OK
OK
OK
Failed to install esp32
Failed to extract tar.bz2. "C:/Sloeber/arduinoPlugin/downloads/xtensa-esp32-elf-win32-1.22.0-80-g6c4433a-5.2.0.zip".
Truncated ZIP file
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK

jantje, 2021-02-14:
I see this in your wifi.cpp
```
#include "Arduino.h"
#include "Wifii.h"
#include "C:\Program Files (x86)\Arduino - Copie\libraries\WiFi\src\WiFi.h"
#include "C:\Program Files (x86)\Arduino - Copie\libraries\WiFi\src\WiFiClient.h" //C:\Users\cabrol\git\arduino-esp32\libraries\WiFi\src\//
#include "C:\Program Files (x86)\Arduino - Copie\libraries\WiFiClientSecure\src\WiFiClientSecure.h"
#include "C:\Program Files (x86)\Arduino - Copie\libraries\HTTPSRedirect\src\HTTPSRedirect.h"
```
I won't say this will not work but it for sure doesn't make the code easy to investigate

jantje, 2021-02-14:
>Failed to install esp32
Failed to extract tar.bz2. "C:/Sloeber/arduinoPlugin/downloads/xtensa-esp32-elf-win32-1.22.0-80-g6c4433a-5.2.0.zip".
Truncated ZIP file

The download of xtensa-esp32-elf-win32-1.22.0-80-g6c4433a-5.2.0.zip failed.
delete C:/Sloeber/arduinoPlugin/downloads/xtensa-esp32-elf-win32-1.22.0-80-g6c4433a-5.2.0.zip and try again

uzi18, 2021-02-14:
Indeed it is a problem

uzi18, 2021-02-14:
@jentie maybe after succesfull installation of pkg, add one file "ok" inside directory and reinstall later?

Brolensky, 2021-02-14:
I do again and same problem just killing esp32 1.0.4 file
I restarts kimming the 2 files esp32 1.04 et xtenza

jantje, 2021-02-14:
@uzi18 basically the problem is that Sloeber does not use the checksum/size info in the json file to validate correct download.
Mostly this is not a problem so it is not high on my list.

Brolensky, 2021-02-14:
> @uzi18 basically the problem is that Sloeber does not use the checksum/size info in the json file to validate correct download.
Mostly this is not a problem so it is not high on my list.

I think it is not a problem: i'am just using Sloeber since 15 days, and it will be my first operation... I have the message but ESP8266 works so i go on this... and dont remenber for whats happen the first time in this new strange world

i try

> #include "Arduino.h"
> #include "WiFi.h"
> #include "SD.h"
> 
> void setup()
> {
> 
> 	WiFi.begin();
> 	SD.begin();
> }
> 
> // The loop function is called in an endless loop
> void loop()
> {
> //Add your repeated code here
> }

And i have

> 22:46:34 **** Build of configuration Release for project toto11 ****
> "C:\\Sloeber\\arduinoPlugin\\tools\\make\\make" all 
> 'Building file: ..\sloeber.ino.cpp'
> 'Starting C++ compile'
> "C:\Sloeber\arduinoPlugin\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX "-IC:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/config" "-IC:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/app_trace" "-IC:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/app_update" "-IC:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/asio" "-IC:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/bootloader_support" "-IC:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/bt" "-IC:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/coap" "-IC:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/console" "-IC:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/driver" "-IC:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp-tls" "-IC:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp32" "-IC:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_adc_cal" "-IC:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_event" "-IC:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_http_client" "-IC:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_http_server" "-IC:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_https_ota" "-IC:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_ringbuf" "-IC:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/ethernet" "-IC:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/expat" "-IC:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/fatfs" "-IC:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/freemodbus" "-IC:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/freertos" "-IC:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/heap" "-IC:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/idf_test" "-IC:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/jsmn" "-IC:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/json" "-IC:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/libsodium" "-IC:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/log" "-IC:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/lwip" "-IC:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/mbedtls" "-IC:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/mdns" "-IC:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/micro-ecc" "-IC:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/mqtt" "-IC:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/newlib" "-IC:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/nghttp" "-IC:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/nvs_flash" "-IC:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/openssl" "-IC:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/protobuf-c" "-IC:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/protocomm" "-IC:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/pthread" "-IC:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/sdmmc" "-IC:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/smartconfig_ack" "-IC:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/soc" "-IC:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/spi_flash" "-IC:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/spiffs" "-IC:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/tcp_transport" "-IC:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/tcpip_adapter" "-IC:\Sloeber\/arduinoPlugin/packages/e [... cut ...] :\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi\src" -MMD -MP -MF"sloeber.ino.cpp.d" -MT"sloeber.ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "..\sloeber.ino.cpp"   -o "sloeber.ino.cpp.o"
> xtensa-esp32-elf-g++: error: CreateProcess: No such file or directory
> make: *** [subdir.mk:20: sloeber.ino.cpp.o] Error 1
> "C:/Sloeber/arduinoPlugin/tools/make/make all" terminated with exit code 2. Build might be incomplete.
> 
> 22:46:35 Build Failed. 1 errors, 0 warnings. (took 376ms)

Brolensky, 2021-02-14:
i check #1290, have a special think to do?

For ESP32 1.0.4 #define __IN_ECLIPSE__ is false so Sloeber.ino.cpp is unused.
For ESP32S2 2 errors

> 22:55:38 **** Incremental Build of configuration Release for project toto11 ****
"C:\\Sloeber\\arduinoPlugin\\tools\\make\\make" all 
C:/Sloeber/arduinoPlugin/tools/make/make --no-print-directory pre-build
cmd /c if exist "C:\Sloeber\Workspace2\toto11\partitions.csv" COPY /y "C:\Sloeber\Workspace2\toto11\partitions.csv" "C:\Sloeber\Workspace2\toto11\Release\partitions.csv"
        1 fichier(s) copi‚(s).
cmd /c if not exist "C:\Sloeber\Workspace2\toto11\Release\partitions.csv" COPY "C:/Users/cabrol/git/arduino-esp32S2\tools\partitions\default_ffat.csv" "C:\Sloeber\Workspace2\toto11\Release\partitions.csv"
' '
C:/Sloeber/arduinoPlugin/tools/make/make --no-print-directory main-build
'Building file: ..\sloeber.ino.cpp'
'Starting C++ compile'
"/bin/xtensa-esp32-elf-g++"   -DF_CPU=240000000L -DARDUINO=10812 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ARDUINO-ESP32S2 "-DARDUINO_BOARD=\"ESP32_DEV\"" -DARDUINO_VARIANT="esp32"  -DESP32 -DCORE_DEBUG_LEVEL=1 -DARDUINO_RUNNING_CORE=1 -DARDUINO_EVENT_RUNNING_CORE=0 -DBOARD_HAS_PSRAM -mfix-esp32-psram-cache-issue    -I"C:\Users\cabrol\git\arduino-esp32S2\cores\esp32" -I"C:\Users\cabrol\git\arduino-esp32S2\variants\esp32" -I"C:\Users\cabrol\git\arduino-esp32S2\libraries\SD\src" -I"C:\Users\cabrol\git\arduino-esp32S2\libraries\WiFi\src" -MMD -MP -MF"sloeber.ino.cpp.d" -MT"sloeber.ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "..\sloeber.ino.cpp"   -o "sloeber.ino.cpp.o"
sh: /bin/xtensa-esp32-elf-g++: Permission denied
make[1]: *** [subdir.mk:20: sloeber.ino.cpp.o] Error 126
make: *** [makefile:67: all] Error 2
"C:/Sloeber/arduinoPlugin/tools/make/make all" terminated with exit code 2. Build might be incomplete.

22:55:39 Build Failed. 2 errors, 0 warnings. (took 943ms)

Brolensky, 2021-02-14:
This is an other project for MEGA2560 
[Compacto_MEGA.zip](https://github.com/Sloeber/arduino-eclipse-plugin/files/5978631/Compacto_MEGA.zip)

Brolensky, 2021-02-14:
If i put Toto11 (WiFi.h + SD.h) on MEGA2560 impossible to have WiFi.h 

> Could not create folder "libraries/WiFi".
> Cannot create linked resource '/toto11/libraries/WiFi'.  Files cannot be linked to folders.
> Cannot create linked resource '/toto11/libraries/WiFi'.  Files cannot be linked to folders.

Brolensky, 2021-02-14:
> I see this in your wifi.cpp
>
> ```
> #include "Arduino.h"
> #include "Wifii.h"
> #include "C:\Program Files (x86)\Arduino - Copie\libraries\WiFi\src\WiFi.h"
> #include "C:\Program Files (x86)\Arduino - [...]

Oh, this the past

jantje, 2021-02-15:
I guess there are enough messages in the log stating "this is not supported " to indicate your Sloeber configuration is bad.
I feel like you are biting off more than you can chew.

[13 more replies]

closed 2021-06-04 by jantje