Sloeber/arduino-eclipse-plugin #1297: Disable [-Werror=return-type]

wm1962, 2021-02-13:
Hello, 
when I build my project, I get the error 

> error: control reaches end of non-void function [-Werror=return-type]

The error happens in some included libraries and i don't want to change them.
How can I disable the error ?

I´ve tried to change it in the Preferences\C/C++\Code Analysis
or by a compiler directive like
> #pragma GCC diagnostic ignored "-Wreturn-type"

jantje, 2021-02-13:
and the question/issue is.

wm1962, 2021-02-13:
How can I disable the error-message?

jantje, 2021-02-13:
Is the error message from the build or the code analyser?

wm1962, 2021-02-13:
Sorry I'm no expert. Here is the output

```
22:29:30 **** Incremental Build of configuration Release for project DHZ_2021 ****
"D:\\Sloeber\\arduinoPlugin\\tools\\make\\make" all 
'Building file: D:\User\Maier\Documents\Eigene Dateien\Modellbau\Arduino\libraries\Nextion\NexRtc.cpp'
'Starting C++ compile'
"D:\Sloeber\arduinoPlugin\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX "-ID:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/config" "-ID:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/app_trace" "-ID:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/app_update" "-ID:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/asio" "-ID:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/bootloader_support" "-ID:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/bt" "-ID:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/coap" "-ID:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/console" "-ID:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/driver" "-ID:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp-tls" "-ID:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp32" "-ID:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_adc_cal" "-ID:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_event" "-ID:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_http_client" "-ID:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_http_server" "-ID:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_https_ota" "-ID:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_ringbuf" "-ID:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/ethernet" "-ID:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/expat" "-ID:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/fatfs" "-ID:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/freemodbus" "-ID:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/freertos" "-ID:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/heap" "-ID:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/idf_test" "-ID:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/jsmn" "-ID:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/json" "-ID:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/libsodium" "-ID:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/log" "-ID:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/lwip" "-ID:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/mbedtls" "-ID:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/mdns" "-ID:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/micro-ecc" "-ID:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/mqtt" "-ID:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/newlib" "-ID:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/nghttp" "-ID:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/nvs_flash" "-ID:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/openssl" "-ID:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/protobuf-c" "-ID:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/protocomm" "-ID:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/pthread" "-ID:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/sdmmc" "-ID:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/smartconfig_ack" "-ID:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/soc" "-ID:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/spi_flash" "-ID:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/spiffs" "-ID:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/tcp_transport" "-ID:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/tcpip_adapter" "-ID:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/ulp" "-ID:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/vfs" "-ID:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/wear_levelling" "-ID:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/wifi_provisioning" "-ID:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/wpa_supplicant" "-ID:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/xtensa-debug-module" "-ID:\Sloeber\/arduino [... cut ...]
D:\User\Maier\Documents\Eigene Dateien\Modellbau\Arduino\libraries\Nextion\NexRtc.cpp:72:1: error: control reaches end of non-void function [-Werror=return-type]
 }
 ^
cc1plus.exe: some warnings being treated as errors
libraries\Nextion\subdir.mk:182: recipe for target 'libraries\Nextion\NexRtc.cpp.o' failed
make: *** [libraries\Nextion\NexRtc.cpp.o] Error 1
"D:/Sloeber/arduinoPlugin/tools/make/make all" terminated with exit code 2. Build might be incomplete.

22:29:32 Build Failed. 6 errors, 3 warnings. (took 2s.172ms)

```

jantje, 2021-02-13:
These are build errors and should not be discarded without knowing what you are doing.
Are you sure this works in the arduino IDE? I would expect not.
If there is a "command line option to ignore this error" I would strongly advice not to use it.
But if you insist you could add it to project properties->arduino->compile options->append to C++

wm1962, 2021-02-13:
Yes, this works in the arduino IDE without any problems

uzi18, 2021-02-13:
Is it possible to paste here your project? Zipped?

uzi18, 2021-02-13:
Please paste here zipped project will try it

jantje, 2021-02-13:
> 
> 
> Yes, this works in the arduino IDE without any problems

Are you sure arduino IDE uses the same library,  the same board with the same platform version?
Did you look at the verbose compile output in Arduino IDE?

uzi18, 2021-02-13:
Arduino ide shows what libraries etc. Uses to compile project at beginning of log, please paste it

wm1962, 2021-02-13:
@uzi18 
> Arduino ide shows what libraries etc. Uses to compile project at beginning of log, please paste it

the log is too long, I will zip the project tomorrow. Thanks and good night so long.

wm1962, 2021-02-14:
Hi uzi18  and jantje,
here are the files. In "**Test_ESP32.zip**" I did remove the directory "**Release**" because the zipped file was to big, I suppose sloeber will generate it again.
In project  **properties->arduino->compile options->append to C++** i added **"-Wno-return-type"**, so the error-messages caused by [-Werror=return-type] are eliminated, but the are still errors.
In Arduino IDE are no errors.

[Test_ESP32.zip](https://github.com/Sloeber/arduino-eclipse-plugin/files/5977496/Test_ESP32.zip)
[libraries.zip](https://github.com/Sloeber/arduino-eclipse-plugin/files/5977503/libraries.zip)
[Messages ArduinoIDE.txt](https://github.com/Sloeber/arduino-eclipse-plugin/files/5977509/Messages.ArduinoIDE.txt)
[Messages Sloeber.txt](https://github.com/Sloeber/arduino-eclipse-plugin/files/5977510/Messages.Sloeber.txt)

wm1962, 2021-02-14:
Hi uzi18 and jantje,
here are the files. In "**Test_ESP32.zip**" I did remove the directory "**Release**" because the zipped file was to big, I suppose sloeber will generate it again.
In project **properties->arduino->compile options->append to C++** i added "**-Wno-return-type**", so the error-messages caused by [-Werror=return-type] are eliminated, but the are still errors.
In Arduino IDE are no errors.

[Test_ESP32.zip](https://github.com/Sloeber/arduino-eclipse-plugin/files/5977593/Test_ESP32.zip)
[libraries.zip](https://github.com/Sloeber/arduino-eclipse-plugin/files/5977594/libraries.zip)
[Messages ArduinoIDE.txt](https://github.com/Sloeber/arduino-eclipse-plugin/files/5977595/Messages.ArduinoIDE.txt)
[Messages Sloeber.txt](https://github.com/Sloeber/arduino-eclipse-plugin/files/5977596/Messages.Sloeber.txt)

jantje, 2021-02-14:
First of all: **always** tell which board you are using. There are some problems with ESP32 latest platform changes (not sure yet they have been released and based on the analysis of the command it seems fine) which should be fixed in Sloeber release 4.4 which is currently in testing

Apart from the esp32 issues expected to be fixed in the next release
Investigating the logs you have sent I see following differences in the NexRtc compile command
1)The warning level is set differently between Arduino IDE and Sloeber
2)The CPU frequency is set differently from arduino IDE and Sloeber
I found no other differences worth mentioning.

It seems your command line  (8951) is to longer than the windows allowed 8192 see #1210 for a workaround that seems to work on esp 32
I think if you enable all warnings in arduino ide (in the preferences settings) you will get the same errors. I guess by disabling show all warnings in project properties->arduino->compile options; the errors should be gone (I never use this and it needs to be reworked so it may no longer work , I'm allergic to warnings 😉 )

wm1962, 2021-02-14:
Hi jantje,
my board ist an ESP32
I have now done the following:

1. I did the workaround in #1210 
2. I changed my librarie-path to a short path `D:\Arduino\libraries`
But I still get errors like **undefined reference to `ESPTimeHelper::ESPTimeHelper()** (see the attached file).
[Messages Sloeber.txt](https://github.com/Sloeber/arduino-eclipse-plugin/files/5978171/Messages.Sloeber.txt)

I dont know where it comes from: from the too long command lines or from anywhere else.
In Arduino IDE there are definitiv no error messages.

jantje, 2021-02-14:
According to the Messages Sloeber.txt provided by jou the compile and link commands are ok.
The linker tells you there is some code missing 

```
: undefined reference to `ESPTimeHelper::ESPTimeHelper()'
: undefined reference to `RFC2047_Decoder::RFC2047_Decoder()'
 undefined reference to `RFC2047_Decoder::~RFC2047_Decoder()'
```

wm1962, 2021-02-14:
Yes, I know. 
But why does the linker tell that there is code missing.
When I build the same project with the Arduino IDE there no code missing and the project is well build?

jantje, 2021-02-15:
There are 2 possibilities
1) because of all the building problems somewhere there is a outputfile that is ... wrong. To fix delete the release folder and build again.
2)Something is different between arduino IDE and Sloeber

The great thing is that sloeber has far more possibilities to inspect the code. See if the code analyzer finds the methoods

wm1962, 2021-02-15:
Hi jantje
>1. because of all the building problems somewhere there is a outputfile that is ... wrong. To fix delete the release folder and build again.
2)Something is different between arduino IDE and Sloeber

I've done nr.1 several times. always the same errors, I dont have any ideas anymore.
Can you please build the project in your system with a.) Sloeber and b.) Arduino and tell me  which errors you get?

jantje, 2021-02-15:
>Can you please build the project in your system with a.) Sloeber and b.) Arduino and tell me which errors you get?

I will not try Arduino IDE but I can try Sloeber.
Assuming The code is provided here https://github.com/Sloeber/arduino-eclipse-plugin/issues/1297#issuecomment-778762541
I also need the json url and the platform name and the platform version

wm1962, 2021-02-15:
Please take these, it's the actual version
[Test_ESP32.zip](https://github.com/Sloeber/arduino-eclipse-plugin/files/5981938/Test_ESP32.zip)
[libraries.zip](https://github.com/Sloeber/arduino-eclipse-plugin/files/5981941/libraries.zip)
json url are: 
```
https://downloads.arduino.cc/packages/package_index.json
https://raw.githubusercontent.com/jantje/hardware/master/package_jantje_index.json
https://raw.githubusercontent.com/jantje/ArduinoLibraries/master/library_jantje_index.json
https://arduino.esp8266.com/stable/package_esp8266com_index.json
https://downloads.arduino.cc/libraries/library_index.json
https://dl.espressif.com/dl/package_esp32_index.json
```
[image]
[image]

wm1962, 2021-02-16:
Hi janjte,
have you got around to testing that? 
Many greetings 
Wilfried

jantje, 2021-02-16:
No, I'm focussing on a release right now.

wm1962, 2021-02-17:
okay, I will wait

wm1962, 2021-03-19:
Hi janjte,
I wanna aks you, if you have got around to testing that?
Many greetings
Wilfried

jantje, 2021-03-19:
I just tried it with V4.4 and it seems there is something off with the inclusion of libraries provided by the platform.
The examples are not excluded and that means compilation will never work.

jantje, 2021-03-19:
Seems the exclusion paths are not properly set in my project. Need to investigate that but I guess this is 4.4 regression
This caused some performance problems; so I fixed those

Your problem seems to be in the ESP-Mail library that has an extras folder in the src folder
Sloeber ignores extras folders in libraries (I guess Arduino IDE only does so in the root but who knows) if the exclusion rules are set properly (which wasn't in my case leading to other problems)

2 comments:
1) IMHO Bloody stupid idea to put source code in an folder named extras if you ask me.
2) @wm1962 you own me more than one as you could easily identify this stuff yourself

wm1962, 2021-03-20:
Hi jantje,
thanks for your tips. 
I removed the ESP-Mail library and the corresponding code and the build is well done.  :)

Many greetings
Wilfried

closed 2021-03-20 by wm1962