Sloeber/arduino-eclipse-plugin #1301: Beta 4.4: No such file or directory error ob Ubuntu 20.04

rmspacefish, 2021-02-15:
This problem occurs when I tested the new beta release of Ubuntu 20.04.
I was able to build an application in Sloeber, but I am not able to flash the board on Ubuntu because there is a "No such file or directory error" when launching avrdude. This was not an issue on Windows 10.

Here is a screenshot:
[image]

I ran the command displayed in the Sloeber console in the command line and that worked without issues..

jantje, 2021-02-15:
I see 2 possible reasons:
1) The installation of the avr platform failed to install avrdude
2) Sloeber point to a bad location

Is the path (with the .../Sloeber/Sloeber/... ) correct?
Can you dump the path of the installed avrdude?

jantje, 2021-02-15:
And thanks for testing :-)

rmspacefish, 2021-02-15:
The command works in the command line and I also checked the folder manually, avrdude is installed. If I do it in command line, I still have to run the Sloeber upload before that because it is also forcing the reset.

jantje, 2021-02-15:
After some thinking I assume this change causes the problem
https://github.com/Sloeber/arduino-eclipse-plugin/commit/d1edd474e437135809373c70e39f8ae4aeb52053#diff-8f40796928c32b6418568f97c8547fdc5c080c49410de1cd8297daf3d3ab29ad
Later on (after testing on windows) I deleted the whole lot. But maybe this is needed for Linux.
Note that this code is not used for the build but only for uploads. As you upload command contains quite a bit of quotes this may be the root cause.
Can you test this on your own?

jantje, 2021-02-15:
You can even see the double quotes in you last error.
So it must be that.
[image]

rmspacefish, 2021-02-15:
I can confirm adding back those lines solved the issue for me.

jantje, 2021-02-15:
great :-)

PR #1302 (2021-02-15): fix for unix systems
fixes #1301
closed 2021-02-15 by jantje