Sloeber/arduino-eclipse-plugin #1308: The type 'BearSSL::WiFiClientSecure' must implement the inherited pure virtual method 'Print::write'

OOHehir, 2021-02-27:
Hello,

I'm working on BasicHTTPSClient.ino [https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266HTTPClient/examples/BasicHttpsClient/BasicHttpsClient.ino](url). It compiles & runs successfully in Arduino but in eclipse I'm getting the error:

The type 'BearSSL::WiFiClientSecure' must implement the inherited pure virtual method 'Print::write' 

Have you seen this before? I attach the setting

Thanks,

[image]

jantje, 2021-02-27:
did you try to turn of warning in project properties->arduino->compile settings?

OOHehir, 2021-02-27:
Apologies for raising this, yes a code based issue. I thought it had something to do with the program not running correctly but it's running now.

Many thanks for all your work by the way, great to have it integrated with eclipse!

Leo

closed 2021-02-27 by OOHehir