Sloeber/arduino-eclipse-plugin #1314: USB CDC not compiling with Core 1.9.0

tyllurius, 2021-03-12:
Hi, my Sloeber (4.3.3.202012182318) with STM Core 1.9.0 doesn't compile projects (STM32F103C8, Maple Mini etc.) with USB CDC enabled.
I get a lot of errors like this:
_\Sloeber\arduinoPlugin\packages\STM32\hardware\stm32\1.9.0\cores\arduino\stm32\usb\usbd_desc.c: In function 'USBD_Class_ProductStrDescriptor':
<command-line>: error: 'MAPLEMINI_F103CB' undeclared (first use in this function); did you mean 'ARDUINO_MAPLEMINI_F103CB'?
E:\_Progs\Sloeber\arduinoPlugin\packages\STM32\hardware\stm32\1.9.0\cores\arduino/utils.h:7:26: note: in definition of macro 'CONCATS'
    7 | #define CONCATS(s1, s2) (s1" " s2)_
    
It seems is has something to do with the concatenation of board names:
_Sloeber\arduinoPlugin\packages\STM32\hardware\stm32\1.9.0\cores\arduino/utils.h:7:28: error: expected ')' before string constant
    7 | #define CONCATS(s1, s2) (s1" " s2)_
    
 With Core 1.7.0 and 1.8.0, I got USB CDC working with this workaround: [#1083](https://github.com/Sloeber/arduino-eclipse-plugin/issues/1083), but in Core 1.9.0 the platform.txt respectively sloeber.platform.txt changed. I digged into this issue and found some issues I guess they've something to do with it: 
 [F103 USB CDC won't build with Sloeber](https://github.com/stm32duino/Arduino_Core_STM32/issues/486)
 [ARDUINO_BOARD not defined correctly on compile command line](https://github.com/Sloeber/arduino-eclipse-plugin/issues/1047)
 [quoted strings from the command line not always working](https://github.com/Sloeber/arduino-eclipse-plugin/issues/1028)
 
But for a less experienced user this is beyond my knowledge and I don't know how to apply the workaround (as they seem to be for ESP32). As the Blue Pill and other STM boards are widely used I can't be only one with this problem, so any help how to solve this is highly appreciated!

jantje, 2021-03-12:
Have you tried with the Sloeber 4.4 beta?

tx0h, 2021-03-14:
i tried and i can compile/link with it (sloeber).

but how can i get it fixed in the eclipse plugin? or do i copy the whole arduinoPlugin tree over into the eclipse directory?

 ty for your work, btw.

jantje, 2021-03-14:
I have not yet made a update site for 4.4 beta so for now the only option is compile yourself as per the instructions on the first page here. That also makes an update site and allows for updating.

tyllurius, 2021-03-17:
Trying Sloeber 4.4 beta will be the next thing I'm going to try...I will report as soon as I got the beta running.

tyllurius, 2021-03-22:
Exact the same compile errors with 4.4 beta.

jantje, 2021-03-22:
I tried it here but I really have no clue what board you are referring to
Please provide the selected board from project properties->arduino
[image]

tyllurius, 2021-03-22:
It doesn't matter which board I choose, I get the error on every board.

The full error message:
E:\_Progs\Sloeber_4_4\arduinoPlugin\packages\STM32\hardware\stm32\1.9.0\cores\arduino\stm32\usb\usbd_desc.c: In function 'USBD_Class_ProductStrDescriptor':
<command-line>: error: 'BLUE_F407VE_Mini' undeclared (first use in this function); did you mean 'ARDUINO_BLUE_F407VE_Mini'?
E:\_Progs\Sloeber_4_4\arduinoPlugin\packages\STM32\hardware\stm32\1.9.0\cores\arduino/utils.h:7:26: note: in definition of macro 'CONCATS'
    7 | #define CONCATS(s1, s2) (s1" " s2)
      |                          ^~
E:\_Progs\Sloeber_4_4\arduinoPlugin\packages\STM32\hardware\stm32\1.9.0\cores\arduino\stm32\usb\usbd_desc.c:74:55: note: in expansion of macro 'BOARD_NAME'
   74 |   #define USBD_CLASS_PRODUCT_HS_STRING        CONCATS(BOARD_NAME, "CDC in HS Mode")
      |                                                       ^~~~~~~~~~
E:\_Progs\Sloeber_4_4\arduinoPlugin\packages\STM32\hardware\stm32\1.9.0\cores\arduino\stm32\usb\usbd_desc.c:225:31: note: in expansion of macro 'USBD_CLASS_PRODUCT_HS_STRING'
  225 |     USBD_GetString((uint8_t *)USBD_CLASS_PRODUCT_HS_STRING, USBD_StrDesc, length);
      |                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
<command-line>: note: each undeclared identifier is reported only once for each function it appears in
E:\_Progs\Sloeber_4_4\arduinoPlugin\packages\STM32\hardware\stm32\1.9.0\cores\arduino/utils.h:7:26: note: in definition of macro 'CONCATS'
    7 | #define CONCATS(s1, s2) (s1" " s2)
      |                          ^~
E:\_Progs\Sloeber_4_4\arduinoPlugin\packages\STM32\hardware\stm32\1.9.0\cores\arduino\stm32\usb\usbd_desc.c:74:55: note: in expansion of macro 'BOARD_NAME'
   74 |   #define USBD_CLASS_PRODUCT_HS_STRING        CONCATS(BOARD_NAME, "CDC in HS Mode")
      |                                                       ^~~~~~~~~~
E:\_Progs\Sloeber_4_4\arduinoPlugin\packages\STM32\hardware\stm32\1.9.0\cores\arduino\stm32\usb\usbd_desc.c:225:31: note: in expansion of macro 'USBD_CLASS_PRODUCT_HS_STRING'
  225 |     USBD_GetString((uint8_t *)USBD_CLASS_PRODUCT_HS_STRING, USBD_StrDesc, length);
      |                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
E:\_Progs\Sloeber_4_4\arduinoPlugin\packages\STM32\hardware\stm32\1.9.0\cores\arduino/utils.h:7:28: error: expected ')' before string constant
    7 | #define CONCATS(s1, s2) (s1" " s2)
      |                         ~  ^~~
E:\_Progs\Sloeber_4_4\arduinoPlugin\packages\STM32\hardware\stm32\1.9.0\cores\arduino\stm32\usb\usbd_desc.c:74:47: note: in expansion of macro 'CONCATS'
   74 |   #define USBD_CLASS_PRODUCT_HS_STRING        CONCATS(BOARD_NAME, "CDC in HS Mode")
      |                                               ^~~~~~~
E:\_Progs\Sloeber_4_4\arduinoPlugin\packages\STM32\hardware\stm32\1.9.0\cores\arduino\stm32\usb\usbd_desc.c:225:31: note: in expansion of macro 'USBD_CLASS_PRODUCT_HS_STRING'
  225 |     USBD_GetString((uint8_t *)USBD_CLASS_PRODUCT_HS_STRING, USBD_StrDesc, length);
      |                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
E:\_Progs\Sloeber_4_4\arduinoPlugin\packages\STM32\hardware\stm32\1.9.0\cores\arduino/utils.h:7:28: error: expected ')' before string constant
    7 | #define CONCATS(s1, s2) (s1" " s2)
      |                         ~  ^~~
E:\_Progs\Sloeber_4_4\arduinoPlugin\packages\STM32\hardware\stm32\1.9.0\cores\arduino\stm32\usb\usbd_desc.c:75:47: note: in expansion of macro 'CONCATS'
   75 |   #define USBD_CLASS_PRODUCT_FS_STRING        CONCATS(BOARD_NAME, "CDC in FS Mode")
      |                                               ^~~~~~~
E:\_Progs\Sloeber_4_4\arduinoPlugin\packages\STM32\hardware\stm32\1.9.0\cores\arduino\stm32\usb\usbd_desc.c:227:31: note: in expansion of macro 'USBD_CLASS_PRODUCT_FS_STRING'
  227 |     USBD_GetString((uint8_t *)USBD_CLASS_PRODUCT_FS_STRING, USBD_StrDesc, length);
      |                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
'Building file: E:\_Progs\Sloeber_4_4\arduinoPlugin\packages\STM32\hardware\stm32\1.9.0\cores\arduino\stm32\OpenAMP\virtio\virtio.c'
E:\_Progs\Sloeber_4_4\arduinoPlugin\packages\STM32\hardware\stm32\1.9.0\cores\arduino\stm32\usb\usbd_desc.c: In function 'USBD_Class_ConfigStrDescriptor':
<command-line>: error: 'BLUE_F407VE_Mini' undeclared (first use in this function); did you mean 'ARDUINO_BLUE_F407VE_Mini'?
E:\_Progs\Sloeber_4_4\arduinoPlugin\packages\STM32\hardware\stm32\1.9.0\cores\arduino/utils.h:7:26: note: in definition of macro 'CONCATS'
    7 | #define CONCATS(s1, s2) (s1" " s2)
      |                          ^~
E:\_Progs\Sloeber_4_4\arduinoPlugin\packages\STM32\hardware\stm32\1.9.0\cores\arduino\stm32\usb\usbd_desc.c:89:55: note: in expansion of macro 'BOARD_NAME'
   89 |   #define USBD_CLASS_CONFIGURATION_HS_STRING  CONCATS(BOARD_NAME, "CDC Config")
      |                                                       ^~~~~~~~~~
E:\_Progs\Sloeber_4_4\arduinoPlugin\packages\STM32\hardware\stm32\1.9.0\cores\arduino\stm32\usb\usbd_desc.c:271:31: note: in expansion of macro 'USBD_CLASS_CONFIGURATION_HS_STRING'
  271 |     USBD_GetString((uint8_t *)USBD_CLASS_CONFIGURATION_HS_STRING, USBD_StrDesc, length);
      |                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
E:\_Progs\Sloeber_4_4\arduinoPlugin\packages\STM32\hardware\stm32\1.9.0\cores\arduino/utils.h:7:28: error: expected ')' before string constant
    7 | #define CONCATS(s1, s2) (s1" " s2)
      |                         ~  ^~~
[... cut ...] \hardware\stm32\1.9.0\libraries\SPI\src" -I"E:\_Progs\Sloeber_4_4\arduinoPlugin\packages\STM32\hardware\stm32\1.9.0\libraries\Wire\src" -MMD -MP -MF"core\core\stm32\usb\usbd_if.c.d" -MT"core\core\stm32\usb\usbd_if.c.o" -D__IN_ECLIPSE__=1 "E:\_Progs\Sloeber_4_4\arduinoPlugin\packages\STM32\hardware\stm32\1.9.0\cores\arduino\stm32\usb\usbd_if.c"   -o "core\core\stm32\usb\usbd_if.c.o"
'Starting C compile'
core\core\stm32\usb\subdir.mk:64: recipe for target 'core\core\stm32\usb\usbd_desc.c.o' failed

jantje, 2021-03-22:
Strange
Works fine here
[image]

jantje, 2021-03-22:
Again Please provide the selected board from project properties->arduino
It seems like USBCON needs to be defined to get this issue and I don't see how to activate it

tyllurius, 2021-03-22:
Trying to get it works with Blue Pill:
[image]

jantje, 2021-03-22:
I do not have the USB support option 😕 
[image]

jantje, 2021-03-22:
Given the boards.txt it should be there.
Probably related to another platform installed on my system :-(

jantje, 2021-03-22:
I think if you change
-DBOARD_NAME="{build.board}"
in platform.txt to
"-DBOARD_NAME=\"{build.board}\""
save and delete platform.sloeber.txt
goto project properties->arduino apply/close
it should be fixed
But I can't test due to #1321

tyllurius, 2021-03-22:
Unfortunately not. Added the "" in platform.txt, deleted the old platform.sloeber.txt and checked the new one, the "" are added correctly, also restarted Sloeber, but the error exactly the same :(

jantje, 2021-03-23:
Unfortunately github removed the \'s from the comment.
Changed the string to code and now you see the correct values
Without the \'s it won't work

tyllurius, 2021-03-23:
Yes, it works!! Thank you Jantje, now I got latest STM core finally running with all features, great!

Issue solved, closed.

jantje, 2021-03-23:
Not really solved as the workaround is not yet implemented

ArminJo, 2021-04-15:
I have the same problem.
for BluePill from `GenF1.menu.pnum.BLUEPILL_F103CB.build.board=BLUEPILL_F103CB` in boards.txt 
  with `-DARDUINO_{build.board} -DARDUINO_ARCH_{build.arch} -DBOARD_NAME="{build.board}"`  from platform.txt
   results in `-DARDUINO_BLUEPILL_F103C8 -DBOARD_NAME="BLUEPILL_F103C8"` at the commandline/console.
If you do this in Arduino IDE you see in the commandline `-DARDUINO_BLUEPILL_F103C8 "-DBOARD_NAME=\"BLUEPILL_F103C8\"" `

Changing in **org.eclipse.cdt.core.prefs** the `A.BUILD.INFO.FLAGS/value=-D${A.BUILD.SERIES} -DARDUINO\=${A.RUNTIME.IDE.VERSION} -DARDUINO_${A.BUILD.BOARD} -DBOARD_NAME\="${A.BUILD.BOARD}"` value to `A.BUILD.INFO.FLAGS/value=-D${A.BUILD.SERIES} -DARDUINO\=${A.RUNTIME.IDE.VERSION} -DARDUINO_${A.BUILD.BOARD} -DBOARD_NAME\=\\"${A.BUILD.BOARD}\\"` will do the job.
Is this included in the 4.4 beta?

jantje, 2021-04-15:
It should be in the nightly :-)

closed 2021-04-17 by jantje