Sloeber/arduino-eclipse-plugin #1317: header file: No such file or directory

Roycinger, 2021-03-21:
Hi folks,
pretty new to Sloeber (not to programming in general, but not a pro).
I created a class to handle Button controls.
When I wanted to include that class (its header file respectively), I get a compiling error:
[image]
As you can see on the left, the header file (controller.h) is present in the root folder.
Do I have to place the header and cpp file somewhere else? Am I missing a point or doing it completely wrong?
Thanks for any advise!

jantje, 2021-03-21:
This should work. 

I always use " " instead of < >. I think you need to use " in this case but I'm no expert.
Check for weird characters in the include statement (basically retype it)

As controller.cpp compiled fine have a look there and copy paste the include from there

Note 1 you file midi_controller.cpp has been modified (see the * after it's name in the editor titlebar) so you may not be looking at what is being compiled
Note 2 The compiler complains in Sloeber.ino.cpp (which should be a copy but who knows) take a look there

Roycinger, 2021-03-21:
Thanks for your reply!
I had luck with importing it as a library. but I will check out your solution aswell to clarify!

closed 2021-06-04 by jantje

id1402, 2025-02-09:
Hello.
I use Arduino IDE since 5 years and yesterday decided to try Sloeber. I followed installation instructions for linux and downloaded sloeber-ide-V4.4.3-linux.gtk.x86_64.tar.gz. Extracting archive and linux specific commands for file rights worked well.
I start Sloeber:
```
cd /home/id1402/WCAL74578049/java/Sloeber
./sloeber-ide
```
Sloeber is running.
I made some new projects and imported ino-files from Arduino IDE sketches.

When I try to build one of my new projects **Arduino.h is missing**:
```
14:07:02 **** Incremental Build of configuration Release for project dmd2_anzeige_52 ****
make all 
Building file: ../sloeber.ino.cpp
Starting C++ compile
"/home/id1402/WCAL74578049/java/Sloeber/arduinoPlugin/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10812 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   -I"/home/id1402/WCAL74578049/java/Sloeber/arduinoPlugin/libraries/WiFi/1.2.7/src" -I"/home/id1402/WCAL74578049/software/sloeber-workspace/libraries/DMD2" -I"/home/id1402/WCAL74578049/software/sloeber-workspace/libraries/TimerThree" -I"/home/id1402/WCAL74578049/java/Sloeber/arduinoPlugin/packages/arduino/hardware/avr/1.8.6/libraries/SPI/src" -I"/home/id1402/WCAL74578049/java/Sloeber/arduinoPlugin/packages/arduino/hardware/avr/1.8.6/libraries/Wire/src" -I"/home/id1402/WCAL74578049/software/sloeber-workspace/libraries/ezTime/src" -I"/home/id1402/WCAL74578049/software/sloeber-workspace/libraries/Adafruit_Unified_Sensor" -I"/home/id1402/WCAL74578049/java/Sloeber/arduinoPlugin/packages/arduino/hardware/avr/1.8.6/libraries/EEPROM/src" -I"/home/id1402/WCAL74578049/java/Sloeber/arduinoPlugin/libraries/SD/1.3.0/src" -I"/home/id1402/WCAL74578049/software/sloeber-workspace/libraries/DHT_sensor_library" -I"/home/id1402/WCAL74578049/software/sloeber-workspace/libraries/IniFile/src" -I"/home/id1402/WCAL74578049/software/sloeber-workspace/libraries/Rtc_by_Makuna/src" -MMD -MP -MF"sloeber.ino.cpp.d" -MT"sloeber.ino.cpp.o" -D__IN_ECLIPSE__=1 "../sloeber.ino.cpp" -o "sloeber.ino.cpp.o"
../sloeber.ino.cpp:7:10: fatal error: Arduino.h: No such file or directory
 #include "Arduino.h"
          ^~~~~~~~~~~
compilation terminated.
make: *** [subdir.mk:20: sloeber.ino.cpp.o] Fehler 1
"make all" terminated with exit code 2. Build might be incomplete.

14:07:03 Build Failed. 2 errors, 0 warnings. (took 389ms)
```
Arduino.h is located here:
/home/id1402/WCAL74578049/java/Sloeber/arduinoPlugin/packages/arduino/hardware/avr/1.8.6/cores/arduino/Arduino.h
I checked the file. It looks ok. Why it is missing? In Arduino IDE the same files compile well.

Othe projects build without problems.
For example:
```
14:20:51 **** Build of configuration Release for project ultraschall ****
make all 
Building target: ultraschall
Printing size:
"/home/id1402/WCAL74578049/java/Sloeber/arduinoPlugin/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-size" -A "/home/id1402/WCAL74578049/software/sloeber-workspace/ultraschall/Release/ultraschall.elf"
/home/id1402/WCAL74578049/software/sloeber-workspace/ultraschall/Release/ultraschall.elf  :
section                     size      addr
.data                         34   8389120
.text                       5492         0
.bss                         256   8389154
.comment                      17         0
.note.gnu.avr.deviceinfo      64         0
.debug_aranges               232         0
.debug_info                13105         0
.debug_abbrev               4591         0
.debug_line                 4136         0
.debug_frame                 868         0
.debug_str                  2995         0
.debug_loc                  6024         0
.debug_ranges                368         0
Total                      38182

Finished building target: ultraschall
14:20:52 Build Finished. 0 errors, 0 warnings. (took 315ms)
```
 Why here is "Building target: ultraschall" and in the error-build is "Building file: ../sloeber.ino.cpp"?
Please help.

jantje, 2025-02-10:
@id1402 
It looks like the core and variant are no longer in your project.
This can mostly be fixed by going to project properties->sloeber->apply and close
If that didn't help provide a image of what the project looks like?

id1402, 2025-02-10:
Thank you! That worked.