Sloeber/arduino-eclipse-plugin #1318: STM32 build script generates filename typo ?

mikkeaa, 2021-03-21:
Build script generates a filename with typo:
arm-none-eabi-gcc: error: .\libraries\SrcWrapper\src\HAL\stm32yyxx_hl_smartcard.c.o: No such file or directory
makefile:96: recipe for target 'test_1_0.elf' failed
make: *** [test_1_0.elf] Error 1
"E:/eclipse_emb/eclipse/arduinoPlugin/tools/make/make all" terminated with exit code 2. Build might be incomplete.

It's supposed to be _hal_ not _hl_ ?
If I copy/rename manually the stm32yyxx_hal_smartcard.* -> stm32yyxx_hl_smartcard.* it works.

br. Mike

mikkeaa, 2021-03-21:
An addition: It get even weirder, now it's missing other letters:
arm-none-eabi-gcc: error: .\librries\SrcWrapper\src\HAL\stm32yyxx_hal_smartcard_ex.c.o: No such file or directory

librries->libraries
What might cause this ?

jantje, 2021-03-21:
I have never ever seen such a thing
So I have no clue what this is about
Which version of sloeber do you use?

mikkeaa, 2021-03-21:
Yes very, very weird, here's some versions:

Sloeber.io	Arduino eclipse plugin	4.3.3.202005042010	io.sloeber.core

Eclipse: Eclipse IDE for Embedded C/C++ Developers (includes Incubating components)
Version: 2021-03 (4.19.0)
Build id: 20210312-0638
OS: Windows 10, v.10.0, x86_64 / win32
Java version: 15.0.2

This is a very fresh install, and the sloeber via marketplace...
I'll probably remove all and reinstall as next step...

jantje, 2021-03-21:
As this is a "old" sloeber and this issue has never been reported something is off on your system
java15 may be it. the product comes with a java included on windows
Download the product from githuib see the releases here; it may be related to newer cdt10

mikkeaa, 2021-03-21:
Ok, now I installed the bundle from:
https://github.com/Sloeber/arduino-eclipse-plugin/releases/download/V4_3_3/V4.3.3_win64.2020-04-23_11-36-51.zip

-And same error and always on the smarcard whatever file:
.\libraries\SrcWrapper\src\HAL\stm32yyxx_hal_smartcad_ex.c.o: No such file or directory

Then I tried to update sloeber via eclipse update and now it died completely:

No editor descriptor for id org.eclipse.cdt.ui.editor.CEditor
java.lang.Exception
	at org.eclipse.ui.internal.EditorReference.createErrorPart(EditorReference.java:332)
	at org.eclipse.ui.internal.EditorReference.createPart(EditorReference.java:313)
	at org.eclipse.ui.internal.e4.compatibility.CompatibilityPart.createPart(CompatibilityPart.java:300)
	at org.eclipse.ui.internal.e4.compatibility.CompatibilityEditor.createPart(CompatibilityEditor.java:61)
	at org.eclipse.ui.internal.e4.compatibility.CompatibilityPart.create(CompatibilityPart.java:338)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at org.eclipse.e4.core.internal.di.MethodRequestor.execute(MethodRequestor.java:58)
	at org.eclipse.e4.core.internal.di.InjectorImpl.processAnnotated(InjectorImpl.java:998)
	at org.eclipse.e4.core.internal.di.InjectorImpl.processAnnotated(InjectorImpl.java:963)
	at org.eclipse.e4.core.internal.di.InjectorImpl.internalInject(InjectorImpl.java:139)
	at org.eclipse.e4.core.internal.di.InjectorImpl.internalMake(InjectorImpl.java:408)
	at org.eclipse.e4.core.internal.di.InjectorImpl.make(InjectorImpl.java:331)
	at org.eclipse.e4.core.contexts.ContextInjectionFactory.make(ContextInjectionFactory.java:202)
	at org.eclipse.e4.ui.internal.workbench.ReflectionContributionFactory.createFromBundle(ReflectionContributionFactory.java:91)
	at org.eclipse.e4.ui.internal.workbench.ReflectionContributionFactory.doCreate(ReflectionContributionFactory.java:60)
	at org.eclipse.e4.ui.internal.workbench.ReflectionContributionFactory.create(ReflectionContributionFactory.java:42)
	at org.eclipse.e4.ui.workbench.renderers.swt.ContributedPartRenderer.createWidget(ContributedPartRenderer.java:132)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.createWidget(PartRenderingEngine.java:1002)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.safeCreateGui(PartRenderingEngine.java:662)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.safeCreateGui(PartRenderingEngine.java:768)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.access$0(PartRenderingEngine.java:739)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$2.run(PartRenderingEngine.java:733)
	at org.eclipse.core.runtime.SafeRunner.run(SafeRunner.java:45)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.createGui(PartRenderingEngine.java:717)
	at org.eclipse.e4.ui.workbench.renderers.swt.StackRenderer.showTab(StackRenderer.java:1293)
	at org.eclipse.e4.ui.workbench.renderers.swt.LazyStackRenderer.postProcess(LazyStackRenderer.java:105)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.safeCreateGui(PartRenderingEngine.java:680)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.safeCreateGui(PartRenderingEngine.java:768)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.access$0(PartRenderingEngine.java:739)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$2.run(PartRenderingEngine.java:733)
	at org.eclipse.core.runtime.SafeRunner.run(SafeRunner.java:45)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.createGui(PartRenderingEngine.java:717)
	at org.eclipse.e4.ui.workbench.renderers.swt.SWTPartRenderer.processContents(SWTPartRenderer.java:73)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.safeCreateGui(PartRenderingEngine.java:676)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$1.run(PartRenderingEngine.java:547)
	at org.eclipse.core.runtime.SafeRunner.run(SafeRunner.java:45)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.createGui(PartRenderingEngine.java:531)
	at org.eclipse.e4.ui.workbench.renderers.swt.ElementReferenceRenderer.createWidget(ElementReferenceRenderer.java:73)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.createWidget(PartRenderingEngine.java:1002)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.safeCreateGui(PartRenderingEngine.java:662)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.safeCreateGui(PartRenderingEngine.java:768)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.access$0(PartRenderingEngine.java:739)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$2.run(PartRenderingEngine.java:733)
	at org.eclipse.core.runtime.SafeRunner.run(SafeRunner.java:45)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.createGui(PartRenderingEngine.java:717)
	at org.eclipse.e4.ui.workbench.renderers.swt.SWTPartRenderer.processContents(SWTPartRenderer.java:73)
	at org.eclipse.e4.ui.workbench.renderers.swt.SashRenderer.processContents(SashRenderer.java:140)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.safeCreateGui(PartRenderingEngine.java:676)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.safeCreateGui(PartRenderingEngine.java:768)
[... cut ...]
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:255)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:660)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:597)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1468)

jantje, 2021-03-22:
nightly needs java 10 or higher

tyllurius, 2021-03-22:
I can confirm the same type of errors, which are type of " .\libraries\SrcWrapper\src\HAL\stm32yyxx_somethinghere.c.o: No such file or directory.
I can confirm them on  Sloeber 4.3 and 4.4 beta on Windows 7, I am unable to compile STM core 1.9.0. With Core 1.7.0 works. On my Windows 10 machine I don't have that issues.

mikkeaa, 2021-03-22:
This is very, very sad because I would have needed eclipse debugging etc. There seems to be something tied up the the STM32 core pack for sure, and I really need the 1.9.0, I never tested the 1.7.0, but I have now reinstalled sloeber+eclipse at least 10 times without luck... enough ... So I'll have to skip the sloeber for now as non working sw for me. Would be interesting still to learn how the scripts are generating for the compiler underneath, that is probably where the problem is ?

tyllurius, 2021-03-22:
I'm really not an expert, but I have a slight guess that it has something to do with the archiver getting something wrong, or the combiner messing something up...they can't find the file although it's there. #1287 sounds somewhat similar to this issue.

jantje, 2021-03-22:
Thing is there changed quite a lot in the STM32 ESP32 area and sloeber needed serious updates to fix that (I even had to do a fix to CDT)
At the same time a java upgrade was needed (because of CDT that contained my fix)
And all these changes 4.4 are still in beta resulting in additional issues. like #1316 which will hit you with these platforms as they rely heavily on libraries
So yes ... it is not a great time to join sloeber when you want to use the newest STM32 and ESP32 platforms

tyllurius, 2021-03-27:
```
arm-none-eabi-gcc: error: .\libraries\SrcWrapper\src\HAL\stm32yxx_hal_rtc_ex.c.o: No such file or directory
makefile:108: recipe for target 'StarkerSchritt_v1_6.elf' failed
make: *** [StarkerSchritt_v1_6.elf] Error 1
```
At the combiner, it was stated:
```
'Starting combiner'
[...] .\libraries\SrcWrapper\src\HAL\stm32yyxx_hal_rtc_ex.c.o
```
Note the missing y.
On other computer (Win 10), same project, same Sloeber (beta 4.4), it works fine. Weird.

jantje, 2021-03-29:
>Weird

I fully agree on this one. The makefile that contains the missing chars is generated by io.sloeber.core/src/io/sloeber/core/toolchain/ArduinoGnuMakefileGenerator.java

Apart from adding "-" to the delete command, changing the names for the arduino pre/post.. commands and using replace instead of replaceall; nothing has changes since 2018

Here is a link comparing current master to somewhere in 2018
https://github.com/Sloeber/arduino-eclipse-plugin/compare/9c9e08ef0df..master
search for ArduinoGnuMakefileGenerator to see the changes

jantje, 2021-04-02:
I assumed the error was in the makefile and not during processing.
As similar issues popped up I now see that this issue may be a duplicate of #1210 which has a workaround
@mikkeaa  Please check

closed 2021-04-17 by jantje