Sloeber/arduino-eclipse-plugin #1320: Errors with index files (Beta 4.4)

tyllurius, 2021-03-22:
Since Beta 4.4, Sloeber reports an error when changing board properties or when running clean (but the operation itself seems to work). It says it can't delete (or change) some kind of index files. Perhaps the title of this issue could be changed into something more suitable.

When changing a board property I get this error:

And when running clean I get this error:

jantje, 2021-03-22:
I have not seen this before. I guess it is related to using one drive

tyllurius, 2021-03-22:
But I have this only on one of my PC. On the PC, which run Slober from same OneDrive workspace, don't raise this errors.

jantje, 2021-03-22:
The thing is ... Sloeber is not writing there. I assume it is the CDT indexer that writes there

tyllurius, 2021-03-23:
You were right, Jantje, it's a pure OneDrive related problem. It's caused by subfolders which are only online, but not yet downloaded on the actual machine (e.g. index files created on another PC/Eclipse session). So when Eclipse tries to read/write this files it throws errors.

It can be easily solved by making the folders available (OneDrive "Always keep on this device").

So this issue can be closed.

closed 2021-03-23 by tyllurius