Sloeber/arduino-eclipse-plugin #1322: ESP32 OTA not starting

hreintke, 2021-03-25:
Hi Jantje,

I have been some time away from programming but returned again.

Have an ESP32 project on which I want to use OTA. 
When starting the upload I get the error :  `tools.esptool_py.upload.network_.pattern : not found in the platform.txt file`
Notice the additional point in `network_.pattern`

There is a line in the platform.txt 
`tools.esptool_py.upload.network_pattern={network_cmd} -i "{serial.port}" -p "{network.port}" "--auth={network.password}" -f "{build.path}/{build.project_name}.bin"`

Also tried with adding an additional line in platform.txt with the "point" in but that dis not solve the issue.

I am currently running the "nightly fix" branch (created when there was the serial crashing issue.
But also did a quick test with current master which shows same behavior).

I changed PC's some time ago, on the old one I was running 4.3 from 11/8/20 (not 100% sure) which worked OK with OTA.

Have you seen this issue before ?

jantje, 2021-03-25:
No I haven't seen this but ....
Upload is not the best tested and OTA is even worse
As to https://github.com/Sloeber/arduino-eclipse-plugin/commit/65987154181935575136977411e6541970e93b5c
Thanks for searching
It may be related. The problem with the old code is that when network is not defined you get 2 DOTS
In both the old and new code there will always be a DOT before PATTERN which is what you see and don't want

As you can run from master .. can you try
```
        String networkPrefix = empty;
        if (isNetworkUpload()) {
            networkPrefix = Const.NETWORK_PREFIX;
        }
        String key = Const.A_TOOLS + upLoadTool + DOT + action + DOT + networkPrefix +  Const.PATTERN;
```

hreintke, 2021-03-26:
I go compilation errors on the `empty` `Const.NETWORK_PREFIX`  `Const.A_TOOLS` and `Const.PATTERN' 

Got it to work with : 
```
        String networkPrefix = EMPTY;
        if (isNetworkUpload()) {
            networkPrefix = NETWORK_PREFIX;
        }
        String key = TOOLS + upLoadTool + DOT + action + DOT + networkPrefix +  PATTERN;
```

`toolsesptool_py.upload.network_pattern : not found in the platform.txt file`

Can try with other, just tell me which one you want.

jantje, 2021-03-29:
```
        String networkPrefix = EMPTY;
        if (isNetworkUpload()) {
            networkPrefix = NETWORK_PREFIX;
        }
        String key = TOOLS + DOT+ upLoadTool + DOT + action + DOT + networkPrefix +  PATTERN;
```
Oeps sorry there is a dot missing

jantje, 2021-04-02:
This should be fixed now in the nightly (that is in about 6 hours if all goes well ;-) )

closed 2021-06-04 by jantje