Sloeber/arduino-eclipse-plugin #1323: MacOS Big Sur: "failed to create the java virtual machine"

SundialServices, 2021-03-25:
Downloaded to MacOS Big Sur (latest: 11.2.3), removed the quarantine attribute, and tried to start the application by clicking on the icon.  Received an alert, "failed to create the Java virtual machine."

The Java version is the very latest one:  Version 8 Update 281, just downloaded from "java.com."

There is no prior version of Eclipse on this system.

jantje, 2021-03-25:
Which Sloeber version?

wimjongman, 2021-03-25:
Sloeber latest does not work with java 8. You have to install java 11 or higher.

try an open source download: https://adoptopenjdk.net/

SundialServices, 2021-03-25:
The MacOS install page specifically says: "Sloeber *requires* Java **8**!  Other versions will not work!"

http://eclipse.baeyens.it/stable.php?OS=MacOS

If this is not the case, and if Sloeber needs something other than "what you get for MacOS from java.com," the instructions need to be updated accordingly.

Confirming now that installing OpenJDK-11 (LTS) from the above referenced site **did** resolve the problem.

jantje, 2021-03-25:
Strange Latest Sloeber release is 4.3 and should work fine on Java8
Nightly and 4.4 (currently in beta) must have java 10 or bigger

I agree on the documentation issue so It should not be closed

jantje, 2021-07-12:
updated web site

closed 2021-07-12 by jantje