Sloeber/arduino-eclipse-plugin #1324: Maple Uploader fails with Java Runtime PATH error

tyllurius, 2021-03-27:
When using the Maple bootloader with latest Sloeber following error occurs:
```
maple_loader v0.1
Resetting to bootloader via DTR pulse
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  EXCEPTION_ACCESS_VIOLATION (0xc0000005) at pc=0x000000007110b5db, pid=20276, tid=5780
#
# JRE version: Java(TM) SE Runtime Environment (11.0.5+10) (build 11.0.5+10-LTS)
# Java VM: Java HotSpot(TM) 64-Bit Server VM (11.0.5+10-LTS, mixed mode, tiered, compressed oops, g1 gc, windows-amd64)
# Problematic frame:
# C  [jSSC-2.8_x86_64.dll+0xb5db]
#
# No core dump will be written. Minidumps are not enabled by default on client versions of Windows
#
# An error report file with more information is saved as:
# E:\_Progs\Sloeber_4_4\arduinoPlugin\packages\STM32\tools\STM32Tools\1.4.0\tools\win\hs_err_pid20276.log
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#
```
This issue is not caused by Sloeber itself, it's because of using a Java version the Maple bootloader doesn't like. As the latest versions of Sloeber (nightly and beta 4.4) require using Java 11 users will run into this error when using the maple bootloader.

Clarification and workaround documented here: stm32duino/Arduino_Tools/issues/67

> To solve this problem the following line should be added to the maple_upload.bat:
> if exist "C:\Program Files (x86)\Arduino\java\bin" set PATH=C:\Program Files (x86)\Arduino\java\bin;%PATH%
> 
> This line temporarily changes the PATH during the execution of this batch file and ensures that the correct Java version is used.

jantje, 2021-03-29:
Thanks for the heads up 😄

jantje, 2021-07-23:
Closing this as it is not a sloeber issue and it is marked as a known limitation (which means it should be closed)

closed 2021-07-23 by jantje