Sloeber/arduino-eclipse-plugin #1325: Editor gone if install "Darkest  Dark Theme" from eclipse marketplace on V4.3.3

SturmWolfgang, 2021-03-30:
I like using dark mode - so I am working with the "Darkest Dark Theme" since I am using sloeber. It worked fine till today. As for preperation of the tests for 4.4 I downloaded newest version 4.3 from here: http://eclipse.baeyens.it/stable.php?OS=Windows

on a blank install,  pointing to my workspace, I installed the theme, after this every panels says: something like:
"No editor descriptor for id org.eclipse.cdt.ui.editor.CEditor"
for serial monitor " io.sloeber.ui.monitor.views.SerialMonitor"

**To Reproduce**
- use http://eclipse.baeyens.it/stable.php?OS=Windows
- install "Darkest Dark Theme" from Marketplace
- restart
=> error appears

restart does not help

[image]

SturmWolfgang, 2021-03-30:
the funny thing is: I have another sloeber 4.3 version, (lets say my productive version) where it works fine.
Same sloeber versions as in the picture on the previous post

SturmWolfgang, 2021-03-30:
on 4.4 the theme installed without problems

jantje, 2021-03-30:
I guess you are running in #1280 
Basically it is a compatibility issue between java CDT and Sloeber
With 4.4 and java 11 you are on the latest for everything and that is most likely to work.

closed 2021-06-04 by jantje