Sloeber/arduino-eclipse-plugin #1326: Release testing 4.4 test case 1. os:Win10 20H2

SturmWolfgang, 2021-03-30:
same problem as described in 
https://github.com/Sloeber/arduino-eclipse-plugin/issues/1325

After Update all Editors show:
[image]

The Arduino/Sloeber function is complete gone - no menu anymore:
[image]

To reproduce
- downloaded: http://eclipse.baeyens.it/stable.php?OS=Windows
- download ESP8266 plattform
- downloaded some libraries
- during start I pointed workspace to a existing 4.3 workspace,
- one project is open and it compilers fine
- Help --> Check for Updates
- A Version 4.3.3 update is shown, the date I can not remember exactly but it was 2020 not 2021
- Update successful  but error as described above
- Re Check for Updates results in error:
"no remedy found"+
[image]

update site it is: V4/stable

jantje, 2021-03-30:
Sloeber 4.3 comes with a jre folder containing java V8 and eclipse/sloeber prefer the java which is in the jre folder
you need to either remove the jre folder and install java 11 (or higher) or replace the jre folder content with java V11(or higher)
I have been using openjdk for java
https://adoptopenjdk.net/releases.html?variant=openjdk11&jvmVariant=openj9

jantje, 2021-05-23:
Closing assuming this was indeed a java version issue

closed 2021-05-23 by jantje