Sloeber/arduino-eclipse-plugin #1328: No such file while linking while the file is clearly there

Kashouryo, 2021-04-01:
**Describe the bug**
elf linker complains that a file is not there while the file is clearly exsits in the correct location

**To Reproduce**
Steps to reproduce the behavior:
1. Use ESP8266Audio and ESP8266SAM library
2. Open ESP8266SAM's example file
3. Click verify right away
4. See error

**Expected behavior**
Build successful

**Log**
**xtensa-lx106-elf-gcc: error: .\libraries\ESP8266Audio\src\libous\silk\resampler_private_IIR_FIR.c.o: No such file or directory**
```
"C:\\Software\\Sloeber\\arduinoPlugin\\tools\\make\\make" all 
'Starting combiner'
"C:\Software\Sloeber\arduinoPlugin\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506/bin/xtensa-lx106-elf-gcc" -CC -E -P -DVTABLES_IN_FLASH "C:\Software\Sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.7.4/tools/sdk/ld/eagle.app.v6.common.ld.h" -o "C:\Users\gigabyte\Documents\sloeber-workspace\Speak\Release/local.eagle.app.v6.common.ld"
"C:\Software\Sloeber\arduinoPlugin\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506/bin/xtensa-lx106-elf-gcc" -fexceptions -Wl,-Map "-Wl,C:/Users/gigabyte/Documents/sloeber-workspace/Speak\Release/Speak.map" -g  -Os -nostdlib -Wl,--no-check-sections -u app_entry -u _printf_float -u _scanf_float -Wl,-static "-LC:\Software\Sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.7.4/tools/sdk/lib" "-LC:\Software\Sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.7.4/tools/sdk/lib/NONOSDK22x_191122" "-LC:\Software\Sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.7.4/tools/sdk/ld" "-LC:\Software\Sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.7.4/tools/sdk/libc/xtensa-lx106-elf/lib" "-Teagle.flash.4m1m.ld" -Wl,--gc-sections -Wl,-wrap,system_restart_local -Wl,-wrap,spi_flash_read  -o "C:/Users/gigabyte/Documents/sloeber-workspace/Speak\Release/Speak.elf" -Wl,--start-group    .\sloeber.ino.cpp.o  .\libraries\SPI\SPI.cpp.o  .\libraries\SD\src\SD.cpp.o  .\libraries\SDFS\src\SDFS.cpp.o  .\libraries\ESP8266WiFi\src\BearSSLHelpers.cpp.o .\libraries\ESP8266WiFi\src\CertStoreBearSSL.cpp.o .\libraries\ESP8266WiFi\src\ESP8266WiFi.cpp.o .\libraries\ESP8266WiFi\src\ESP8266WiFiAP.cpp.o .\libraries\ESP8266WiFi\src\ESP8266WiFiGeneric.cpp.o .\libraries\ESP8266WiFi\src\ESP8266WiFiGratuitous.cpp.o .\libraries\ESP8266WiFi\src\ESP8266WiFiMulti.cpp.o .\libraries\ESP8266WiFi\src\ESP8266WiFiSTA-WPS.cpp.o .\libraries\ESP8266WiFi\src\ESP8266WiFiSTA.cpp.o .\libraries\ESP8266WiFi\src\ESP8266WiFiScan.cpp.o .\libraries\ESP8266WiFi\src\WiFiClient.cpp.o .\libraries\ESP8266WiFi\src\WiFiClientSecureAxTLS.cpp.o .\libraries\ESP8266WiFi\src\WiFiClientSecureBearSSL.cpp.o .\libraries\ESP8266WiFi\src\WiFiServer.cpp.o .\libraries\ESP8266WiFi\src\WiFiServerSecureAxTLS.cpp.o .\libraries\ESP8266WiFi\src\WiFiServerSecureBearSSL.cpp.o .\libraries\ESP8266WiFi\src\WiFiUdp.cpp.o  .\libraries\ESP8266SdFat\src\SpiDriver\SdSpiESP8266.cpp.o .\libraries\ESP8266SdFat\src\SpiDriver\SdSpiParticle.cpp.o .\libraries\ESP8266SdFat\src\SpiDriver\SdSpiSAM3X.cpp.o .\libraries\ESP8266SdFat\src\SpiDriver\SdSpiSTM32.cpp.o .\libraries\ESP8266SdFat\src\SpiDriver\SdSpiTeensy3.cpp.o  .\libraries\ESP8266SdFat\src\SdCard\SdSpiCard.cpp.o .\libraries\ESP8266SdFat\src\SdCard\SdSpiCardEX.cpp.o .\libraries\ESP8266SdFat\src\SdCard\SdioCardEX.cpp.o .\libraries\ESP8266SdFat\src\SdCard\SdioTeensy.cpp.o  .\libraries\ESP8266SdFat\src\FatLib\FatFile.cpp.o .\libraries\ESP8266SdFat\src\FatLib\FatFileLFN.cpp.o .\libraries\ESP8266SdFat\src\FatLib\FatFilePrint.cpp.o .\libraries\ESP8266SdFat\src\FatLib\FatFileSFN.cpp.o .\libraries\ESP8266SdFat\src\FatLib\FatVolume.cpp.o .\libraries\ESP8266SdFat\src\FatLib\FmtNumber.cpp.o .\libraries\ESP8266SdFat\src\FatLib\StdioStream.cpp.o .\libraries\ESP8266SdFat\src\FatLib\fstream.cpp.o .\libraries\ESP8266SdFat\src\FatLib\istream.cpp.o .\libraries\ESP8266SdFat\src\FatLib\ostream.cpp.o  .\libraries\ESP8266SdFat\src\MinimumSerial.cpp.o  .\libraries\ESP8266SAM\src\ESP8266SAM.cpp.o .\libraries\ESP8266SAM\src\debug.c.o .\libraries\ESP8266SAM\src\reciter.c.o .\libraries\ESP8266SAM\src\render.c.o .\libraries\ESP8266SAM\src\sam.c.o  .\libraries\ESP8266HTTPClient\src\ESP8266HTTPClient.cpp.o  .\libraries\ESP8266Audio\src\opusfile\info.c.o .\libraries\ESP8266Audio\src\opusfile\internal.c.o .\libraries\ESP8266Audio\src\opusfile\opusfile.c.o .\libraries\ESP8266Audio\src\opusfile\stream.c.o  .\libraries\ESP8266Audio\src\libopus\silk\fixed\LTP_analysis_filter_FIX.c.o .\libraries\ESP8266Audio\src\libopus\silk\fixed\LTP_scale_ctrl_FIX.c.o .\libraries\ESP8266Audio\src\libopus\silk\fixed\apply_sine_window_FIX.c.o .\libraries\ESP8266Audio\src\libopus\silk\fixed\autocorr_FIX.c.o .\libraries\ESP8266Audio\src\libopus\silk\fixed\burg_modified_FIX.c.o .\libraries\ESP8266Audio\src\libopus\silk\fixed\corrMatrix_FIX.c.o .\libraries\ESP8266Audio\src\libopus\silk\fixed\encode_frame_FIX.c.o .\libraries\ESP8266Audio\src\libopus\silk\fixed\find_LPC_FIX.c.o .\libraries\ESP8266Audio\src\libopus\silk\fixed\find_LTP_FIX.c.o .\libraries\ESP8266Audio\src\libopus\silk\fixed\find_pitch_lags_FIX.c.o .\libraries\ESP8266Audio\src\libopus\silk\fixed\find_pred_coefs_FIX.c.o .\libraries\ESP8266Audio\src\libopus\silk\fixed\k2a_FIX.c.o .\libraries\ESP8266Audio\src\libopus\silk\fixed\k2a_Q16_FIX.c.o .\libraries\ESP8266Audio\src\libopus\silk\fixed\noise_shape_analysis_FIX.c.o .\libraries\ESP8266Audio\src\libopus\silk\fixed\pitch_analysis_core_FIX.c.o .\libraries\ESP8266Audio\src\libopus\silk\fixed\process_gains_FIX.c.o .\libraries\ESP8266Audio\src\libopus\silk\fixed\regularize_correlations_FIX.c.o .\libraries\ESP8266Audio\src\libopus\silk\fixed\residual_energy16_FIX.c.o .\libraries\ESP8266Audio\src\libopus\silk\fixed\residual_energy_FIX.c.o .\libraries\ESP8266Audio\src\libopus\silk\fixed\schur64_FIX.c.o .\libraries\ESP8266Audio\src\libopus\silk\fixed\schur_FIX.c.o .\libraries\ESP8266Audio\src\libopus\silk\fixed\vector_ops_FIX.c.o .\libraries\ESP8266Audio\src\libopus\silk\fixed\warped_autocorrelation_FIX.c.o  .\libraries\ESP8266Audio\src\libopus\silk\A2NLSF.c.o .\libraries\ESP8266Audio\src\libopus\silk\CNG.c.o .\libraries\ESP8266Audio\src\libopus\silk\HP_variable_cutoff.c.o .\libraries\ESP8266Audio\src\libopus\silk\LPC_analysis_filter.c.o .\libraries\ESP8266Audio\src\libopus\silk\LPC_fit.c.o .\libraries\ESP8266Audio\src\libopus\silk\LPC_inv_pred_gain.c.o .\libraries\ESP8266Audio\src\libopus\silk\LP_variable_cutoff.c.o .\libraries\ESP8266Audio\src\libopus\silk\NLSF2A.c.o .\libraries\ESP8266Audio\src\libopus\silk\NLSF_VQ.c.o .\libraries\ESP8266Audio\src\libopus\silk\NLSF_VQ_weights_laroia.c.o .\libraries\ESP8266Audio\src\libopus\silk\NLSF_decode.c.o .\libraries\ESP8266Audio\src\libopus\silk\NLSF_del_dec_quant.c.o .\libraries\ESP8266Audio\src\libopus\silk\NLSF_encode.c.o .\libraries\ESP8266Audio\src\libopus\silk\NLSF_stabilize.c.o .\libraries\ESP8266Audio\src\libopus\silk\NLSF_unpack.c.o .\libraries\ESP8266Audio\src\libopus\silk\NSQ.c.o .\libraries\ESP8266Audio\src\libopus\silk\NSQ_del_dec.c.o .\libraries\ESP8266Audio\src\libopus\silk\PLC.c.o .\libraries\ESP8266Audio\src\libopus\silk\VAD.c.o .\libraries\ESP8266Audio\src\libopus\silk\VQ_WMat_EC.c.o .\libraries\ESP8266Audio\src\libopus\silk\ana_filt_bank_1.c.o .\libraries\ESP8266Audio\src\libopus\silk\biquad_alt.c.o .\libraries\ESP8266Audio\src\libopus\silk\bwexpander.c.o .\libraries\ESP8266Audio\src\libopus\silk\bwexpander_32.c.o .\libraries\ESP8266Audio\src\libopus\silk\check_control_input.c.o .\libraries\ESP8266Audio\src\libopus\silk\code_signs.c.o .\libraries\ESP8266Audio\src\libopus\silk\control_SNR.c.o .\libraries\ESP8266Audio\src\libopus\silk\control_audio_bandwidth.c.o .\libraries\ESP8266Audio\src\libopus\silk\control_codec.c.o .\libraries\ESP8266Audio\src\libopus\silk\debug.c.o .\libraries\ESP8266Audio\src\libopus\silk\dec_API.c.o .\libraries\ESP8266Audio\src\libopus\silk\decode_core.c.o .\libraries\ESP8266Audio\src\libopus\silk\decode_frame.c.o .\libraries\ESP8266Audio\src\libopus\silk\decode_indices.c.o .\libraries\ESP8266Audio\src\libopus\silk\decode_parameters.c.o .\libraries\ESP8266Audio\src\libopus\silk\decode_pitch.c.o .\libraries\ESP8266Audio\src\libopus\silk\decode_pulses.c.o .\libraries\ESP8266Audio\src\libopus\silk\decoder_set_fs.c.o .\libraries\ESP8266Audio\src\libopus\silk\enc_API.c.o .\libraries\ESP8266Audio\src\libopus\silk\encode_indices.c.o .\libraries\ESP8266Audio\src\libopus\silk\encode_pulses.c.o .\libraries\ESP8266Audio\src\libopus\silk\gain_quant.c.o .\libraries\ESP8266Audio\src\libopus\silk\init_decoder.c.o .\libraries\ESP8266Audio\s [... cut ...] espnow -lsmartconfig -lairkiss -lwpa2 -lstdc++-exc -lm -lc -lgcc -Wl,--end-group  "-LC:/Users/gigabyte/Documents/sloeber-workspace/Speak\Release"
xtensa-lx106-elf-gcc: error: .\libraries\ESP8266Audio\src\libous\silk\resampler_private_IIR_FIR.c.o: No such file or directory
makefile:104: recipe for target 'Speak.elf' failed
make: *** [Speak.elf] Error 1
```

**Desktop (please complete the following information):**
 - OS: Windows 7 64bit pro
 - Version Nightly and 4.3 stable

**Additional context**

jantje, 2021-04-01:
Wierd
This instantly reminded me #1318
Checking it looks like the same/similar issue but I don't think it is. I currently think this is issue #1210 

the command contains
.\libraries\ESP8266Audio\src\**libopus**\silk\resampler_private_IIR_FIR.c.o
while the error states
.\libraries\ESP8266Audio\src\**libous**\silk\resampler_private_IIR_FIR.c.o
Note the missing p in the error output

Given your description I would assume that in this case libopus is the right name and as such Sloeber generated the correct command
Which makes this different from #1318

tyllurius, 2021-04-01:
That's exactly the error I have on my Win7 64bit pro PC; I referred it to #1318. Still I don't understand what is the difference to #1210?

jantje, 2021-04-01:
#1318 sloeber makes a wrong command
#1210 windows does something with a to long command

Your command is way to long for windows so you need the workaround of #1210

Kashouryo, 2021-04-02:
Big success! Thank you guys! Btw, how to do parallel make in sloeber?(make -j8)
```
'Finished building target: Speak'
' '

10:23:14 Build Finished. 0 errors, 0 warnings. (took 25s.39ms)
```

closed 2021-04-02 by Kashouryo

tyllurius, 2021-04-02:
Big sucess here, too! It also was issue #1210 in my case.