Sloeber/arduino-eclipse-plugin #1330: why two include here?

youxiaojie, 2021-04-08:
```
#ifdef __IN_ECLIPSE__
//This is a automatic generated file
//Please do not modify this file
//If you touch this file your change will be overwritten during the next build
//This file has been generated on 2021-04-08 05:38:07

#include "Arduino.h"
#include "Arduino.h"

void setup() ;
void loop() ;

#include "test1.ino"

#endif

```

jantje, 2021-04-08:
Because
`#include "Arduino.h"`
is always added.
And if it is found to be included in the sketch it is added again
And it is completely harmless because Arduino.h must have include guards

youxiaojie, 2021-04-08:
sure, in my past experience include some file two times will lead to some problems eg, double definition, some warnings.
Description	Resource	Path	Location	Type
Unused declaration of function 'loop'	sloeber.ino.cpp	/test1	line 11	Code Analysis Problem
Description	Resource	Path	Location	Type
Unused declaration of function 'setup'	sloeber.ino.cpp	/test1	line 10	Code Analysis Problem

what is this file's function? sloeber.ino.cpp?

jantje, 2021-04-08:
>sure, in my past experience include some file two times will lead to some problems eg, double definition, some warnings.

See https://riptutorial.com/cplusplus/example/3525/include-guards

>what is this file's function? sloeber.ino.cpp?

A Arduino ino file does not follow C/C++ compilation requirements.
In other words if you change a .ino file in a .cpp file you will not have sloeber.ino.cpp and also no working project

Sloeber supports .ino files but we advice to not use them unless needed. They may be needed because you are writing libraries and want to add examples or because you want arduino ide compatibility
But if you can get away without .ino files sloeber will be faster (no need to generate sloeber.ino.cpp) and more CDT like (which I feel will make things easier)
However you will have to make your own function declarations and all the other stuff .ino files hide.
But if you like that ... IMHO you should stick to arduino.ide

youxiaojie, 2021-04-08:
ut if you can get away without .ino files sloeber will be faster (no need to generate sloeber.ino.cpp) and more CDT like (which I feel will make things easier)
However you will have to make your own function declarations and all the other stuff .ino files hide.
But if you like that ... IMHO you should stick to arduino.ide

that is to say to creat a c/c++ merely? and do left thing by hand?

closed 2021-06-04 by jantje