Sloeber/arduino-eclipse-plugin #1331: Can't find private libraries

judge2005, 2021-04-11:
The 'Import Arduino Libraries' dialog no longer lists any libraries in my private directory for any of my projects. I have no idea when this stopped working.

I deleted the path from the 'Private Library Path' section of the 'Arduino' preferences dialog and re-added it, but it doesn't help. I checked the 'Automatically import libraries based on includes', that doesn't help. I restarted eclipse - doesn't help. I moved the path to the top of the list - doesn't help.

 - OS: Windows 10
 - Eclipse:
Eclipse IDE for C/C++ Developers

Version: Photon Release (4.8.0)
Build id: 20180619-1200

Sloeber:
  Sloeber, the Eclipse Arduino IDE (plugin version)	4.3.1.201810101819	io.sloeber.feature.feature.group	Sloeber.io

jantje, 2021-04-11:
as this is a really old Sloeber version it must be some config error or some incompatibility with a newer plugin

judge2005, 2021-04-12:
Does 4.3.3 work with Eclipse Photon 4.8.0?

jantje, 2021-04-12:
I honestly don't have a clue.

judge2005, 2021-04-13:
OK. What version of Eclipse was it built for?

judge2005, 2021-05-22:
It was the presence of a couple of .cpp and/or .h files in the library folder. e.g. if the library folder is mylibs, which has this structure:
```
mylibs
    \lib1
    \lib2
    afile.cpp
    afile.h
```

closed 2021-05-22 by judge2005

jantje, 2021-05-22:
That may be the root cause". However that should mean mylibs is available as library.