Sloeber/arduino-eclipse-plugin #1332: Serial Monitor prints single characters only in pairs

dwittenb, 2021-04-13:
I am using Sloeber Version 4.3.3 and an Arduino Uno. On writing single characters to the Serial Port the Serial Monitor in Sloeber/Eclipse shows single characters only in pairs. This is my Example:

#include "Arduino.h"
void setup()
{
  Serial.begin(9600, SERIAL_8N1);
}

void loop()
{
  while (Serial.available() > 0)
  {
    char Zeichen = Serial.read();
   	Serial.print(Zeichen);
  }
}

The baudrate in the Serial Monitor is set to 9600 and the checkbox of DTR/DSR is unset.

On sending a single character with no CR/LF the first character is not shown in the Serial Monitor.  On sending the next Character the first an the next character is printet on the Serial Monitor. It seems that the Monitor is waiting that at least for two characters to be sended or to be recieved.
If I am using a external Terminal-Program, each charachter is looped back right away.

dwittenb, 2021-04-13:
```
#include "Arduino.h"
void setup()
{
Serial.begin(9600, SERIAL_8N1);
}

void loop()
{
while (Serial.available() > 0)
{
char Zeichen = Serial.read();
Serial.print(Zeichen);
}
}
```

dwittenb, 2021-04-13:
OS: WIN10

jantje, 2021-04-13:
That might be related to the fact Sloeber supports the scope.
Do you have scope filtering on?

dwittenb, 2021-04-14:
Thank you, thats was the problem. In the Serial Monitor the Button "Hide Plotter Data" was pressed. I unpressed it and everything works fine.

closed 2021-04-14 by dwittenb