Sloeber/arduino-eclipse-plugin #1333: Release testing 4.4 test case 4.3 -> 4.4 Win10

ArminJo, 2021-04-15:
AVR projects which were closed in 4.3 and opened under 4.4 loose the` Use alternative size command`? check, i.e. we have `Config.Release.compile.sloeber.size.switch=false`.
If this cannot be fixed, it should be noted in the Upgrade_Instructions.txt , so that the user must not seach for the reason like I did, but can edit the .sproject file, which is much faster, especially for multi configuration projects.

P.s. can the old 4.3 org.eclipse.cdt.core.prefs files be removed (or stripped)

jantje, 2021-04-15:
>projects which were closed in 4.3 and opened under 4.4 loose the Use alternative size command? check, i.e. we have Config.Release.compile.sloeber.size.switch=false

Thanks for the report. I'll investigate

> can the old 4.3 org.eclipse.cdt.core.prefs files be removed (or stripped)

I'm not sure which files you mean. Are this workspace files, project files or install files?

ArminJo, 2021-04-15:
> I'm not sure which files you mean. Are this workspace files, project files or install files?

the <project root>/.settings/org.eclipse.cdt.core.pref where the old board settings were located.

jantje, 2021-04-15:
>the /.settings/org.eclipse.cdt.core.pref where the old board settings were located.

So you are talking about the project preference settings.
If you only have set Sloeber values they should be empty and gone
If you have set other CDT preferences they should still be in these files

ArminJo, 2021-04-15:
Thanks for the reply, mostly I have no other than Sloeber preferenced set in an project, but since I know, I can check for it 😀 .

jantje, 2021-04-17:
>projects which were closed in 4.3 and opened under 4.4 loose the Use alternative size command? check, i.e. we have Config.Release.compile.sloeber.size.switch=false

I checked and if the original value of "show all warnings" was true it is still true after upgrade.
I even have a unit test testing this

however I see that the flag is not taken into account properly

ArminJo, 2021-04-17:
Where are nightly builds to test?
I can not compile it for myself now.

jantje, 2021-04-17:
The nightly is updated every night.
Should be available from the nightly tomorrow.
I hope to finish #1335 tonight as well

ArminJo, 2021-04-17:
But where can I download the nightly? I my be a bit of a fool, but I did not found it yet.

jantje, 2021-04-17:
http://eclipse.baeyens.it/nightly.php
add this site to your update sites in windows->preferences->update sites
and run updates

ArminJo, 2021-04-18:
After updating right now I get:
[image]
I use working sets and cannot access any files in these anymore.
But even in the plain project view no project file is shown 😢 .
[image]

jantje, 2021-04-18:
Can you zip and provide me your workspace as is before upgrade so I can try?

ArminJo, 2021-04-18:
It is here.
https://www.dropbox.com/sh/gbzu4ow2jndyzme/AACzyi1CBwybfdfF_ktU9rIPa?dl=0
But 1.3 GByte and still uploading.

jantje, 2021-04-18:
Thinking about this a bit more I think I know what the problem is.
I assume that you have partially migrated that is you already have a .sproject and that .sproject contains warning_level=true
With the new nightly this will cause problems as it has to be NONE ALL MORE or CUSTOM
I assume that the other problems are consequences of sloeber "crashing" on this error.
I think it is a good idea to make sloeber mlore resiliant for badly formatted .sproject files and sloeber.cfg file

jantje, 2021-04-18:
The dropbox map is still empty
I made a fix for .sproject containing a wrong value like true or false

ArminJo, 2021-04-18:
> The dropbox map is still empty

Sorry, I was not at home and the upload was interrupted.

> I made a fix for .sproject containing a wrong value like true or false

THANKS! I will test it tomorrow

jantje, 2021-05-23:
I think we can close this issue. I tested and the warning levels (like in Arduino IDE) are now selectable and custom and they work fine on my system
Please reopen if not fixed

closed 2021-05-23 by jantje