Sloeber/arduino-eclipse-plugin #1334: Release testing 4.4 test case .sprojects Win10

ArminJo, 2021-04-15:
After changing the Arduino Board Selection parameter I have this contents in .sproject.
It is **a merge of digispark-tiny values** and other old values from a **former instance** where I edited a Blue Pill STM32F103C8 and an ESP8266 board.
no multi configuration project, but it also happens there.

Config.Release.board.BOARD.ID=digispark-tiny
Config.Release.board.BOARD.MENU.CpuFrequency=80
Config.Release.board.BOARD.MENU.Debug=Disabled
Config.Release.board.BOARD.MENU.DebugLevel=None____
Config.Release.board.BOARD.MENU.FlashErase=none
Config.Release.board.BOARD.MENU.FlashSize=4M1M **-> ESP8266 **
Config.Release.board.BOARD.MENU.LwIPVariant=v2mss536 **-> ESP8266 * 👎 
Config.Release.board.BOARD.MENU.UploadSpeed=921600
Config.Release.board.BOARD.MENU.VTable=flash
Config.Release.board.BOARD.MENU.burnmode=upgrade
Config.Release.board.BOARD.MENU.clock=clock165
Config.Release.board.BOARD.MENU.cpu_speed=speed_72mhz
Config.Release.board.BOARD.MENU.device_variant=STM32F103C8
Config.Release.board.BOARD.MENU.micronucleus=default **->Digispark** 👍 
Config.Release.board.BOARD.MENU.opt=osstd
Config.Release.board.BOARD.MENU.pnum=**BLUEPILL_F103C8** 👎 
Config.Release.board.BOARD.MENU.rtlib=nano
Config.Release.board.BOARD.MENU.upload_method=dfuoMethod
Config.Release.board.BOARD.MENU.usb=CDCgen
Config.Release.board.BOARD.MENU.xserial=generic
Config.Release.board.BOARD.MENU.xusb=FS
Config.Release.board.BOARD.TXT=${SLOEBER_HOME}/arduinoPlugin/packages/digistump/hardware/avr/1.7.3/boards.txt
Config.Release.board.PROGRAMMER.NAME=Default

The same happens for an ESP8266 board d1_mini (micronucleus is missing here even when I edited a digispark before.)
Config._UnitTest.board.BOARD.ID=d1_mini
Config._UnitTest.board.BOARD.MENU.CpuFrequency=80
Config._UnitTest.board.BOARD.MENU.Debug=Disabled
Config._UnitTest.board.BOARD.MENU.DebugLevel=None____
Config._UnitTest.board.BOARD.MENU.FlashErase=none
Config._UnitTest.board.BOARD.MENU.FlashSize=4M1M
Config._UnitTest.board.BOARD.MENU.LwIPVariant=v2mss536
Config._UnitTest.board.BOARD.MENU.UploadSpeed=921600
Config._UnitTest.board.BOARD.MENU.**VTable**=flash **->ESP** 👍 
Config._UnitTest.board.BOARD.MENU.baud=921600
Config._UnitTest.board.BOARD.MENU.cpu_speed=speed_72mhz
Config._UnitTest.board.BOARD.MENU.dbg=Disabled
Config._UnitTest.board.BOARD.MENU.device_variant=**STM32F103C8**
Config._UnitTest.board.BOARD.MENU.eesz=4M3M
Config._UnitTest.board.BOARD.MENU.exception=disabled
Config._UnitTest.board.BOARD.MENU.ip=lm2n
Config._UnitTest.board.BOARD.MENU.lvl=None____
Config._UnitTest.board.BOARD.MENU.opt=osstd
Config._UnitTest.board.BOARD.MENU.pnum=**BLUEPILL_F103C8** 👎 
Config._UnitTest.board.BOARD.MENU.rtlib=nano
Config._UnitTest.board.BOARD.MENU.ssl=all
Config._UnitTest.board.BOARD.MENU.upload_method=dfuoMethod
Config._UnitTest.board.BOARD.MENU.usb=CDCgen
Config._UnitTest.board.BOARD.MENU.vt=flash
Config._UnitTest.board.BOARD.MENU.wipe=none
Config._UnitTest.board.BOARD.MENU.xserial=generic
Config._UnitTest.board.BOARD.MENU.xtal=80
Config._UnitTest.board.BOARD.MENU.xusb=FS
Config._UnitTest.board.BOARD.TXT=${SLOEBER_HOME}/arduinoPlugin/packages/esp8266/hardware/esp8266/2.7.4/boards.txt
Config._UnitTest.board.PROGRAMMER.NAME=Default

jantje, 2021-04-15:
can you provide the platform json url and screenshots of the project properties->arduino settings

ArminJo, 2021-04-15:
These are my  platform json URL's:
```
https://dl.espressif.com/dl/package_esp32_index.json
http://drazzy.com/package_drazzy.com_index.json
http://digistump.com/package_digistump_index.json
https://github.com/stm32duino/BoardManagerFiles/raw/master/STM32/package_stm_index.json
https://raw.githubusercontent.com/sparkfun/Arduino_Apollo3/master/package_sparkfun_apollo3_index.json
https://files.pololu.com/arduino/package_pololu_index.json
https://raw.githubusercontent.com/ArminJo/DigistumpArduino/master/package_digistump_index.json
http://dan.drown.org/stm32duino/package_STM32duino_index.json
http://raw.githubusercontent.com/MHEtLive/arduino-boards-index/master/package_mhetlive_index.json
https://sandeepmistry.github.io/arduino-nRF5/package_nRF5_boards_index.json
https://mcudude.github.io/MiniCore/package_MCUdude_MiniCore_index.json
https://raw.githubusercontent.com/xukangmin/TinyCore/master/avr/package/package_tinycore_index.json
https://mcudude.github.io/MegaCore/package_MCUdude_MegaCore_index.json
https://raw.githubusercontent.com/xukangmin/TinyCore/master/avr/package/package_tinycore_index.json
```
The digispark settings
[image]
And the merged in Bluepill and ESP8266 settings
[image]
[image]

And thanks for maintaining this helpful tool!!!

jantje, 2021-06-27:
I tried to reproduce this with the latest version and I found following issues
1) Your json urls contains 
```
http://digistump.com/package_digistump_index.json
https://raw.githubusercontent.com/ArminJo/DigistumpArduino/master/package_digistump_index.json
```
That is 2 times package_digistump_index.json which means one will not be accessible. This will work but it is not a good practice.
Your example uses  https://raw.githubusercontent.com/ArminJo/DigistumpArduino/master/package_digistump_index.json

2) FYI sseems the json utrl for bluepill is no longer valid https://github.com/stm32duino/BoardManagerFiles/raw/master/STM32/package_stm_index.json
And the new one https://github.com/stm32duino/BoardManagerFiles/raw/master/package_stmicroelectronics_index.json seems to only contain 2.0
 (it goes fast I know :-( )

So I tried to reproduce your case and could not because of the delay on my behalf.
On the other hand I did a lot of work in this area and I have not seen this issue pop up.
So if you have this issue with the nightly (hoping to release a latest beta soon) can you provide easy reproducible steps?

ArminJo, 2021-06-28:
I will try to reproduce it with the nightly, but it will not before mid of July and I hope I am not successful 😀

jantje, 2021-06-28:
I hope 4.4 is out before mid July :-)

jantje, 2021-09-10:
I guess I can close this now

closed 2021-09-10 by jantje