Sloeber/arduino-eclipse-plugin #1337: STM32F103 crashes when using Wire.h

DrHock, 2021-04-17:
i am working Sloeber Beryllium 4.3.3 for a few month and every worked perfect in my projects. 
I am just busy porting my code to several platforms. Arduino Nano, Arduino Mega 2560, Arduino Due, ESP32, STM32F103C8, with good result.

After the attempt to integrate a I2C device (DS3231) all plattforms work as expected, but STM32 crashes. I have not tested any other I2C Application before on STM32.

i traced the problem down to the simple example "i2c_scanner_wire.ino" which comes with
http://dan.drown.org/stm32duino/package_STM32duino_index.json

the same example compiled in Arduino IDE leads to working code on STM32.

**To Reproduce**
compile Example "i2c_scanner_wire.ino" with Wire.h Library selected.
Download to STM32F103C8  board

**observations**
der code seem to crash after the first loop of the I2C scan with the call to wire.Endtransmision()
you can see a kind of fast blinking of the LED.

Additional information.
I tried several STM32F103Cx Boards but they all behave the same.
I am guessing the issue here:  https://github.com/Sloeber/arduino-eclipse-plugin/issues/1005  or http://stm32duinoforum.com/forum/viewtopic.php?t=3941 
might address a similar problem. Unfortunatelly i am not able to compile the plugin source code myself. so and hoping this will be fixed in the next bundle.

jantje, 2021-04-17:
provide the name of the board and the platform version you are using.
Provide the json url and a screenshot of project properties->arduino

DrHock, 2021-04-17:
thanks for immediate answer.   
this is the URL 
http://dan.drown.org/stm32duino/package_STM32duino_index.json
I tried Version 2020.12.26  and 2021.3.18  with the same result.
in Arduino IDE the same URL and both Versions are working.
here is the screenshot.
[image]

DrHock, 2021-04-17:
Sorry. I am using the bluepill board..

jantje, 2021-04-17:
I assume you get a warning when you start telling you something about the path.
So it might be #1210 related
and you may need Sloeber 4.4 as there have been STM32 changes but I don't know whether these are for 
`http://dan.drown.org/stm32duino/package_STM32duino_index.json`
or
`https://raw.githubusercontent.com/stm32duino/BoardManagerFiles/master/STM32/package_stm_index.json`

DrHock, 2021-04-17:
No. There is no error or warning message. Neither with my project nor with the reported sample.nevertheless I try to move closer to the root to exclude risk of the reported issue. 
 I would be happy to test it with version 4.4. Is a win64 (beta) bundle already available?

jantje, 2021-04-17:
!!!Do not forget to take backups of your installs and workspaces!!!!
The 4.4 bundle is here https://github.com/Sloeber/arduino-eclipse-plugin/releases/tag/V4_4_0-beta2
You will still need to apply the workaround for #1210 
You may also try to upgrade to the nightly which does contain a fix for #1210 (more risky)
!!!Do not forget to take backups of your installs and workspaces!!!!

DrHock, 2021-04-18:
thnak you for the link to the beta
I could try the beta I downloaded 11a.m. today.
unfortunatly still the same behaviour.  here are my findings:
- we need a JAVA VM11. the newest JRE (281) is not enough.
[image]

- another Win10 machine was used
-      machine                  | sketch | result | size     || sketch2 | result
- +----------------------+---------------------------------------------
- Machine 1 Arduino IDE || I2C | OK  |  18116B   ||  blink.ino | OK
- Machine 1 Sloeber 4.3  || I2C | fail  | 18184B     || blink.ino | OK
- Machine 1 Sloeber 4.4  || I2C | fail  | 15324B     || blink.ino | OK
- Machine 2 Sloeber 4.4 || I2C  | fail | 15324B     || blink.ino | OK

if I can do anything to support, just let me know.

jantje, 2021-04-18:
yes you need java 11 or higher

jantje, 2021-04-18:
check https://openjdk.java.net/ to download

DrHock, 2021-04-19:
Thx,  I know.  this was just for information, Non-Java Programmers will face the issue.
the result of my findings are in the table. summary: 
blink.ino builds with all toolchains OK and runs on the bluepill as expected
the unchanged example "i2c_scanner_wire.ino" only on Arduino IDE.
The resulting binary is different; at least the size which is given in the table as well.
I spent several hour to compare the Command lines of Sloeber and Arduino but could not find anything significantly different besides some "-I" switches outside the quote and missing quotes.

jantje, 2021-04-19:
To be able to understand why I need command line differences
Maybe the STM32 people can help out

DrHock, 2021-04-19:
don't get me wrong. that's not my question and I don't mis use you to be my manual for gcc

fact: there is an issue why Arduino IDE does work with the the same code base with simple original example and Sloeber does not. in V4.3 as well as V4.4
I invested many hours on this and just wanted let you know my findings. that's all.
whether you put e.g. "-IC:/sloeber/file1.c" in the command line or -I"C:/Sloeber/file1.c" might or might not be of interest. I am just saying this is the only difference I found which might be of semantical interest.

I don't want to bother you anymore. thx.

jantje, 2021-04-19:
don't get me wrong 😄 
I like things fixed and working properly 👍 

However I do not own a STM32 so after "it builds" and "the commands look the same" there is little I can do

The only "real difference" I know of between Arduino IDE and Sloeber build commands is when making the Archive. Sloeber does it in one call. Arduino IDE does is source file by source file.

closed 2024-02-20 by jantje