Sloeber/arduino-eclipse-plugin #1338: Unable to compile when workspace is in Windows user directory

stscde, 2021-04-20:
**Describe the bug**
Compilation of blink sketch failes when workspace is in Windows user directory.
- Workspace directory "C:\Users\Stefan\Documents\sloeber-workspace" => build of blink sketch fails
- Workspace directory "C:\dev\sloeber-workspace" => build of blink sketch is successful

```
16:07:37 **** Incremental Build of configuration Release for project M1 ****
"C:\\dev\\Sloeber\\arduinoPlugin\\tools\\make\\make" all 
'Building file: ..\sloeber.ino.cpp'
'Starting C++ compile'
"C:\dev\Sloeber\arduinoPlugin\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\dev\Sloeber\/arduinoPlugin/packages/esp8266/hardware/esp8266/2.7.4/tools/sdk/include" "-IC:\dev\Sloeber\/arduinoPlugin/packages/esp8266/hardware/esp8266/2.7.4/tools/sdk/lwip2/include" "-IC:\dev\Sloeber\/arduinoPlugin/packages/esp8266/hardware/esp8266/2.7.4/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\Stefan\Documents\sloeber-workspace\M1\Release/core" -c -Wall -Wextra -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++11 -MMD -ffunction-sections -fdata-sections -fno-exceptions -DNEW_OOM_ABORT  -DNONOSDK22x_190703=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=1460 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DDEBUG_ESP_PORT=Serial  -DARDUINO=10812 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_NODEMCU\"" -DLED_BUILTIN=2 -DFLASHMODE_DIO  -DESP8266   -I"C:\dev\Sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266" -I"C:\dev\Sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.7.4\variants\nodemcu" -MMD -MP -MF"sloeber.ino.cpp.d" -MT"sloeber.ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "..\sloeber.ino.cpp"   -o "sloeber.ino.cpp.o"
In file included from ..\sloeber.ino.cpp:13:0:
..\Blink.ino:37:1: fatal error: opening dependency file sloeber.ino.cpp.d: Permission denied
 }
 ^
compilation terminated.
subdir.mk:18: recipe for target 'sloeber.ino.cpp.o' failed
make: *** [sloeber.ino.cpp.o] Error 1
"C:/dev/Sloeber/arduinoPlugin/tools/make/make all" terminated with exit code 2. Build might be incomplete.

16:07:38 Build Failed. 3 errors, 0 warnings. (took 1s.300ms)
```

**To Reproduce**
Use "V4.3.3_win64.2020-04-23_11-36-51.zip" from Sloeber download page.

1. Use a Windows system with a non admin user
2. Create a workspace in the users home directory (was suggested for me by default)
3. Configure a new sketch with template "blink" for nodemcu 2.7.4
4. Click the build button

**Expected behavior**
Project should compile.

**Screenshots**

[image]

[image]

jantje, 2021-04-20:
Strange
>fatal error: opening dependency file sloeber.ino.cpp.d: Permission denied

As this file should be in the Release folder; it looks like you do not have write access to the Release folder
Which is wierd because the make files should be there as well
Is there a read only sloeber.ino.cpp.d file?
Can you check the content of the Release folder and the permissions of the Release folder?

stscde, 2021-04-23:
Hi jantje, thank you for your response, sorry for getting back late. Lot of work to do ...

I tried to reproduce the issue today on another PC but was not able to do so. Workspace folder there is "C:\Users\Stefan.LAPTOP-VP1A7M22\Documents\sloeber-workspace" and compilation works fine.

You asked me for files in the release folder. I have access to the PC with the issue tomorrow and will investigate.

OT: Meanwhile I spend some time developing with sloeber and its really great. Much faster compilation time and a the comfort I am used to from Eclipse. I really like it and would like to say a big thanks to you for creating and supporting it!

stscde, 2021-05-07:
Hi jantje,

I finally got access to the PC with the problem again. I checked the release folder and it does not contain the sloeber.ino.cpp.d file. See the following screenshot for the contents.

[image]

If I create an empty file "sloeber.ino.cpp.d" there manually then I get the same error for "sloeber.ino.cpp.o". If I create this file then I have missing "core" files.

Meanwhile I also deactivated my AV scanner. No difference, same issue.

As this is only happening on one PC/workspace and the community does not seem to have this issue too - feel free close this issue. If you like to investigate further I am there to assist. I can also offer a TeamViewer/Teams/... session.

Best wishes and a nice weekend
Stefan

jantje, 2021-05-14:
Thanks for the update.
It is really wierd. Looking at this again I would take a look at the content of blink.ino and sloeber.ino.cpp. Maybe that will provide a clue.
Then delete the release folder and build again
run the make from the command line

I have however very little time I can spend on Sloeber right now. So a remote session won't work.

wimjongman, 2021-05-14:
You installed or ran as administrator. This will create the files as administrator leaving you without access for the normal user. 

Delete the workspace and try again, or use file/switch workspace to create a new workspace. Never install or run Sloeber as administrator. It will bring nothing but problems.

stscde, 2021-05-20:
@jantje I found the reason for the behavior. It is that the "Documents" Folder is on the list of Windows Protected Folders. 
When using Windows Defender these folders are protected by Windows Defender.
When using Bitdefender (as I do) these folders are protected by Bitdefender. When I turn of the protection in Bitdefender compilation works instantly. See the following screenshot.
[image]

See also:
https://docs.microsoft.com/en-us/microsoft-365/security/defender-endpoint/controlled-folders?view=o365-worldwide

As Windows is so eager to protect the "Documents" folder and I do not want to reduce my security settings I decided not to use the "Documents" folder for my workspace anymore. Usually I would have not used this folder initially If it would not have been the default suggestion :)

@wimjongman Thank you for your feedback, I used the ZIP "V4.3.3_win64.2020-04-23_11-36-51.zip" for "installation" and did not ran Sloeber with elevated priviledges.

Best wishes
Stefan

closed 2021-05-20 by stscde