Sloeber/arduino-eclipse-plugin #1339: Failed to parse json file + Please wait for the 'arduino installer job' to finish

opc241, 2021-04-21:
**Describe the bug**
I am trying to get ready with arduino on eclipse but when i hit "upload" the example code to the esp8266 it says:
"Please wait for the 'arduino installer job' to finish. Look at right bottom of the eclipse window to see if it is still running."

error log says: 
Failed to parse json file "C:\eclipse\arduinoPlugin\package_jantje_index.json"

java.lang.reflect.InaccessibleObjectException: Unable to make field private final java.lang.String java.io.File.path accessible: module java.base does not "opens java.io" to unnamed module @4fd74814
	at java.base/java.lang.reflect.AccessibleObject.checkCanSetAccessible(AccessibleObject.java:357)
	at java.base/java.lang.reflect.AccessibleObject.checkCanSetAccessible(AccessibleObject.java:297)
	at java.base/java.lang.reflect.Field.checkCanSetAccessible(Field.java:177)
	at java.base/java.lang.reflect.Field.setAccessible(Field.java:171)
	at com.google.gson.internal.reflect.UnsafeReflectionAccessor.makeAccessible(UnsafeReflectionAccessor.java:44)
	at com.google.gson.internal.bind.ReflectiveTypeAdapterFactory.getBoundFields(ReflectiveTypeAdapterFactory.java:159)
	at com.google.gson.internal.bind.ReflectiveTypeAdapterFactory.create(ReflectiveTypeAdapterFactory.java:102)
	at com.google.gson.Gson.getAdapter(Gson.java:458)
	at com.google.gson.internal.bind.ReflectiveTypeAdapterFactory.createBoundField(ReflectiveTypeAdapterFactory.java:117)
	at com.google.gson.internal.bind.ReflectiveTypeAdapterFactory.getBoundFields(ReflectiveTypeAdapterFactory.java:166)
	at com.google.gson.internal.bind.ReflectiveTypeAdapterFactory.create(ReflectiveTypeAdapterFactory.java:102)
	at com.google.gson.Gson.getAdapter(Gson.java:458)
	at com.google.gson.Gson.fromJson(Gson.java:931)
	at com.google.gson.Gson.fromJson(Gson.java:870)
	at io.sloeber.core.api.PackageManager.loadPackage(PackageManager.java:685)
	at io.sloeber.core.api.PackageManager.loadJson(PackageManager.java:673)
	at io.sloeber.core.api.PackageManager.loadJsons(PackageManager.java:645)
	at io.sloeber.core.api.PackageManager.setJsonURLs(PackageManager.java:847)
	at io.sloeber.ui.preferences.ThirdPartyHardwareSelectionPage.performOk(ThirdPartyHardwareSelectionPage.java:42)
	at org.eclipse.jface.preference.PreferenceDialog$7.run(PreferenceDialog.java:905)
	at org.eclipse.core.runtime.SafeRunner.run(SafeRunner.java:45)
	at org.eclipse.jface.util.SafeRunnable.run(SafeRunnable.java:174)
	at org.eclipse.jface.preference.PreferenceDialog.okPressed(PreferenceDialog.java:889)
	at org.eclipse.ui.internal.dialogs.FilteredPreferenceDialog.okPressed(FilteredPreferenceDialog.java:461)
	at org.eclipse.jface.preference.PreferenceDialog.buttonPressed(PreferenceDialog.java:233)
	at org.eclipse.jface.dialogs.Dialog.lambda$0(Dialog.java:619)
	at org.eclipse.swt.events.SelectionListener$1.widgetSelected(SelectionListener.java:84)
	at org.eclipse.swt.widgets.TypedListener.handleEvent(TypedListener.java:252)
	at org.eclipse.swt.widgets.EventTable.sendEvent(EventTable.java:89)
	at org.eclipse.swt.widgets.Display.sendEvent(Display.java:4209)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1043)
	at org.eclipse.swt.widgets.Display.runDeferredEvents(Display.java:4026)
	at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:3626)
	at org.eclipse.jface.window.Window.runEventLoop(Window.java:823)
	at org.eclipse.jface.window.Window.open(Window.java:799)
	at org.eclipse.ui.internal.OpenPreferencesAction.run(OpenPreferencesAction.java:66)
	at org.eclipse.jface.action.Action.runWithEvent(Action.java:474)
	at org.eclipse.jface.action.ActionContributionItem.handleWidgetSelection(ActionContributionItem.java:580)
	at org.eclipse.jface.action.ActionContributionItem.lambda$4(ActionContributionItem.java:414)
	at org.eclipse.swt.widgets.EventTable.sendEvent(EventTable.java:89)
	at org.eclipse.swt.widgets.Display.sendEvent(Display.java:4209)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1043)
	at org.eclipse.swt.widgets.Display.runDeferredEvents(Display.java:4026)
	at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:3626)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$5.run(PartRenderingEngine.java:1157)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:338)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.run(PartRenderingEngine.java:1046)
	at org.eclipse.e4.ui.internal.workbench.E4Workbench.createAndRunUI(E4Workbench.java:155)
	at org.eclipse.ui.internal.Workbench.lambda$3(Workbench.java:644)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:338)
	at org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Workbench.java:551)
	at org.eclipse.ui.PlatformUI.createAndRunWorkbench(PlatformUI.java:156)
	at org.eclipse.ui.internal.ide.application.IDEApplication.start(IDEApplication.java:152)
	at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:203)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:134)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:104)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:401)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:255)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:78)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:567)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:653)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:590)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1461)

provide the name of the board and the platform version you are using.  
Provide the json url and a screenshot of project properties->arduino  

**Desktop (please complete the following information):**
 - OS: Windows 10 

[image]

jantje, 2021-04-21:
I guess something went wrong when downloading the file https://raw.githubusercontent.com/jantje/hardware/master/package_jantje_index.json
and the downloaded file C:\eclipse\arduinoPlugin\package_jantje_index.json on your system is not what it should be
Which caused the install job not to finish
Which causes the problem
Restarting should fix the problem

Don't close this issue as this exception should be caught and handled properly

opc241, 2021-04-22:
So i restartet the IDE and now there are additional packages failing like that: 
[image]

The path fot the packages the ide is searching for is at: C:\eclipse\arduinoPlugin\
So i copied the following files to this path and restarted the ide: 
1) 
C:\eclipse\arduinoPlugin\packages\esp8266\hardware\esp8266\2.7.4\package_esp8266com_index.json -----------> C:\eclipse\arduinoPlugin\package_esp8266com_index.json

2) C:\eclipse\arduinoPlugin\packages\arduino\hardware\avr\1.8.3\package_index.json ------------> C:\eclipse\arduinoPlugin\package_index.json

3) https://raw.githubusercontent.com/jantje/hardware/master/package_jantje_index.json -------------> C:\eclipse\arduinoPlugin\package_jantje_index.json

I can´t find file nr. 3 in the directory and had to download it from your source. 
After rebooting eclipse, i could observe the files getting deleted when the error(s) occured as in the screenshot above.

jantje, 2021-04-22:
I guess looking at the content of the files will give you a clue why they are corrupt.
If the files are deleted then that mean sloeber is set to refresh these files look in windows->preferences->arduino to disable the update of json files

opc241, 2021-04-22:
I did a online json validator on all 3 files, but the format of the json files are valid. 
I also disabled the update of json files, but the error still occurs.

jantje, 2021-04-22:
Did you upgrade your java version recently?
Which version of sloeber are you using?
https://stackoverflow.com/questions/41265266/how-to-solve-inaccessibleobjectexception-unable-to-make-member-accessible-m#41265267

SigP226, 2021-04-24:
Hello,

Same kind of issue. 
- Windows 7
- Eclipse 2021-03 (installed with zip, not with the installer which doesn't work, I don't know why...)
- CDT C++ IDE
- GNU MCU arm
- darkest theme
- Sloeber 4.3.3
- Java 16 (C:/program files/Java/jdk-16.0.1)

installed Apr 24th

After downloading Sloeber (plugin), it returns Java error immediately, I tried too ad the json link for STM32duino and same. Here is a screen copy.
I check the stackoverflow link, but don't understand what I have to modify and where. Do I need to add --illegal-access somewhere in the files or in the eclipse.ini?

[image]

SigP226, 2021-04-24:
Hello again, 

my problem is solved by downgrade Java in eclipse.ini file and using the JRE version 15

Now it doesn't compile properly, 
F:\Eclipse\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.0.0\cores\arduino/variant.h:15:10: error: 
#include expects "FILENAME" or < FILENAME >
   15 | #include VARIANT_H

but this is another story :)

jantje, 2021-04-24:
Great thanks. Now we know java 16 or higher is a no go for now.
The json parser will need to be replaced before we can use java 16 or higher.
Do you use Oracle Java or another? (There may be a difference)

opc241, 2021-04-24:
I could confirm. I downgraded to java 16, reinstalled the plugin and all errors were gone. Thanks!

SigP226, 2021-04-26:
Open JDK
build 15.0.2

jantje, 2021-04-26:
I reopen this issue as people do not search for closed issues

jantje, 2021-07-23:
Closing this as it is a known limitation and known limitations should be closed

jantje, 2021-09-14:
Reopening as we will need to find a fix given the changes in java management in eclipse
Seems this is how to fix this issue though I do not understand it.
https://github.com/google/gson/issues/1540#issuecomment-641642692

wimjongman, 2021-09-14:
to workaround this issue, add the following line to your eclipse.ini or sloeber.ini
--illegal-access=permit

closed 2021-12-05 by jantje

Antonio-Raposo, 2021-12-11:
Hi. I just switched to Java 17 and here we go again... It crashes even with the workaround given at the beginning.  So I leave here the workaround for Java 17 until a new version of Sloeber is out. Add the following line to sloeber.ini or eclipse.ini:
 `--add-opens=java.base/java.io=ALL-UNNAMED`

jantje, 2021-12-12:
Can you test with the latest nightly?
https://github.com/Sloeber/arduino-eclipse-plugin/actions/runs/1554243582
It should be fixed there but lots of changes so I would like a decent test.

Antonio-Raposo, 2021-12-12:
Hello. The nightly eclipse plugin works with the workaround removed. I had only time to test my old faithful serial blinker and it compiled ok, uploaded ok and then blinked and was able to write and read the serial port. So far no issues.

jantje, 2021-12-12:
Thanks for the feedback
As this change involved coding the json reading I couldn't withstand changing the classes which caused lots of improvements and as such change.
But this reflection issue in Sloeber should be fixed now 😄

Fubbel, 2022-01-02:
I use Opensuse 15.2 with OpenJDK 11.
In Eclipse 2021-12 is the same error.
The workaround
--add-opens=java.base/java.io=ALL-UNNAMED
in eclipse.ini is ok, all other is not ok.
Update to the nightly build (Sloeber, the Eclipse Arduino IDE (plugin version) 4.4.1.202201020122) is ok - without the workaround.

The product Sloeber 4.4.1.202201020122 ends with error:
An error occurred while installing the items
session context was:(profile=epp.package.cpp, phase=org.eclipse.equinox.internal.p2.engine.phases.Install, operand=null --> [R]io.sloeber.product.executable.gtk.linux.x86_64 4.4.1.202201020122, action=org.eclipse.equinox.internal.p2.touchpoint.natives.actions.ChmodAction).
The action chmod failed - file /opt/eclipse.2021.12/sloeber-ide does not exist

where Eclipse is installed in /opt/eclipse.2021.12 ans SLOEBER_HOME is ~/
you use an fixed path for the ide - no good idea... eclipse-path is read-only....

jantje, 2022-01-02:
>I use Opensuse 15.2 with OpenJDK 11.
Update to the nightly build (Sloeber, the Eclipse Arduino IDE (plugin version) 4.4.1.202201020122) is ok - without the workaround.

Thanks for the confirmation

wimjongman, 2022-01-03:
> --add-opens=java.base/java.io=ALL-UNNAMED

Should we add this to the ini?

jantje, 2022-01-03:
> --add-opens=java.base/java.io=ALL-UNNAMED

I don't think so as I implemented json readers for all json files.
I uses Java 17 for debugging and I have no issues
As to this post it is not needed.  https://github.com/Antonio-Raposo