Sloeber/arduino-eclipse-plugin #1340: Compilation issue with Core STM32 2.0.0 - makefile generation

SigP226, 2021-04-25:
Hello,

environment : Win7 + eclipse 2021-03 + sloeber + JRE15 + gnu MCU.
Test with the blink example
generation of the project (arduino sketch).

[image]

Add library srcWrapper.
Start the compilation
compilation issue : 
[image]

I have found what I believe to be the preliminary root cause : quote not at the right place which cause the compiler to not use the include file to define VARIANT_H.
extract hereunder: look difference between 1st and 2nd line, the quote after the I

This concern only the 3 last line of include of the command line.
"-IF:\Eclipse\arduinoPlugin\packages\STMicroelectronics\tools\CMSIS\5.7.0/CMSIS/DSP/PrivateInclude"   
-I"F:\Eclipse\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.0.0\cores\arduino" 
-I"F:\Eclipse\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.0.0\variants\STM32F1xx\F103C8T_F103CB(T-U)" 
-I"F:\Eclipse\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.0.0\libraries\SrcWrapper\src" 

Is it an issue coming from sloeber or from Eclipse?

SigP226, 2021-04-27:
Hello,
I forget to precise : Sloeber 4.3.3

First progress : the error come from quotes "" or bracket angle <> missing in the definition of the environment variable VARIANT_H
I have corrected like this, by adding manually the bracket. [image]

Of course this is a manual workaround. If you change the board type in your project, this will be lost and you have to change again manually.

Now all file compile BUT, with some error after compilation (combiner). Let's continue analysis to understand why.

jantje, 2021-04-27:
There are quite a lot of fixes for the newer versions of STM32 in the nightly
This is one of the reasons I'm trying to make a release

SigP226, 2021-05-01:
Issue closed. 
Update of Sloeber done with a nightly build on May 1st. Compilation works fine. 
Thanks a lot !

closed 2021-05-01 by SigP226

hreintke, 2021-05-05:
Hi Jantje,
I am also hitting this issue.
Tried both the nightly and git master but for me it doesn't solve.
Used an existing project moving from STM32 v1.9.0 to V2.0.0 and a new project directly on v2.0.0. Same result.

The manual workaround from @SigP226 makes it work.

Is there some additional setting I need to apply ?

jantje, 2021-05-05:
@hreintke 
If you have the problem with the nightly please create a new issue.
FYI I'm pretty busy right now and I have little time left to spend on SLOEBER