Sloeber/arduino-eclipse-plugin #1341: Sloeber toolbar and menu disappear if I install subclipse

judge2005, 2021-04-26:
Sloeber toolbar and menu disappear if I install subclipse

1. Download and unzip the windows zip from here: https://github.com/Sloeber/arduino-eclipse-plugin/releases/tag/V4_3_3
2. Run it and let it initialize (windows asks me if I want to trust it).
3. Select 'Install new software...' and add this repo: https://subclipse.github.io/updates/
4. Select everything but the maven support and install it.
5. Sloeber restarts and all the sloeber menu and toolbar items are gone.

jantje, 2021-04-26:
Are you still in arduino perspective?

judge2005, 2021-04-27:
Yes

jantje, 2021-04-27:
I assume you are on Sloeber 4.3 and the update updated CDT to CDT10
Which cause hot issue #1263

judge2005, 2021-04-27:
K. I’ll try and build a nightly...

jantje, 2021-05-23:
Closing due to no response
Please reopen if not fixed

closed 2021-05-23 by jantje