Sloeber/arduino-eclipse-plugin #1343: Release testing 4.4 test case 4.3 -> 4.4 Win10 (Use Alternative size command not reacting on Apply button, only Apply and Close works)

ArminJo, 2021-04-27:
I checked Use Alternative size command, clicked Apply, switched tab to board selection and back and voila, **the box is unchecked again** 😞

jantje, 2021-04-27:
Actually .... in my book, you should not even have to press apply before switching, the value should still be there.
Thanks for testing :-)

ArminJo, 2021-04-27:
Thanks for developing this great tool 🥇 
I considered to switch my huge workspace to 4.4 so I made a few last test, but I think it is better to wait a bit.

jantje, 2021-04-27:
>I considered to switch my huge workspace to 4.4 so I made a few last test, but I think it is better to wait a bit.

Thanks for doing the tests. And indeed waiting is better. There may be incompatibilities between beta releases.

jantje, 2021-05-23:
This one seems fixed :-)

closed 2021-05-23 by jantje