Sloeber/arduino-eclipse-plugin #1344: Release testing 4.4 test case 4.3 -> 4.4 Win10 (makefile: missing separator)

ArminJo, 2021-04-27:
I converted the example project I used for the [documentation](https://github.com/Sloeber/arduino-eclipse-plugin/discussions/1342) to 4.4 following the rules:
```
Upgrade to a new Sloeber version
Copy files from the old to the new Sloeber directory before running it the first time.
1. All prefs files from Sloeber/configuration/settings/
2. jargon-wordlist.txt
3. The complete arduinoPlugin directory
```
then I get the error:
```
07:43:01 **** Incremental Build of configuration BlinkWithAnalogIn for project Blink_test ****
"C:\\Sloeber\\arduinoPlugin\\tools\\make\\make" all 
makefile:73: *** missing separator.  Stop.
"C:/Sloeber/arduinoPlugin/tools/make/make all" terminated with exit code 2. Build might be incomplete.
```
In the makefile I see the lines
```
# Add inputs and outputs from these tool invocations to the build variables 
AR += \
C:\sloeber-workspace\Blink_test\BlinkWithAnalogIn\ \
```
and
```
C:\sloeber-workspace\Blink_test\BlinkWithAnalogIn\: $(AR_OBJ)
	@echo 'Starting archiver'
	"C:\Sloeber\arduinoPlugin\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7/bin/avr-gcc-ar" rcs     C:\sloeber-workspace\Blink_test\BlinkWithAnalogIn/ $(AR_OBJ)
```

In my other (errorless) workspace I see the lines with an **arduino.ar** after the backslash.
`E:\WORKSPACE_SLOEBER_44\IRremote_Leonardo\_ReceiveDemo\arduino.ar \`
and
```
E:\WORKSPACE_SLOEBER_44\IRremote_Leonardo\_ReceiveDemo\arduino.ar: $(AR_OBJ)
	@echo 'Starting archiver'
	"E:\Elektro\Sloeber_44\arduinoPlugin\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7/bin/avr-gcc-ar" rcs     E:\WORKSPACE_SLOEBER_44\IRremote_Leonardo\_ReceiveDemo/arduino.ar $(AR_OBJ)

```
switching project to arduino version 1.8.12 and back to 1.8.13 cures the error.

BTW. Sloeber sometimes hangs forever if I try to open a closed former 4.3 project. Killing sloeber, restarting it and then deleting and importing the project works.

jantje, 2021-04-27:
I still hadn't time to take a good look at the documentation you made as I'm struggling with 
#1334 #1333 #1321 which should also fix #1343
Basically the whole boardDescription/multiple config contained some really bad code. I knew that but he it worked it only annoyed me in test environment with 100ths of boards as it was slow.
I've been trying to make it better but bumped into all kind of problems. I think I reached a breakthrough yesterday
Not pretty but it works and is consistent

And then I noticed modifying sloeber.cfg no longer worked :-(

ArminJo, 2021-04-27:
I planned to ask [Andreas Spiess ](https://www.youtube.com/c/AndreasSpiess) to present Sloeber, after he presented [Arduino2.0](https://www.youtube.com/watch?v=nlI_5vxm3bk),  [Visual micro ](https://www.youtube.com/watch?v=7SiprW-goj0) and [PlatformIO](https://www.youtube.com/watch?v=0poh_2rBq7E).
The tutorial was meant as a support for my request.
But I think I will wait until 4.4 is stable.

Good luck with your fight against the nasty bugs!
Armin from Cologne.

jantje, 2021-06-27:
I think I fixed the multiple config/multiple os issues and so I'm back looking at the remaining issues.
Reading your original remakt the things that come to mind are
1) I'm not sure if the action "All prefs files from Sloeber/configuration/settings/' is a smart thing to do. Yes it will keep your preference settings but it may also keep "dirt around"

Secondly if something is wrong always open project properties->arduino->apply close. Though 4.4 should be much better at this  we are talking migration here. (I mean "It is complicated and very likely not covering all cases".)
Basically your workaround includes this action so I don't know whether the "changing the platform" was needed.

closed 2021-08-10 by ArminJo