Sloeber/arduino-eclipse-plugin #1345: build fails for esp32 due to error in subdir.mk : path to compiler is missing

lvlerber, 2021-04-29:
**Describe the bug**
The build stops at the first compile:
[image]
As you see the command in the subdir.mk is "/bin/xtensa-esp32-elf-gcc".  The complete path is missing and the cmd can't be executed in the shell.
[image]

**To Reproduce**
Steps to reproduce the behavior:
1.  In arduino settings add this third party url: https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_dev_index.json
2. In project arduino settings select platform esp32/2.0.0-alpha1 and board NodeMCU-32S
3. try to build your sketch'
4. 

provide the name of the board and the platform version you are using.  
ESP32 nodeMCU-32S

Provide the json url and a screenshot of project properties->arduino  
[image]

**Expected behavior**
The .mk file should contain a correct compile command.  In the 1.6.0 version that is 
"C:\Apps\eclipse\arduinoPlugin\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM .......(with many, many extra parameters.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: w10
 - Browser chrome
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
I know it is an alpha version, but 1.6.0 doesn't contain the core sources for - amongst others - esp_event.

jantje, 2021-04-29:
Newer versions of ESP32 do not work on Sloeber V4.3
V4.4 is in beta and still has some issues

closed 2021-06-04 by jantje

gfvalvo, 2021-06-10:
I see this issue was closed. Is there now a Sloeber with support for latest ESP32 core?

jantje, 2021-06-10:
Due to massive support; V4.4 is still in beta.
Issues are closed when "fixed" not when released. This is what the "fixed in nightly" tag is used for.