Sloeber/arduino-eclipse-plugin #1346: Unable to install Sloeber

croccodillo, 2021-05-14:
**Describe the bug**
I can't install the plugin into Eclipse (last version, 2021-03).
I can't install the Loeber IDE (4.3.3).

Everytime I try to install the plugin or the complete IDE, I get the following error:
[image]

**To Reproduce**
Simply try to install the plugin

**Expected behavior**
I'd like to be able to install the Sloeber

**Screenshots**
See above

**Desktop (please complete the following information):**
 - OS: Windows 10 20H2
 - Browser: Chrome

Thank you in advance!

Ciao,
Giovanni

jantje, 2021-05-14:
It seems to me sloeber plugin is installed fine (which it should not due to #1263) but the configuration fails.
Sloeber does not come with the avr platform and therefore installs a platform at startup as part of what we call configuration.
In your case it seems Sloeber decided to download a file and fails to unzip it.
As Sloeber caches these files it could be the first download was corrupt and so stopping sloeber, deleting the file, starting sloeber may fix your problem. If that does not fix it try to download the file yourself and put it in the downloads location. Your local settings may make sloeber to fail to download (This is no good workaround as Sloeber needs to download stuff (libraries/platform/compilers...) from time to time and that will fail)
The location of the file I would delete is in the error message

jantje, 2021-05-23:
closing as to no response

closed 2021-05-23 by jantje