Sloeber/arduino-eclipse-plugin #1347: Release testing 4.4 test case 4.3 -> 4.4 Win10 (Environment is empty for 2. config + opening a project hangs forever)

ArminJo, 2021-05-20:
I just updated eclipse.

Environment is almost empty after startup for the second and following configurations! 
Like in this example Test_No_GOLD_CAP_ATTACHED configuration is the second one. The first configuration works well. Even after deleting the Test_No_GOLD_CAP_ATTACHED config and building anew I get the same empty Environment after startup.
I can see this for other projects too!
[image]

Content of .sproject
```
Config._Release_GOLD_CAP_ATTACHED.board.BOARD.ID=attinyx7micr
Config._Release_GOLD_CAP_ATTACHED.board.BOARD.MENU.LTO=enable
Config._Release_GOLD_CAP_ATTACHED.board.BOARD.MENU.bod=disable
Config._Release_GOLD_CAP_ATTACHED.board.BOARD.MENU.burnmode=upgrade
Config._Release_GOLD_CAP_ATTACHED.board.BOARD.MENU.millis=enabled
Config._Release_GOLD_CAP_ATTACHED.board.BOARD.MENU.neopixelport=porta
Config._Release_GOLD_CAP_ATTACHED.board.BOARD.MENU.pinmapping=new
Config._Release_GOLD_CAP_ATTACHED.board.BOARD.MENU.resetpin=reset
Config._Release_GOLD_CAP_ATTACHED.board.BOARD.MENU.sketchclock=1external16
Config._Release_GOLD_CAP_ATTACHED.board.BOARD.TXT=${SLOEBER_HOME}/arduinoPlugin/packages/ATTinyCore/hardware/avr/1.5.2/boards.txt
Config._Release_GOLD_CAP_ATTACHED.board.PROGRAMMER.NAME=Default
Config._Release_GOLD_CAP_ATTACHED.board.UPLOAD.PORT=COM8
Config._Release_GOLD_CAP_ATTACHED.compile.sloeber.extra.all=
Config._Release_GOLD_CAP_ATTACHED.compile.sloeber.extra.archive=
Config._Release_GOLD_CAP_ATTACHED.compile.sloeber.extra.assembly=
Config._Release_GOLD_CAP_ATTACHED.compile.sloeber.extra.c.compile=
Config._Release_GOLD_CAP_ATTACHED.compile.sloeber.extra.compile=
Config._Release_GOLD_CAP_ATTACHED.compile.sloeber.extra.cpp.compile=-DGOLD_CAP_ATTACHED -DNO_BUTTON_RELEASE_CALLBACK -DUSE_ALWAYS_CLI_SEI_GUARD_FOR_OUTPUT
Config._Release_GOLD_CAP_ATTACHED.compile.sloeber.extra.link=-fmerge-all-constants
Config._Release_GOLD_CAP_ATTACHED.compile.sloeber.size.switch=true
Config._Release_GOLD_CAP_ATTACHED.compile.sloeber.warning_level=ALL
Config._Release_GOLD_CAP_ATTACHED.compile.sloeber.warning_level.custom=
Config._Release_GOLD_CAP_ATTACHED.other.IS_VERSION_CONTROLLED=false
Config._Test_No_GOLD_CAP_ATTACHED.board.BOARD.ID=attinyx7micr
Config._Test_No_GOLD_CAP_ATTACHED.board.BOARD.MENU.LTO=enable
Config._Test_No_GOLD_CAP_ATTACHED.board.BOARD.MENU.bod=disable
Config._Test_No_GOLD_CAP_ATTACHED.board.BOARD.MENU.burnmode=upgrade
Config._Test_No_GOLD_CAP_ATTACHED.board.BOARD.MENU.millis=enabled
Config._Test_No_GOLD_CAP_ATTACHED.board.BOARD.MENU.neopixelport=portb
Config._Test_No_GOLD_CAP_ATTACHED.board.BOARD.MENU.pinmapping=new
Config._Test_No_GOLD_CAP_ATTACHED.board.BOARD.MENU.resetpin=reset
Config._Test_No_GOLD_CAP_ATTACHED.board.BOARD.MENU.sketchclock=1external16
Config._Test_No_GOLD_CAP_ATTACHED.board.BOARD.TXT=${SLOEBER_HOME}/arduinoPlugin/packages/ATTinyCore/hardware/avr/1.5.2/boards.txt
Config._Test_No_GOLD_CAP_ATTACHED.board.PROGRAMMER.NAME=Default
Config._Test_No_GOLD_CAP_ATTACHED.board.UPLOAD.PORT=COM8
Config._Test_No_GOLD_CAP_ATTACHED.compile.sloeber.extra.all=
Config._Test_No_GOLD_CAP_ATTACHED.compile.sloeber.extra.archive=
Config._Test_No_GOLD_CAP_ATTACHED.compile.sloeber.extra.assembly=
Config._Test_No_GOLD_CAP_ATTACHED.compile.sloeber.extra.c.compile=
Config._Test_No_GOLD_CAP_ATTACHED.compile.sloeber.extra.compile=
Config._Test_No_GOLD_CAP_ATTACHED.compile.sloeber.extra.cpp.compile=-DNO_BUTTON_RELEASE_CALLBACK -DUSE_ALWAYS_CLI_SEI_GUARD_FOR_OUTPUT
Config._Test_No_GOLD_CAP_ATTACHED.compile.sloeber.extra.link=-fmerge-all-constants
Config._Test_No_GOLD_CAP_ATTACHED.compile.sloeber.size.switch=true
Config._Test_No_GOLD_CAP_ATTACHED.compile.sloeber.warning_level=ALL
Config._Test_No_GOLD_CAP_ATTACHED.compile.sloeber.warning_level.custom=
Config._Test_No_GOLD_CAP_ATTACHED.other.IS_VERSION_CONTROLLED=false
```

after changing one Arduino board specification field and clicking on apply the environment tab changes to:
[image] and everything is fine until the next Sloeber startup.

**AND**

Changing boards does not change the board specification fields immediately,
Below I changed board type, but see the fields for the old board type.
[image]
After apply+close and reopen properties again I see the changes values.
[image]

**AND**
Opening a 4.3 project (RobotArmControl or OpenWindowAlarm and a lot others) does hang up Sloeber (but CPU is at 2% => deadlock?) 

After more than 2 hours of documenting bugs I submit this issue now, it may not be complete...

jantje, 2021-05-20:
Thanks for testing
I made the last changes knowing they are not complete
They had been on my pc and I had to check them in because of problems on my pc I didn't want to risk to loose them.
Basically I had to change quite a lot of some basic functionality to fix the issue and there are probably more knock off effects than problems solved right now (for sure with multiple config)
I'll us these conclusions to test the chanbges

jantje, 2021-05-23:
>Opening a 4.3 project (RobotArmControl or OpenWindowAlarm and a lot others) does hang up Sloeber (but CPU is at 2% => deadlock?)

If this issue still exists after the fixes can you send me the project?

jantje, 2021-05-23:
I noticed the upgrade tests fail (recursively).
More changes coming ....

ArminJo, 2021-05-23:
> If this issue still exists after the fixes can you send me the project?
You already have all my projects 😀 see https://github.com/Sloeber/arduino-eclipse-plugin/issues/1333

jantje, 2021-06-27:
I confirm the hang on robotarmcontrol :-(

jantje, 2021-06-27:
I'm not sure you had the exact same issue as I had but The hang on my system is fixed with this change :-)
As on my system it was a deadlock with low cpu usage I am hopeful

closed 2021-07-11 by jantje