Sloeber/arduino-eclipse-plugin #1349: How to freeze the libraries versions in Slober?

marcelocsi, 2021-06-02:
This is a request. When developing code, it is a very good practice to freeze source code and tools in order to replicate to solve issues from the field.

Therefore, how do I freeze the source code and tools or have a list of the libraries versions in a file where this file can be saved with the project and automatically reconfigure the project when a specific version is required?

jantje, 2021-06-03:
>Therefore, how do I freeze the source code and tools

Like everybody else in the it industry .... make a virtual machine

> or have a list of the libraries versions in a file where this file can be saved with the project and automatically reconfigure the project when a specific version is required?

Well .... it is very likely that in a real project you will change the code of some libraries so you can not just "download an reconfigure" unless you make your own library json file referring to your libraries (which can be easily done in sloeber) which implies a whole release approach for the libraries probably independent of your projects.

However Sloeber currently only supports 1 version of a library to be installed in a Sloeber install (Something which may change one day) so storing the library version for the project can currently not be supported as you can have 2 projects in the sloeber install requiring a different version of the same library.

In other words these are the typical questions a "development shop consultant" gets and the only honest answer is "it depends, if you pay me well enough I will look at the details of your situation and figure something out that can work, but don't be surprised if you change something and the whole thing falls apart"

marcelocsi, 2021-06-03:
This is obvious that this is for industry, this is where I came from for more than a decade working on embedded coding releasing code for mass production 2M+ units/year, and this is what I am looking for it right now.

Virtual machine? No, thanks. If something crashes on Eclipse on that machine, my work is to rebuild everything once again - this does not seem to be smart and, this is not using versioning control rationally. My goal was to setup something in git with sloeber in a fashion that I can easily recreate a new sandbox from a git tag and recompile my source code (this is ok) with libraries used in that version, which is not ok because libraries from Arduino might have new versions from time to time.

It looks like my answer is no, I can't pay you money but I would certainly pay beers on a bar!

Thank you for your prompt reply as usual.

closed 2021-06-03 by marcelocsi

marcelocsi, 2021-06-03:
You know what? In my former employer, we have stored everything in the repository. Everything! Like libraries, project specific codes, tools, compilers, everything under one tag. That would have no point in have virtual machine anymore.

That company has several sites across globe and there is a considerable turnover of people, so, the key point was to install the repository tools to create sandboxes, unzip the configured Eclipse with all required plugins preconfigured (you needed to enter only your user id and password) and in few minutes you could grab any version from production and recompile "as it for production". The point here is to rebuild the executable from source, replicate the failure from field and solve it. So, the code was for embedded systems, so even the hash table of source code would match because even a change in the compiler version would mess it up and the code was tracked by some agencies like UL/IEC/VDE (or any other country with standards).

I truly understand your pain about people; sometimes I wanna beat somebody with a dead cat until the caw meows again. 

Well, regarding beer, certainly there is no distance to have a good beer! I am planning to do some travel, btw, where are you from? Maybe I can visit your country, learn your culture by drinking good beer.

jantje, 2021-06-03:
>You know what? In my former employer, we have stored everything in the repository. Everything! Like libraries, project specific codes, tools, compilers, everything under one tag. That would have no point in have virtual machine anymore.

That is untill you can't run the compilers on "modern OS'es". We had problems with legacy build software in the build chain which could only run on a very specific os (that ran end of support/hardware) so we ran these build machines on a vm anyway so taking a step further and put the build vm's in the version control tool came naturally.
The problem with this whole thing is .... it never stops. As I hinted on before, once you check in your vm how are you going to guarantee that future vm hosting software is going to support your vm? And do you really care that the version you build today is still build-able in 10 years? I would guestimate about 99% of all software that was out there 10 years ago will need serious work to be rebuild today.

> I wanna beat somebody with a dead cat until the caw meows again.

I like cat's. I wouldn't do that to them it would break my hart 😸 

>Maybe I can visit your country, learn your culture by drinking good beer.

If you like beer and you end up in western Europe; visit a small country Belgium. I'll be there ;-)
https://en.wikipedia.org/wiki/Beer_in_Belgium

marcelocsi, 2021-06-03:
Embedded compilers runs fine in old machines like Win98 or in newest machines like Windows 10, without any issues. So, no need to use VM but I do relate your pain about the need when using some other languages.

Belgium is my favorite place for beer! I need to explore more, I was at "drei punke Ländern" from Germany side but I need for sure explore more! Let's discuss it in a near future! lol

Thanks a lot for your support. 

Kind regards!