Sloeber/arduino-eclipse-plugin #1351: Create new sloeber cpp/h project from existing Arduino sketch

Pestimist, 2021-06-11:
Hi,

I wondered if there is a direct way to import existing Arduino sketches into a new eclipse cpp/h project?
I eventually want to get rid of the .ino files...

What I now did to achieve this task is 

- copy the folder with the existing project into sloeber workspace
- get rid of unnecessary files
- create a new "sketch" in sloeber choosing cpp template
- copy contents of .ino file in corresponding .cpp file, taking care not to overwrite the auto-generated #include
- compile
- some errors where shown under "problems" although the build was successful
- rebuilding index solved that

Now my question is if I choose the right way to do this or is there a more straight-forward option to import existing sketches?

Cheers,
Bob

jantje, 2021-06-11:
I don't think there is a "right way"
I mostly create a new project and then drag/drop the code onto the project and select copy/link as what is appropriate at the time.
When stuff is in version control I prefer taking that route

Pestimist, 2021-06-16:
Hi jantje,

Thanks for the reply.
So I basicall took the right route to create a new project (with the .ino<->.cpp rename detour)

P.S. thank you and all @sloeber for the work and support !

closed 2021-06-16 by jantje