Sloeber/arduino-eclipse-plugin #1352: Sloeber 4.3.3, partial recompiling on ESP8266 Arduino Core 3.0.0 not working

SturmWolfgang, 2021-06-13:
**Describe the bug**
on ESP8266 Arduino Core 3.0.0 after a code change (in the main .ino file) the changes are not compiled.
I always have to "clean project" "rebuild project"

**To Reproduce**
this works, and is expected behavior for 3.0.0 also
1.) have ESP8266 Arduino Core 2.7.4 installed, and configured in the projects properties
2.) have a project with it, built it and flash it (currently it is an IOD-09Th, but hardware is not relevant, same on Huzzah)
3.) change a line of code
4.) built again (only changes are rebuild), so takes seconds, console output:
```
10:16:39 **** Incremental Build of configuration Release for project ESP8266 ****
"C:\\Programmieren\\Sloebe433\\arduinoPlugin\\tools\\make\\make" all 
'Building file: ..\sloeber.ino.cpp'
'Starting C++ compile'
```
5.) upload --> changes are included

this does not work:
1.) have ESP8266 Arduino Core 3.0.0 installed, and configured in the projects properties
2.) have a project with it, built it and flash it
3.) change a line of code
4.) built again , console output:
```
10:21:07 **** Incremental Build of configuration Release for project ESP8266 ****
"C:\\Programmieren\\Sloebe433\\arduinoPlugin\\tools\\make\\make" all 
make: *** No rule to make target 'C\:\Programmieren\Sloebe433\arduinoPlugin\packages\esp8266\hardware\esp8266\3.0.0\cores\esp8266/Arduino.h', needed by 'sloeber.ino.cpp.o'.  Stop.
"C:/Programmieren/Sloebe433/arduinoPlugin/tools/make/make all" terminated with exit code 2. Build might be incomplete.
```
5.) upload --> changes are not included

workaround
1.) clean project
2.) rebuild project (takes minutes as all the core is rebuild)

**Desktop:**
 -Windows 10 Home, 21H1

Any comments how to rebuild projects with core 3.0.0 without rebuilding the entire one (and waiting minutes) are appreciated
Is there a way to "clean" just a single file to reduce rebuild time?

jantje, 2021-06-13:
your output shows a incorrect filename and the build fails. So partial recompile starts but fails
C\:\Programmieren\Sloebe433\arduinoPlugin\packages\esp8266\hardware\esp8266\3.0.0\cores\esp8266/Arduino.h

Have a look at what is inside sloeber.ino.cpp

uzi18, 2021-06-13:
Path is wrong: C\:\Programmieren\Sloebe433\arduinoPlugin\packages\esp8266\hardware\esp8266\3.0.0\cores\esp8266/Arduino.h

SturmWolfgang, 2021-06-13:
thanks for the quick reply, unfortunately the answers do not help, as I did nothing with the path

I stripped the problem down to the absolute minimum:
1.) Create a new empty sketch with Core 3.0.0, and compile it 
- so empty setup, empty loop, it will be built successfully
2.) just insert some line e.g. "Serial.begin(9600);" in setup and rebuilt by clicking the check mark from the Arduino toolbar
error as above (10:21:07), no rebuilt
- did not touch sloeber.ino.cpp, which looks good to me:
```
#include "Arduino.h"
#include "Arduino.h"

void setup() ;
void loop() ;

#include "ESP8266_IOD9TH_Core3.ino"
```
- did not change any path

uzi18, 2021-06-13:
Generated makefile has some errors because path mentioned in error looks wrong

hreintke, 2021-06-15:
See issue https://github.com/Sloeber/arduino-eclipse-plugin/issues/1198

Will probably also be fixed when https://github.com/esp8266/Arduino/commit/74d675cffab94d18c709f7a322a537d8957b8e7e is included (is in master but not yet in a released version.

jantje, 2021-06-15:
@hreintke 
Thanks :-) . I had already forgotten about that issue
I'm closing this as a duplicate of #1198 
Please reopen with comments when this is not correct

closed 2021-06-15 by jantje